zephyrproject-rtos/zephyr #26830: Support FIFO mode in sensor drivers for sensors with internal FIFO such as lis2ds12, lis2dw12

songshan99, 2020-07-13:
**Is your enhancement proposal related to a problem? Please describe.**

I'm using lis2ds12/lis2dw12 with nRF9160 DK. I add the sensor in overlay file in the device tree:

```
&spi3 {
	compatible = "nordic,nrf-spim";
	status = "okay";
	sck-pin = <23>;
	mosi-pin = <24>;
	miso-pin = <25>;
	cs-gpios = <&gpio0 22 0>;

	lis2dw12@0 {
		compatible = "st,lis2dw12";
		spi-max-frequency = <1000000>;
		reg = <0>;
		irq-gpios = <&gpio0 21 GPIO_ACTIVE_HIGH>;
		label = "LIS2DW12";
	};
};
```
in the project configuration file, sensor configuration is:
```
CONFIG_SENSOR=y
CONFIG_SENSOR_LOG_LEVEL_DBG=y
CONFIG_LIS2DW12=y
CONFIG_LIS2DW12_TRIGGER_OWN_THREAD=y
CONFIG_LIS2DW12_INT_PIN_1=y
CONFIG_LIS2DW12_ODR_200=y
```

In the code, register a trigger handler function, and read the x/y/z acceleration after every triggered event:
```
bool init_LIS2DW12()
{
	struct device *lis2dw12 = device_get_binding(DT_LABEL(DT_INST(0, st_lis2dw12)));
	struct sensor_trigger trig;
	trig.type = SENSOR_TRIG_DATA_READY;
	trig.chan = SENSOR_CHAN_ACCEL_XYZ;
	sensor_trigger_set(lis2dw12, &trig, lis2dw12_trigger_handler);
}

static int sample_count=0
static void lis2dw12_trigger_handler(struct device *dev,
				    struct sensor_trigger *trig)
{
	struct sensor_value accel[3];
	sensor_sample_fetch_chan(dev, SENSOR_CHAN_ACCEL_XYZ);
	sensor_channel_get(dev, SENSOR_CHAN_ACCEL_XYZ, accel);
	sample_count++;
	printf("sample count: %d\r\n", sample_count);

	//put the samples to another buffer and wait for further process in another task...
}
```
So the sensor_sample_fetch_chan() and sensor_channel_get() only reads one sample at a time.
After a bunch of test, I found that the code only works when LIS2DW12 ODR (output date rate) is <=100Hz. 
At 200Hz ODR, the interrupt is so frequent that the above code of reading the X/Y/Z acceleration sample one by one is not good enough to handle such frequent interruption.

**Describe the solution you'd like**

A lot of sensors supported by Zephyr has FIFO function. (e.g. lis2ds12, lis2dw12, LSM6DSO , MPU6050). It is nice if the sensor driver can offer functions to enable the FIFO mode and perform bulk SPI reading (prefer to use DMA too).

**Describe alternatives you've considered**
For now it is also possible to NOT to use sensor_sample_fetch_chan() and sensor_channel_get() functions, but to get the context pointer and directly call the LIS2DW12 register read function, or even SPI read functions, such as example below:
```
static void lis2dw12_trigger_handler(struct device *dev, struct sensor_trigger *trig)
{
	struct lis2dw12_data *data = (struct lis2dw12_data *)dev->driver_data;
	stmdev_ctx_t *ctx = ((struct lis2dw12_data *)dev->driver_data)->ctx;

	///////////////////////////////////////////
	//Below is blocking read, this is the most stable when working at 200Hz ODR
	if(ctx->read_reg(data, 0x28U, buffer_rx, 192)) {
		printf("lis2dw12_spi_read_async() error.\r\n");
	}
}
```

This approach works. But the code depends on what function are available in the sensor's driver/middleware. 
It would still be better if sensor driver provides FIFO functions since that'll be a generic implementation for all sensors.

**Additional context**
NO.

gmarull, 2020-08-24:
I think it would be worth re-opening https://github.com/zephyrproject-rtos/zephyr/pull/17434 as it would better suit applications where high sensor data rates need to be processed. Support for sensors that have a FIFO can be implemented on current sensor API, e.g. could be done reading FIFO contents on trigger or using a periodic timer and then `_fetch` would be a consumer. 

Still there are a couple of problems I see in the current API. First is an efficient way of accessing data in blocks. In case sensor FIFO was used data would be read internally in bursts, but external access would made on a sample basis (requiring two calls, `_fetch` + `_get`). The second point is that sensor API does data conversion on `_get`. This is not always desired, e.g. I may just want to transmit raw values, apply some data processing on non-converted data, etc.

jchan123456, 2022-10-20:
Is there any status update on this FIFO support issue?  I am looking for a solution to use the internal hw FIFO of the sensor.  Thanks,

JC

teburd, 2023-10-23:
Should be fixed by #60063

teburd, 2025-01-13:
Sensor WG 01/13/2025 Re-assigned to @avisconti

avisconti, 2025-01-15:
Mmmh, I need to implement a similar work done for lsm6dsv16x in #83094. The lis2dw12 sensor was not the next one I had in mind, moreover this request is very old. Wondering if it is still needed...

avisconti, 2025-01-15:
But if it is I can work on it in the next days...

teburd, 2025-01-16:
Would need issue authors to chime in on importance. I don’t need it myself.

avisconti, 2025-01-17:
> Would need issue authors to chime in on importance. I don’t need it myself.

Just because it seems a Sensor WG decision

teburd, 2025-01-17:
> > Would need issue authors to chime in on importance. I don’t need it myself.
> 
> Just because it seems a Sensor WG decision

Re-assigning to you was. It would be great to see ST take maintainership of the ST sensor drivers