zephyrproject-rtos/zephyr #26833: RFC: Support file open flags to fs and POSIX API

de-nordic, 2020-07-13:
## Introduction

The `fs_open` lacks support for flags that would allow to choose whether file is created when does not exist, only opened when exists, is opened for read or write, etc. This is proposal to add support for such flags and propagate this support to POSIX file API.

### Problem description

Addition of open flags allows to control whether file is created when does not exist or only opened when exists. It enables opening file for append or select whether file will be readable and/or writable after opening.
Right now file is always opened, when it does not exist it will be created, which sometimes may not be desired effect.

### Proposed change

Add support for flags to `fs_open` that would allow to choose whether file will be created, if it does not exist, opened at the end (append) and whether read and/or write will be allowed.

## Detailed RFC

The detailed RFC is provided via implementation: https://github.com/zephyrproject-rtos/zephyr/pull/26305
Short description of changes:
 - The `fs_open` function declaration has been changed to accept one more parameter for flags, in addition to the two it is already taking.
 - The `FS_O_CREATE`, `FS_O_APPEND`, `FS_O_READ`, `FS_O_WRITE` have been defined.
 - The cpp macro has been provided that keeps original behavior, which resembles combination of flags `FS_O_CREATE | FS_O_READ | FS_O_WRITE`, for code that invokes function by two parameters.
 - API tests have been provided and FAT FS and LittleFS tests have been changed to check the new feature.
 - POSIX `open` function has been modified to accept POSIX flags that are then translated and passed to `fs_open`

### Dependencies

The change to `fs_open` does not affect old code, as the "default" behavior of two parameter function calls have been preserved.
The change to POSIX `open` breaks compatibility with previous behavior, but the behavior has been POSIX non-compliant anyway.

carlescufi, 2020-07-14:
API meeting 14th July: no objections to the change, barring finishing  the review comments in the PR being addressed.

de-nordic, 2020-08-03:
Completed with https://github.com/zephyrproject-rtos/zephyr/pull/26305.

closed 2020-08-03 by de-nordic