zephyrproject-rtos/zephyr #26857: High CPU load when running samples/subsys/usb/audio/headphones_microphone

koffes, 2020-07-14:
**Is your enhancement proposal related to a problem? Please describe.**
Running the mentioned sample on the nRF52840. Vanilla setup, no changes to config.

The CPU loads are:
Idle: 4 %
Playback 33 %
Paused playback: 33 %. Goes down to 26 % after a while
Play and record: 52 %

**Describe the solution you'd like**
Please try to enable DMA. This will likely reduce CPU load significantly.

@emob-nordic @alexsven

alexsven, 2020-12-03:
I have done some testing on the headset sample with a nRF5340.

I have tested two set ups, one with stereo hp + stereo mic (vanilla sample) and another where I remove one channel from both hp and mic from the overlay file, thus creating a mono device.

The starting CPU load before any audio has been played is 10%, then I start the audio and I see that the CPU load goes up to 40% for stereo hp and 37% for mono hp. When I at the same time sink the mic line by recording with Audacity I get 63% for stereo and 57% for mono. When stopping audio and stopping sink I get 33% for both usecases after a while.

alexsven, 2020-12-03:
Ran one more test: Headphones only
Starting CPU load: 7%
Playing audio: 28%
Paused: 21%

ddavidebor, 2021-04-24:
@koffes I see this has been opened for a long time. Is there an update on this?

koffes, 2021-04-26:
@ddavidebor We are waiting for this issue to be resolved as part of #26935

alexsven, 2022-02-02:
Testing again with v3.0.0-rc1 and it seems headphones only are using 19% CPU, and turning on mic it goes up to 22% and then an additional 10% is consumed by sysworkq

alexsven, 2022-02-02:
Testing on nRF52840 I get:
Stereo headphones : 21%
Stereo headset: 24% for USB thread, 11% for sysworkq

tmon-nordic, 2023-10-20:
#64174 makes rather significant amount of cpu time available to application because it eliminates busy loop inside interrupt while waiting for DMA to finish.