zephyrproject-rtos/zephyr #26862: Bluetooth: GATT: CCC is not properly stored

MarekPieta, 2020-07-14:
**Describe the bug**
Zephyr Bluetooth peripheral may not store CCC in non-volatile memory in case of multiple CCC change (due to race condition).

**To Reproduce**
I replicated the problem using nrf_desktop peripheral connected to my Windows10 host (the host performs multiple CCC writes). That causes a race condition and triggers the issue.

`ccc_delayed_store` is called as `k_delayed_work` handler. Before its finished, the `bt_gatt_attr_write_ccc` may be called. In that case the new CCC is not stored properly in non-volatile memory (it may not be taken into account yet). The work is not resubmitted too. I think there is false assumption that the `ccc_delayed_store` cannot be interrupted by `bt_gatt_attr_write_ccc`. I added additional logs to see what caused the problem.

**Expected behavior**
All CCCs for bonded peer should be properly stored in non-volatile memory.

**Impact**
Bluetooth notifications are not properly enabled after reboot and reconnection.

**Environment**
 - OS: Linux Ubuntu
 - Zephyr SDK 0.11.1 
 - commit SHA: ac117c8d6377834f4b89c195d7ee4711d8ee32f3

MarekPieta, 2020-07-14:
I tried moving `ccc_store->conn_list[i] = NULL;` above `bt_gatt_store_ccc`. It worked fine as hotfix (the issue was not replicated - the work was resubmitted). I may provide the fix as PR.

joerchan, 2020-07-15:
Based on the bug-report a workaround would be to disable `CONFIG_BT_SETTINGS_CCC_STORE_ON_WRITE`.

PR #26872 (2020-07-16): bluetooth: gatt: Fix storing subscriptions
Change fixes storing subscriptions in settings. CCC write can interrupt the ccc_delayed_store. Without the change, new CCC was not stored in non-volatile memory and the work that stored CCC was not resubmitted. That resulted in lost subscription after reboot.

This fixes: #26862
commit a6e8cf43 (2020-07-16): bluetooth: gatt: Fix storing subscriptions

Change fixes storing subscriptions in settings. CCC write can interrupt
the ccc_delayed_store. Without the change, new CCC was not stored in
non-volatile memory and the work that stored CCC was not resubmitted.
That resulted in lost subscription after reboot.

Fixes: #26862
closed 2020-07-16 by carlescufi