zephyrproject-rtos/zephyr #26865: Generic function for obtaining the privilege stack bounds for x86

maksimmasalski, 2020-07-14:
**Describe the bug**
`arch.priv_stack_start` is implemented only for ARM and ARC. Necessary to have a generic function for obtaining the privilege stack bounds for x86 too.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to use `thread->arch.priv_stack_start` somewhere in the code
2. Build it for x86
3. It can't build for x86, because that feature not implemented for x86

**Expected behavior**
Obtaining the privilege stack bounds must be implemented for all architectures.

**Impact**
For x86 no generic function for obtaining the privilege stack bounds

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.11.4
 - Commit SHA d1f4a91183ac77cd3e38ad897ac2cc191b23661a

maksimmasalski, 2020-09-07:
@andrewboie Could you take a look at that issue? So after solving it, I can finish my test cases PR #25323

andrewboie, 2020-09-16:
yes will spend some time on this, thanks

maksimmasalski, 2020-11-02:
@andrewboie Hello, any news from you regards that issue?

enjiamai, 2021-06-21:
Hi @nashif @andyross , do we indeed need a function/interface like this?  or we can implement this as a test util? thank you!

andyross, 2021-06-22:
It doesn't seem like a **bad** API to have, but clearly we don't need it generically right now.  In some architectures, there's no MMU state behind kernel stacks, so it's not necessarily something that matters to the hardware.

I guess if nothing else is should be part of CONFIG_THREAD_STACK_INFO, which is where we put this information (optionally) for apps that require it.

enjiamai, 2021-06-22:
Hi @andyross , thanks for your guidance! 
So my understanding is to implement this kind of API within CONFIG_THREAD_STACK_INFO=y, we might not need to implement it for all of the ARCH/platform at this moment.

edersondisouza, 2025-10-06:
This is implemented by https://github.com/zephyrproject-rtos/zephyr/pull/77425

closed 2025-10-06 by edersondisouza