zephyrproject-rtos/zephyr #26868: qemu_x86_64 icount support with SMP

andrewboie, 2020-07-14:
**Describe the bug**
Enabling icount for qemu_x86_64 results in hangs / tests not passing. There has been some discussion about this in https://github.com/zephyrproject-rtos/zephyr/pull/26394

At this time, this is the only platform that doesn't have icount support, fixes for nios2 and arc are in-flight.

**To Reproduce**
Enable CONFIG_QEMU_ICOUNT=y for qemu_x86_64, set a shift value, and watch stuff fail when `sanitycheck` is invoked.

**Expected behavior**
All tests passing.

**Impact**
Not having icount is a severe annoyance as it causes sporadic timing-related test failures which can conceal real bugs.

wentongwu, 2020-07-15:
> I would expect the CPU emulator to emulate instructions on each CPU round-robin, regardless of what the OS is doing.

It is round-robin way on a single vCPU thread, but there is some kind of granularity, TB or TB chains, for icount there is also soonest deadline.

> In other words, it doesn't make sense to me that if we call k_busy_wait() on CPU 0, that nothing happens on CPU 1 the entire time. That seems like broken behavior. Is that really what's happening? 

enabled icount, this is the working modle for multi CPUs case, k_busy_wait is also common instructions, qemu doesn't insert any break/exit code for common instruction.

wentongwu, 2020-08-25:
I think icount mode doesn’t work well on qemu_x86_64 multi core, because QEMU can’t provide accurate timer on multi core platform. And QEMU can’t provide true physical concurrency.

closed 2020-08-27 by andrewboie