zephyrproject-rtos/zephyr #26873: WIFI_ESP: sockets left opened after unexpected reset of ESP

niziak, 2020-07-15:
**Describe the bug**
There are multiple scenario when cold start of ESP chip can occur (deep sleep modes, turn off to save energy, spurious resets).
Reset is detected by driver, but UDP sockets are still kept in connected state (so `AT+CIPSTART=` is not sent)

**Expected behavior**
All sockets should be closed upon reset.

**Impact**
DNS resolver works only during first "session". After powering ESP chip off and on, there is not possible to use resolver.

**Logs and console output**
Correct flow (`AT+CIPSTART` followed by `AT+CIPSEND`):
```
uart:~$ net dns google.com
Query for 'google.com' sent.
[00:00:36.901,428] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x245b9, data 0x20005320
[00:00:36.901,489] <dbg> wifi_esp_offload.esp_sendto: link 0, timeout 0
[00:00:36.901,519] <dbg> wifi_esp_offload.esp_connect: link 0, timeout 5000
[00:00:36.901,580] <dbg> wifi_esp_offload._sock_connect: link 0, ip_proto UDP, addr 8.8.8.8
[00:00:36.901,611] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 49 50 53 54  41 52 54 3d 30 2c 22 55 |AT+CIPST ART=0,"U
44 50 22 2c 22 38 2e 38  2e 38 2e 38 22 2c 35 33 |DP","8.8 .8.8",53
[00:00:36.910,339] <dbg> modem_cmd_handler: RECV
30 2c 43 4f 4e 4e 45 43  54                      |0,CONNEC T
[00:00:36.910,369] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [0,CONNECT] (len:9)
[00:00:36.910,858] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK
[00:00:36.910,858] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [OK] (len:2)
[00:00:36.910,980] <dbg> wifi_esp_offload._sock_send: link 0, len 28
[00:00:36.911,041] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 49 50 53 45  4e 44 3d 30 2c 32 38 2c |AT+CIPSE ND=0,28,
22 38 2e 38 2e 38 2e 38  22 2c 35 33             |"8.8.8.8 ",53
[00:00:36.916,137] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK
[00:00:36.916,137] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [OK] (len:2)
[00:00:36.921,997] <dbg> modem_cmd_handler: RECV
52 65 63 76 20 32 38 20  62 79 74 65 73          |Recv 28  bytes
[00:00:36.927,703] <dbg> modem_cmd_handler: RECV
53 45 4e 44 20 4f 4b                             |SEND OK
[00:00:36.927,764] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [SEND OK] (len:7)
dns: 172.217.16.46
dns: All results received

```
After ESP chip reset (Only `AT+CIPSEND` is sent):
```
[00:00:47.368,591] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x245b9, data 0x20005320
[00:00:47.368,652] <dbg> wifi_esp_offload.esp_sendto: link 0, timeout 0
[00:00:48.168,029] <dbg> wifi_esp_offload._sock_send: link 0, len 28
[00:00:48.168,090] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 49 50 53 45  4e 44 3d 30 2c 32 38 2c |AT+CIPSE ND=0,28,
22 38 2e 38 2e 38 2e 38  22 2c 35 33             |"8.8.8.8 ",53
[00:00:48.173,797] <dbg> modem_cmd_handler: RECV
45 52 52 4f 52                                   |ERROR
[00:00:48.173,828] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [ERROR] (len:5)
[00:00:48.173,858] <dbg> wifi_esp_offload._sock_send: Failed to send command, -5
[00:00:48.173,889] <err> wifi_esp_offload: Failed to send data: link 0, ret -5
```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain: zephyr-sdk-0.11.1
 - Zephyr version: 2.3.99 2eb633d12ed96b90babda1e3ade76941f33375d5

**Additional context**
Reset is detected by driver and  handled. 
```c
MODEM_CMD_DEFINE(on_cmd_ready) {
...
	if (net_if_is_up(dev->net_iface)) {
		net_if_down(dev->net_iface);
		LOG_ERR("Unexpected reset");
	}
...
       net_if_ipv4_addr_rm(dev->net_iface, &dev->ip);
       k_work_submit_to_queue(&dev->workq, &dev->init_work);
}

static void esp_init_work(struct k_work *work)
{
...
	net_if_set_link_addr(dev->net_iface, dev->mac_addr,
			     sizeof(dev->mac_addr), NET_LINK_ETHERNET);
	LOG_INF("ESP Wi-Fi ready");
	net_if_up(dev->net_iface);
...
}
```

I don't know network stack architecture of Zephyr, but at least after removing od IP address from interface all opened sockets should be closed (is this flow possible with offloaded driver?)

There is only one place in driver code, where socket can be freed - by call to  `esp_offload->put()`.

@tsvehagen Please propose correct to way to implement fix.

niziak, 2020-07-15:
See my fix proposal: https://github.com/niziak/zephyr/tree/esp_close_sockets_on_reset

mniestroj, 2020-07-15:
What I see is that net_context is allocated once in DNS implementation in `subsys/net/lib/dns/resolve.c` in `dns_resolve_init()`. `net_context_put()` is never called (when using fixed list of DNS servers), which is why `esp_offload->put()` is never called as well.

So first question is: is DNS resolve implementation correct? Maybe we should change the code to allocate (`net_context_get()` and deallocate (`net_context_put()`) new network context with each resolve attempt? @jukkar could you feedback on this?

jukkar, 2020-07-20:
> So first question is: is DNS resolve implementation correct? Maybe we should change the code to allocate (`net_context_get()` and deallocate (`net_context_put()`) new network context with each resolve attempt?

The `dns_init_resolver()` is called when the network stack is initialized and there is no cleanup for the statically allocated addresses. It would be very heavy operation to do this init for each resolve operation. We could probably have a cleanup function if needed, or allow `dns_init_resolver()` to be called multiple times in which case it could cleanup things from earlier invocation.

mniestroj, 2020-07-20:
> The dns_init_resolver() is called when the network stack is initialized and there is no cleanup for the statically allocated addresses. It would be very heavy operation to do this init for each [...]

`dns_init_resolver()` calls `net_context_bind()` function in which network interface is assigned to the context. I don't think this is right to do during system init, because (available) network interfaces might change in runtime. For example there might be GSM and WiFi, which are both "dynamic". If application tries to resolve some domain, then socket (or net_context) should be bound to the network interface, which is online (has configured IP addresses) at the time.

jukkar, 2020-07-21:
> I don't think this is right to do during system init, because (available) network interfaces might change in runtime. For example there might be GSM and WiFi, which are both "dynamic". If [...]

I agree, if we have interfaces coming up/down, the system should re-do the DNS init or we could do the bind() only when application is trying to resolve address. We could start to monitor network connectivity events or similar, and act accordingly. Atm I do not have time to look into this, patches are welcome if there are anyone willing to investigate this.

closed 2020-10-04 by github-actions[bot]