zephyrproject-rtos/zephyr #26874: STM32: Multiple ADCs using same interrupt not possible

lsangild, 2020-07-15:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
Some STM32 chips have multiple ADCs. At the moment e.g. the STM32F407 ADC1 is the only available one out of three.

**Describe why you are asking for this support?**
On some of the larger chips (e.g. LFPQ144) the ADC3 has dedicated inputs I would like to use.

**If this is a new board or SoC, please state whether you are willing to maintain the Zephyr support for it if it is included in the main tree**
I am willing to implement this feature if I get some help.

I have attempted to add ADC2 and ADC3 to the device tree, but when multiple of them have the status "okay", I get the error 
```
gen_isr_tables.py: error: multiple registrations at table_index 18 for irq 18 (0x12)
Existing handler 0x801c283, new handler 0x801c283
Has IRQ_CONNECT or IRQ_DIRECT_CONNECT accidentally been invoked on the same irq multiple times?
```
Which is due to the fact that the three ADCs share interrupt 18 as a global ADC interrupt.

Addition is done in a .overlay:
```
/ {
    soc {
        adc2: adc@40012100 {
            compatible = "st,stm32-adc";
            reg = <0x40012100 0x100>;
            clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00000100>;
            interrupts = <18 0>;
            status = "disabled";
            label = "ADC_2";
            #io-channel-cells = <1>;
        };
        adc3: adc@40012200 {
            compatible = "st,stm32-adc";
            reg = <0x40012200 0x100>;
            clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00000100>;
            interrupts = <18 0>;
            status = "disabled";
            label = "ADC_3";
            #io-channel-cells = <1>;
        };
};
```
Could there be an option to nest the ADCs? This could be beneficial as they also share the ADC_Common register, which defines e.g. interleaving options.

intercreate, 2020-08-27:
This enhancement would be great!  @lsangild how did you end up dealing with this limitation?

lsangild, 2020-08-28:
> This enhancement would be great! @lsangild how did you end up dealing with this limitation?

I have not dealt with it. My application does not necessarily need the dedicated inputs at the moment, so I am using only ADC1.
Later on in the development process, if this feature is not implemented, I will probably use the HAL or LL libraries to circumvent the limitations present in Zephyr.

erwango, 2021-11-02:
@Jeepgoing please check https://github.com/zephyrproject-rtos/zephyr/pull/37585

Jeepgoing, 2021-11-03:
> @Jeepgoing please check #37585

LGTM

```
#ifdef CONFIG_ADC_STM32_SHARED_IRQS
static bool init_irq = true;
#endif
```

But i am not sure is the best solution.

erwango, 2022-01-28:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/41497

closed 2022-01-28 by erwango