zephyrproject-rtos/zephyr #26889: Need syscall that allows to force sleep state

albertofloyd, 2020-07-16:
**Is your enhancement proposal related to a problem? Please describe.**
Certain applications do require minimum latency to enter low power state under certain external conditions:
i.e. Multi CPU-s systems performing power throttling.

If one of these SoC is running Zephyr the host/master controller would request the main app to put the SoC in lower power state.

While latest update to power subsystem has decreased the latency to enter deep sleep, this greatly varies depending of the number of threads used in a application. 

Refer to samples/boards/mec/power_management and see latency values.

**Describe the solution you'd like**
Add a syscall like _sys_suspend that allow to bypass residency policy from user mode/app side.

**Describe alternatives you've considered**
Forcing by suspending all threads and calling k_sleep(MIN_RESIDENCY + delta) works but time to reach low power state is reached is unbounded.

ceolin, 2020-09-26:
I've mixed feelings about it. `sys_pm_force_power_state` don't have a kernel object associated so, there is not way to limit it to specific thread which mean that all user threads will be able to do it ....

albertofloyd, 2020-09-30:
@ceolin  In the scenario mentioned above not all users threads need to force system sleep state, probably just a designated thread.
If we can have that mechanism to force system sleep limited to a specific thread that would satisfy the scenario.

nashif, 2020-09-30:
> @ceolin In the scenario mentioned above not all users threads need to force system sleep state, probably just a designated thread.
> If we can have that mechanism to force system sleep limited to a specific thread that would satisfy the scenario.

@albertofloyd  you are referring to usermode and system calls here which is terminology used when memory protection and isolation are involved. Are you specifically asking for this to be a system call that can be called from a usermode thread in an environment where MPU/MMU is being used?

albertofloyd, 2020-10-06:
@nashif  yes.
Need a way to bypass power subsystem policy and force sleep state from usermode side. 
If only possible from supervisor thread when MMU is being used that still enables the scenario we want to cover

closed 2021-03-07 by nashif