zephyrproject-rtos/zephyr #26896: STM32: mcu goes to deep sleep inadvertently when using PM.

BBekker, 2020-07-16:
**Describe the bug**
When using the (experimental) STM32L4 power management, deepsleep is not disabled when waking up, resulting in the next `WFI` call also going to deepsleep.

**To Reproduce**
Steps to reproduce the behavior:
1. use stm32l* PM
2. Do a long wait to go to deep sleep.
3. Do a short wait that should not go do deep sleep
4. System goes to deep sleep anyway

**Expected behavior**
System only goes to deep sleep when executing `sys_set_power_state` with the appropriate power state.

**Fix**

In the soc specific power.c, add  `LL_LPM_EnableSleep` to `_sys_pm_power_state_exit_post_ops`. This sets the system configuration bit to use regular sleep instead of deepsleep when executing WFI.

I'm not sure if this is a bug, or if i'm just using the Power management code outside its expected use case.

FRASTM, 2020-07-23:
@BBekker  the soc residency policy does not go to sleep for a _too_ "short wait"
do you mean that `_sys_pm_power_state_exit_post_ops`() has to LL_LPM_EnableDeepSleep (like the `sys_set_power_state` function does) ?

BBekker, 2020-07-24:
The bug is that the MCU goes to sleep, even when the residency policy does not decide it has to go to sleep.

`sys_set_power_state` uses `LL_LPM_EnableDeepSleep` to set the `SLEEPDEEP` bit in the cortex-m33 system control register.
https://developer.arm.com/documentation/100235/0003/the-cortex-m33-peripherals/system-control-block/system-control-register?lang=en

But since that bit is not reset after wakeup, it stays active.
`_sys_pm_power_state_exit_post_ops` should disable that bit again to prevent the system going to deep sleep at every WFI instruction.

FRASTM, 2020-08-25:
This is the position of the SLEEPONEXIT bit of the System Control Register (SCR) of the cortex
Clearing this bit will avoid going to sleep when returning to Thread mode
This is the LL_LPM_DisableSleepOnExit() call during _sys_pm_power_state_exit_post_ops

PR #28501 (2020-09-21): soc: arm: stm32l4: fix power state exit
When we wake-up of deep sleep power state, we want to disable it.
Otherwise, when the cpu will go next to idle mode during a
SYS_POWER_STATE_ACTIVE, it will go into deep sleep mode
instead of a sleep mode.

fixes: #26896
commit babc6b9e (2020-09-21): soc: arm: stm32[l4|wb]: fix power state exit

When we wake-up of deep sleep power state, we want to disable it.
Otherwise, when the cpu will go next to idle mode during a
SYS_POWER_STATE_ACTIVE, it will go into deep sleep mode
instead of a sleep mode.

fixes: #26896
closed 2020-09-21 by MaureenHelm