zephyrproject-rtos/zephyr #26900: Bluetooth: host: bt_conn_recv() assumes ACL data is >= 2 bytes

thomasstenersen, 2020-07-16:
**Describe the bug**
An new `BT_ACL_START` fragment in `bt_conn_recv()` of less than 2 bytes will have unintended behavior.
The particular line of code with the problem is https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/conn.c#L1208

A link layer may choose to fragment an L2CAP PDU in any way it sees fit -- including fragmenting the first four bytes (the L2CAP header).

This behavior is tested in the LL/DFL/SLA/BV-02-C and LL/DFL/MAS/BV-02-C Link Layer conformance tests https://www.bluetooth.com/specifications/qualification-test-requirements/.

Here's a relevant section from the Bluetooth Core Specification v5.2, Vol. 3, Part A, section 1.4, table 1.1:

> A part of a PDU, as resulting from a fragmentation operation. Fragments are used only in the delivery of data to and from the lower layer. They are not used for peer-to-peer transportation. A fragment may be a Start or Continuation Fragment with respect to the L2CAP PDU. A fragment does not contain any protocol information beyond the PDU; the distinction of start and continuation fragments is transported by lower layer protocol provisions. Note: Start Fragments always either begin with the first octet of the Basic L2CAP header of a PDU or they have a length of zero (see [Vol 2] Part B, Section 6.6.2).

**Expected behavior**
The host stores the fragment and checks the length field as soon as it has received enough bytes for it to be valid (i.e., 2 bytes).

**Impact**
This will in worst case cause memory corruption, hard faults, buffer overflow attacks (?), etc. As the `conn->rx_len` effectively will be set as a combination of a value from a previously allocated buffer and the first byte in the new PDU.

**Environment (please complete the following information):**
 - Commit SHA or Version used: Current master (36707cd)

Vudentz, 2020-07-21:
This is quite weird given that BlueZ does exactly the same and tracing back:

commit aae7fe22a875a84e328469e228cba033ebbf20cb
Author: Andrei Emeltchenko <andrei.emeltchenko@nokia.com>
Date:   Wed Sep 15 14:28:43 2010 +0300

    Bluetooth: check for l2cap header in start fragment
    
    BLUETOOTH SPECIFICATION Version 4.0 [Vol 3] page 36 mentioned
    "Note: Start Fragments always begin with the Basic L2CAP header
    of a PDU."
    
    Signed-off-by: Andrei Emeltchenko <andrei.emeltchenko@nokia.com>
    Acked-by: Marcel Holtmann <marcel@holtmann.org>
    Signed-off-by: Gustavo F. Padovan <padovan@profusion.mobi>

It appeas that text was updated to allow fragmenting the L2CAP headers but I wonder if that is valid given the old specification still requires the full header. (@holtmann we may need to update this on BlueZ as well if that L2CAP can be fragmented).

thomasstenersen, 2020-07-22:
Yes, the Zephyr host is far from the only one with this assumption. There is a relevant TSE with more background here: https://www.bluetooth.org/tse/errata_view.cfm?errata_id=10216

Vudentz, 2020-07-23:
So they made the change knowingly it would break hosts such as Linux, for something that adds no value to the host given that if there is no L2CAP header there is basically no packet to report, while we can make this work going forward I wonder if we want to go backward and fix every release where we drop this packets, or we just push it back and make every controller willing to work with zephyr to not fragment the heade, as it have worked so far.

thomasstenersen, 2020-07-24:
This was my initial reaction as well. It seems a bit arbitrary.

However, I _do_ acknowledge the use-case for fragmenting L2CAP PDUs arbitrarily, i.e., to 1 + 23 bytes of LL data. Let's say you have a dual-mode chip with both Bluetooth and another 2.4GHz protocol (e.g., Wi-Fi). You will need to arbitrate between the two protocols, and perhaps you only have a time slice for sending a 1-byte LL PDU...?

> [...] where we drop these packets [...]
 
It looks like we're not dropping the packets, but actually misbehave. The Host doesn't check if the length of the ACL data is `>=2` before pulling the length field from the buffer.

> [...] or we just push it back and make every controller willing to work with zephyr to not fragment the header

I don't think this is not a good idea for a couple of reasons:

1. The controller is forced to be L2CAP aware.
   - ... which is technically a layer violation.
   - It puts unnecessary intelligence in the controller.
2. The controller is required to have memory buffers larger than the maximum receive LL Data payload length (`connMaxRxOctets`).
   - When `connMaxRxOctets = 23`, the controller is required to have RX buffer headroom of at least 26 bytes of payload. This is because it can, in the worst case, receive two LL PDUs of 3 + 23 bytes and is not able to send any data to the Host before the second PDU has been received.
3. It is an easy fix in the Host. And it belongs in the Host.
   - If the start fragment is `<2` bytes, wait for a continuation fragment before processing any further.
4. I'm not sure if the Zephyr LL(s) does any introspection into the LL PDUs to ensure that they're `>=2` bytes before sending them up to the host.
   - Maybe @cvinayak can confirm whether or not that's the case.

> [...] as it have worked so far.
 
I wouldn't quite agree to that, there's just not a report of the Zephyr Host stack actually crashing because of this. It would require pretty in-depth knowledge to diagnose this. 

I've attached a screenshot below of this actually occuring in the wild. In the first packet, only one byte of the L2CAP header is transmitted, and in the second, the remaining 23 bytes are sent.
[image]

Vudentz, 2020-07-24:
> This was my initial reaction as well. It seems a bit arbitrary.
>
> However, I _do_ acknowledge the use-case for fragmenting L2CAP PDUs arbitrarily, i.e., to 1 + 23 bytes of LL data. Let's say you have [...]

It doesn't necessarily need to parse any data though, it is just a minimal of 2 bytes, on a OS like Zephyr where we have a combined build I guess it makes no diference where the buffering happens, in fact it may take more memory to buffer this on the host side.

> 2. The controller is required to have memory buffers larger than the maximum receive LL Data payload length (`connMaxRxOctets`).
>
>    * When `connMaxRxOctets = 23`, the controller is required to [...]

Well we code that checks the rx_len:

```
		if (!conn->rx_len) {
			BT_ERR("Unexpected L2CAP continuation");
			bt_conn_reset_rx_state(conn);
			net_buf_unref(buf);
			return;
		}

		if (buf->len > conn->rx_len) {
			BT_ERR("L2CAP data overflow");
			bt_conn_reset_rx_state(conn);
			net_buf_unref(buf);
			return;
		}

```
I guess that means we would not be able to do these checks anymore, or need some code before to handle rx_len not being set ready the following by and then proceed to do these checks, anyway if the solution is that simple we are probably wasting our time discussing here.

> 4. I'm not sure if the Zephyr LL(s) does any introspection into the LL PDUs to ensure that they're `>=2` bytes before sending them up to the host.
>
>    * Maybe @cvinayak can confirm whether or not [...]

Yep, I confused with Linux that does drop the packet were we just assume it is 2 bytes longs which needs fixing.
> I've attached a screenshot below of this actually occuring in the wild. In the first packet, only one byte of the L2CAP header is transmitted, and in the second, the remaining 23 bytes are sent.
> [image]

On a split build where avoiding waking up the host may save some power may not be the best practice to split the packet in 2, it is also wasting throughput with 1 bytes + headers, on a combined build where controller and host are on running on same cpu I guess it makes very little difference.

thomasstenersen, 2020-07-27:
> It doesn't necessarily need to parse any data though, it is just a minimal of 2 bytes, on a OS like Zephyr where we have a combined build I guess it makes no diference where the buffering happens, in fact it may take more memory to buffer this on the host side.

> On a split build where avoiding waking up the host may save some power may not be the best practice to split the packet in 2, it is also wasting throughput with 1 bytes + headers, on a combined build where controller and host are on running on same cpu I guess it makes very little difference.

This may very well be, but in any case, if a controller decides to send a 1-byte `BT_ACL_START` message, the host must handle it. The Zephyr host should work with any (spec. conforming) controller.

---

> Well we code that checks the rx_len:
> [...]
> I guess that means we would not be able to do these checks anymore, or need some code before to handle rx_len not being set read the following byte and then proceed to do these checks, anyway if the solution is that simple we are probably wasting our time discussing here.

I think we're wasting our time :slightly_smiling_face: 
I suppose a quick patch would look something like this:
```diff
diff --git a/subsys/bluetooth/host/conn.c b/subsys/bluetooth/host/conn.c
index 97185a87f1..ad68815d76 100644
--- a/subsys/bluetooth/host/conn.c
+++ b/subsys/bluetooth/host/conn.c
@@ -1204,14 +1204,30 @@ void bt_conn_recv(struct bt_conn *conn, struct net_buf *buf, uint8_t flags)
 	/* Check packet boundary flags */
 	switch (flags) {
 	case BT_ACL_START:
+		if (conn->rx) {
+			BT_ERR("Unexpected first L2CAP frame");
+			bt_conn_reset_rx_state(conn);
+		}
+
+		if (buf->len < sizeof(len)) {
+			/* Got ACL data of size 1. Don't parse the length field
+			 * until the next BT_ACL_CONT when it might be valid
+			 * (=>2 bytes).
+			 */
+			conn->rx = buf;
+			return;
+		}
+
 		hdr = (void *)buf->data;
 		len = sys_le16_to_cpu(hdr->len);
 
 		BT_DBG("First, len %u final %u", buf->len, len);
 
-		if (conn->rx_len) {
-			BT_ERR("Unexpected first L2CAP frame");
+		if (buf->len > (sizeof(*hdr) + len)) {
+			BT_ERR("L2CAP overflow");
 			bt_conn_reset_rx_state(conn);
+			net_buf_unref(buf);
+			return;
 		}
 
 		conn->rx_len = (sizeof(*hdr) + len) - buf->len;
@@ -1223,13 +1239,27 @@ void bt_conn_recv(struct bt_conn *conn, struct net_buf *buf, uint8_t flags)
 
 		break;
 	case BT_ACL_CONT:
-		if (!conn->rx_len) {
+		if (!conn->rx) {
 			BT_ERR("Unexpected L2CAP continuation");
 			bt_conn_reset_rx_state(conn);
 			net_buf_unref(buf);
 			return;
 		}
 
+		/* Have we received the full length field in the L2CAP header
+		 * yet?
+		 */
+		if (conn->rx->len < sizeof(uint16_t)) {
+			uint16_t len = (conn->rx->data[0] | buf->data[0] << 8);
+
+			conn->rx_len = sizeof(struct bt_l2cap_hdr) + len;
+
+			/* Subtract 1 for the first 1-byte segment received in
+			 * BT_ACL_START.
+			 */
+			conn->rx_len -= 1;
+		}
+
 		if (buf->len > conn->rx_len) {
 			BT_ERR("L2CAP data overflow");
 			bt_conn_reset_rx_state(conn);

 }
```

PR #27169 (2020-09-18): Bluetooth: host: Support fragmented L2CAP header
Fixes: #26900.

A controller may fragment an L2CAP SDUs in any way it sees fit,
including fragmenting the L2CAP header. Likewise, the receiving
controller may send the fragmented header as ACL data to the host.

The Zephyr host assumed that a `BT_ACL_START` was at least 2 bytes long,
and consequently read the two-byte length field from the buffer without
length checks.

This commit allows the `BT_ACL_START` to be less than two bytes,
updating the `conn->rx_len` onces the `BT_ACL_CONT` with the remaining
part of the length field has been received [... cut]
closed 2020-09-18 by jhedberg