zephyrproject-rtos/zephyr #26912: arm: cortex_r: config_userspace: V2.3 hang during early power-up

pderwin, 2020-07-16:
**Describe the bug**

In v2.3 when building with CONFIG_USERSPACE enabled, I see that we do not do a context switch a short time after power-up.  I've chased this back and found the the cpus[0].nested has incremented to 6 in my example, so the code in **z_arm_int_exit()** thinks we are in a nested interrupt, and does not switch to another task.

The flaw is because of our syscall path.  On each syscall, we increment the nested count upon entry to the SVC (z_arm_svc in swap_helper.S), but then we go through z_arm_do_syscall (userspace.S) and return from the syscall without ever decrementing nested. 

The Cortex M code appears to work the same way to me.

**To Reproduce**
Any unit test for Cortex-R with CONFIG_USERSPACE enabled, should show that context switching fails after we do a system call or two.  

In my example, I have a user-mode thread the created a couple of message queues, and also interrupt driver shell was enabled.  After my thread completed its queue creation, I found I could not interact with the shell.  The uart interrupts were taken, but the process context  switch would not occur.  We stayed in the idle task.

**Expected behavior**
The shell should have been useable.

**Impact**
The system is not useable.  High implact

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gcc
 - Commit SHA:  b8c78e254 from 6/3/20 by Carlos Cufi

**Additional context**

I've already fixed a couple of other errors for R4 in V2.3 , but those are not upstreamed yet.  They are easy patches that I can supply if needed.

pderwin, 2020-07-20:
I did get a functional build by applying these 3 patches.  Patches 1 & 2 I think are solid.  The 3rd one, I doubt is what we need to end up with, but it gets me running for the time being.

[0003-arch-arm-userspace-decrement-nested-before-calling-s.patch.gz](https://github.com/zephyrproject-rtos/zephyr/files/4947559/0003-arch-arm-userspace-decrement-nested-before-calling-s.patch.gz)
[0002-arch-arm-userstate-reload-r2-and-r3.patch.gz](https://github.com/zephyrproject-rtos/zephyr/files/4947560/0002-arch-arm-userstate-reload-r2-and-r3.patch.gz)
[0001-arch-arm-userspace-use-rfeia-to-return-from-syscall.patch.gz](https://github.com/zephyrproject-rtos/zephyr/files/4947561/0001-arch-arm-userspace-use-rfeia-to-return-from-syscall.patch.gz)

carlescufi, 2020-07-20:
@pderwin assuming the bug exists in master, could you send a Pull Request against it and then backport it to the `v2.3-branch`?

stephanosio, 2020-07-23:
The upstream Cortex-R arch port does not support userspace.

As @carlescufi already pointed out, I believe the first step should be to submit a feature pull request with userspace support for the AArch32 Cortex-A/-R arch port.

pderwin, 2020-07-23:
We have locally maintained userspace for some time.  We'll get it upstreamed

ioannisg, 2020-07-24:
@pderwin could you please confirm that you do not see this issue in the _upstream_ project?

> Any unit test for Cortex-R with CONFIG_USERSPACE enabled, should show that context switching fails after we do a system call or two.

CONFIG_USERSPACE cannot be selected upstream for Cortex-R CPUs, because it depends on ARCH_HAS_USERSPACE, and this option is not selected for CPU_CORTEX_R.

I'd like to remove the bug label then, and you could choose to re-scope this ticket,

pderwin, 2020-07-24:
I agree that that is appropriate.

Thanks,

Phil

nashif, 2020-10-01:
this is already covered here https://github.com/zephyrproject-rtos/zephyr/issues/20594, no need for this ticket IMO

closed 2020-10-01 by nashif