zephyrproject-rtos/zephyr #26914: gen_kobject_list.py didn't generate correcet gperf info fro arc mwdt toolchain

vonhust, 2020-07-17:
**Describe the bug**
We are working on adding the support of ARC MWDT toolchain, which be found at
#22668 
and https://github.com/foss-for-synopsys-dwc-arc-processors/zephyr/tree/topic-mwdt-new.

We found  a new corner case that gen_kobject_list.py didn't gengerate corrert gperf info for arc mwdt toolchain.
gpfer info is necessary for Zephyr's USERSPACE feature.

arc mwdt toolchain only supports DWARF 4
ARC gcc in Zephyer SDK will generate DWARF 2 info.

a case is the tests/kernel/pipe/pipe. Here we select target nsim_sem, and enable verbose in gen_kobject_list.py. we can get the
following info.

* For ARC GCC

gen_kobject_list.py: found 18 kernel object instances total

* For ARC MWDT

gen_kobject_list.py: found 15 kernel object instances total

Obviously, there are some info lost for ARC MWDT, e.g., the info of stack_1.

* For ARC GCC, we can see

gen_kobject_list.py: symbol 'test_pipe' at 0x80003110 contains 1 object(s)
**gen_kobject_list.py: symbol 'stack_1' at 0x80000700 contains 1 object(s)**
gen_kobject_list.py: symbol 'get_sem' at 0x8000313c contains 1 object(s)

* For ARC MWDT, we didn't see the info ot stack_1

gen_kobject_list.py: symbol 'test_pipe' at 0x80003130 contains 1 object(s)
gen_kobject_list.py: symbol 'get_sem' at 0x8000315c contains 1 object(s)

But we found info of stack_1 in the map file.

What's more, for found kobjects,  the info in gerf  for ARC MWDT is not correct, e.g.  wrong stack size info about thread stack.

We have some questions:

* Does the pyelftools used in gen_kobject_list.py support DWARF 4
* Does gen_kobject_list.py support the parse of DWARF 4

* about Fix
It's better to make a debug on the gen_kobjet_list.py  with the elf file generated by ARC MWDT.

Considering ARC MWDT is a commerical toolchain, requires licenses.  We can provide a elf file and map file.

**To Reproduce**
Steps to reproduce the behavior:
1. use https://github.com/foss-for-synopsys-dwc-arc-processors/zephyr/tree/topic-mwdt-new
2. cd test/kernel/pipe
3. export ZEPHYR_TOOLCHAIN_VARIANT=ardmwdt
4. mkdir build; cd build
5. cmake -DBOARD=nsim_sem
6. make
7. see the gperf related files.

andrewboie, 2020-07-17:
> Does the pyelftools used in gen_kobject_list.py support DWARF 4

It's an open source project, found here: https://github.com/eliben/pyelftools
If we need to upgrade the version we are using, that isn't hard to do.

> Does gen_kobject_list.py support the parse of DWARF 4

Unknown if this has ever been tested. What does GCC 9.x produce?

> Considering ARC MWDT is a commerical toolchain, requires licenses. We can provide a elf file and map file.
> about Fix
> It's better to make a debug on the gen_kobjet_list.py with the elf file generated by ARC MWDT.

Given that this is against a proprietary compiler you guys are probably going to have to take the lead on debugging this.

MaureenHelm, 2020-07-21:
release meeting 21 jul: Changing to enhancement since we don't yet support this toolchain

andrewboie, 2020-08-10:
It appears that according to the release notes pyelftools 0.26 supports DWARF v4 or at least a significant subset of it.

I have no idea what changes required to gen_kobject_list.py are needed (if any) to support it.

I'm removing myself from the assignee list for now.

ruuddw, 2020-08-14:
PR #27008 fixes this. Note that testing can be done with gcc, when compiling with -gdwarf-version 4.

PR #27008 (2020-08-14): scripts: add the support of DW_AT_count for array type
In DWARF 4, e.g. ARC Metaware toolchain, DW_AT_count is
used not DW_AT_upper_bound. We should consider this corner
case.

Fixes #26914
closed 2020-08-14 by carlescufi