zephyrproject-rtos/zephyr #26923: [RFC] API change - Normalize callback signature include the caller's device pointer.

tbursztyka, 2020-07-17:
## Introduction

Adding callback caller's device pointer to DMA, IPM and UART callbacks.

### Problem description

DMA, IPM and UART callbacks do not provide the device's callback caller as parameter, forcing users to use the callback_arg parameter instead for this purpose, which in turn limits its usage (they cannot provide some private data, or have to mangle thing together storing the device pointer into the private data).

### Proposed change

Adding such device pointer to DMA, IPM and UART callbacks.

## Detailed RFC

Problem description section is clear enough
The issue was spotted while switching all struct device * to constant. But it is also a normalization procedure, as all other APIs always provide the device pointer to the callbacks.

### Proposed change (Detailed)

Add struct device *dev to DMA, IPM and UART callbacks (it's pretty trivial so it does not need more details)

### Dependencies

All users of these callbacks need to adapt their callbacks accordingly.
PRs will fix all zephyr internal code, but there are module dependencies as well (mbedtls and Segger)

### Concerns and Unresolved Questions

None.

## Alternatives

None.

**PRs fixing this:**
- [[RFC] API Change: DMA - callback improvement](#26781)
- [[RFC] API Change: IPM - callback improvement](#26780)
- [[RFC] API Change: UART - callback improvement](#26426)

tbursztyka, 2020-07-17:
@carlescufi @nashif @pabigot @galak

carlescufi, 2020-07-21:
API meeting 21st July:

- Why is UART being added a context/user data pointer but not others that are also missing it (eg. gpio)
- Can we take this opportunity to settle on a name for data that is carried transparently by the driver without ever accessing it?
  - `context` +
  - `user_data` +++++ (selected)
  - `callback_args`

carlescufi, 2020-07-28:
> API meeting 21st July:
>
> * Why is UART being added a context/user data pointer but not others that are also missing it (eg. gpio)
> * Can we take this opportunity to settle on a name for data that is [...]

API meeting 28th July

- @tbursztyka will update the PRs to unify the user param to be named `user_data`

commit 04d6d0b1 (2020-07-30): drivers: serial: Fix uart_irq_callback_user_data_t usage

Now providing the device pointer that calls the callback.

Fixes #26923
commit 4644d4a4 (2020-07-30): uart: Normalize user callback signature

All other API provide the device pointer to the callback it's going to
call. So fixing this in UART.

This will avoid passing the actual device pointer as a user data (thus
enabling user data for something else).

Fixes #26923
commit 4c78d0e3 (2020-07-30): tests: uart: Fix uart_callback_set usage

The device pointer is now provided to the callback, there is no need to
call uart_callback_set() with the device being set a user data.

Fixes #26923
commit 701869fc (2020-07-30): uart: Fix uart_irq_callback_user_data_set usage

Now providing the struct device * to the callback.

Fixes #26923
closed 2020-07-30 by carlescufi