zephyrproject-rtos/zephyr #26924: no space to store ccc cfg

johnsmithecon, 2020-07-17:
**Describe the bug**
I will occasionally see the "no space to store ccc cfg" issue. The output to the console looks like this: 
[image]

When the "no space to store ccc cfg" issue does show up, the node becomes useless. It cannot open bearer anymore. Once I turn the node off then back on, however, the node acts normally again. 

**To Reproduce**
Quite difficult to reproduce but if I have 2 nodes and I play with the nrf provisioning app and unprovision and provision the nodes randomly and wait 10 minutes then repeat the process, I notice that the issue will show itself eventually.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
To me this is a show stopper because it results in the node becoming useless and I think it messes with the entire mesh network.

**Logs and console output**
My prj.conf file:
J-Link Log Viewer
#CONFIG_UART_CONSOLE=n
#CONFIG_HAS_SEGGER_RTT=y
#CONFIG_USE_SEGGER_RTT=y
#CONFIG_RTT_CONSOLE=y
#CONFIG_LOG_BACKEND_RTT=y
#CONFIG_LOG_STRDUP_BUF_COUNT=20
#CONFIG_LOG_STRDUP_MAX_STRING=80

#CONFIG_MAIN_STACK_SIZE=512
#CONFIG_BT_RX_STACK_SIZE=2048
#CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

#CONFIG_BT=y
#CONFIG_BT_OBSERVER=y
#CONFIG_BT_BROADCASTER=y
#CONFIG_BT_DEBUG_LOG=n
#CONFIG_BT_MESH_GATT_PROXY=y
#CONFIG_BT_PERIPHERAL=y
#CONFIG_BT_MESH_PB_GATT=y
#CONFIG_BT_MESH_PB_ADV=y
#CONFIG_BT_DEVICE_NAME="LIGHT"

#CONFIG_BT_MESH=y
#CONFIG_BT_MESH_APP_KEY_COUNT=1

#CONFIG_BT_SETTINGS=y
#CONFIG_FLASH=y
#CONFIG_FLASH_PAGE_LAYOUT=y
#CONFIG_FLASH_MAP=y
#CONFIG_FCB=y
#CONFIG_SETTINGS=y
#CONFIG_SETTINGS_FCB=y

#CONFIG_BT_DEBUG_LOG=y
#CONFIG_BT_MESH_DEBUG=y
#CONFIG_BT_MESH_DEBUG_ACCESS=y
#CONFIG_BT_MESH_DEBUG_MODEL=y
#CONFIG_BT_MESH_DEBUG_TRANS=y
#CONFIG_BT_MESH_DEBUG_NET=y
#CONFIG_BT_MESH_DEBUG_CRYPTO=n

PWM START
#CONFIG_PRINTK=y
CONFIG_PWM=y
#CONFIG_PWM_LOG_LEVEL_DBG=y

PWM END
#CONFIG_SYS_CLOCK_TICKS_PER_SEC=1000

#BT_CTLR=BT_CTLR_TX_PWR_PLUS_8

from prj.conf of mesh
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

CONFIG_BT_SETTINGS=y
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_NVS=y
CONFIG_SETTINGS=y

CONFIG_BT_OBSERVER=y
CONFIG_BT_CTLR_DUP_FILTER_LEN=0
CONFIG_BT_CTLR_LE_ENC=n
CONFIG_BT_DATA_LEN_UPDATE=n
CONFIG_BT_PHY_UPDATE=n
CONFIG_BT_CTLR_CHAN_SEL_2=n
CONFIG_BT_CTLR_MIN_USED_CHAN=n
CONFIG_BT_CTLR_PRIVACY=n

CONFIG_BT_PERIPHERAL=y

CONFIG_BT=y
CONFIG_BT_TINYCRYPT_ECC=y
CONFIG_BT_L2CAP_RX_MTU=69
CONFIG_BT_L2CAP_TX_MTU=69
CONFIG_BT_L2CAP_TX_BUF_COUNT=5

CONFIG_BT_MESH=y
CONFIG_BT_MESH_MODEL_EXTENSIONS=y
CONFIG_BT_MESH_RELAY=y
CONFIG_BT_MESH_FRIEND=y
CONFIG_BT_MESH_PB_GATT=y
CONFIG_BT_MESH_PB_ADV=y
CONFIG_BT_MESH_GATT_PROXY=y

#CONFIG_BT_MESH_LOW_POWER=y
#CONFIG_BT_MESH_LPN_SCAN_LATENCY=30
#CONFIG_BT_MESH_LPN_RECV_DELAY=40
#CONFIG_BT_MESH_LPN_POLL_TIMEOUT=300

CONFIG_BT_MESH_SUBNET_COUNT=2
CONFIG_BT_MESH_APP_KEY_COUNT=2
CONFIG_BT_MESH_MODEL_GROUP_COUNT=2
CONFIG_BT_MESH_LABEL_COUNT=3

CONFIG_BT_DEBUG_LOG=y

CONFIG_BT_MESH_DEBUG=y
#CONFIG_BT_MESH_DEBUG_PROV=y
CONFIG_BT_MESH_DEBUG_PROXY=y
#CONFIG_BT_MESH_DEBUG_BEACON=y
#CONFIG_BT_MESH_DEBUG_NET=y
#CONFIG_BT_MESH_DEBUG_TRANS=y
#CONFIG_BT_MESH_DEBUG_SETTINGS=y
#CONFIG_BT_MESH_DEBUG_LOW_POWER=y
#CONFIG_BT_MESH_DEBUG_FRIEND=y
#CONFIG_BT_MESH_DEBUG_MODEL=y
#CONFIG_BT_MESH_DEBUG_ACCESS=y
#CONFIG_BT_MESH_DEBUG_CRYPTO=y
#CONFIG_BT_MESH_DEBUG_ADV=y

UART STUFF FOR SENSOR
CONFIG_UART_CONSOLE=n
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
CONFIG_LOG_BACKEND_SWO=n
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y

CONFIG_RING_BUFFER=y

#CONFIG_BT_MAX_PAIRED=1

**Environment (please complete the following information):**
MacOS

**Additional context**
Some stuff I've noticed includes:
 I reset node 1 and try to provision the sensor node and it gets stuck at trying to get composition data and my logs for node 2 tells me no space to store  ccc cfg. And literally node 2 is broken now so I need to turn off and back on and now I can get composition data. And now I can’t reset node 2 with my app. But if I reset the node 1 first then try to reset the node 2, I am able to reset.

johnsmithecon, 2020-07-27:
I've been able to recreate the issue by flashing samples/bluetooth/mesh project onto nrf52832 chip (west build nrf52dk_nrf52832). I uncommented BT_MESH_DEBUG_PROV=y and BT_MESH_DEBUG_PROXY=y from prj.conf.

I then go on the nRF mesh app on iOS and do the following:

1. Provision then reset then provision a few times. (Not sure if this step is necessary)
2. Keep in provisioned state for 15 minutes.
3. Reset the node then try to provision again. You should now see the “no space to store ccc cfg” message printed to the console and your node will now become unusable.

johnsmithecon, 2020-07-27:
It's also printing "bt_mesh_proxy.gatt_proxy_advertise: Connectable advertising deferred (max connections)

Maybe the will mean something to your team

joerchan, 2020-07-28:
@johnsmithecon Thank you for the updated way to reproduce this scenario.
The 15 minutes was a good indicator that the privacy configuration of the phone was part of the problem. I was able to reproduce this faster by power-cycling my phone in order to trigger a new RPA.

The problem is that the GATT provisioner service is unregistered while the peer is still disconnected.
Then the CCC configuration of the peer will not be cleared on disconnect anymore.
After resetting the node the service will be added, and the CCC will be "restored", even though the peer is disconnected.
Once a connection is made with a new RPA then there are no space to store a new CCC cfg.

PR #27206 (2020-08-10): Bluetooth: GATT: Unregister CCC when unregistering services
Fix issue with bt_gatt_service_unregister not clearing CCC information.

If the service is unregistered while an unbonded peer is connected and
subscribed then this CCC would be restored again when registering the
service even if the peer is now disconnected.

If the service is unregistered with bonded peers CCCs stored in settings
then this value would never be cleared. If the service is registered
again then the peer would receive a service changed for this service,
but it's CCC value would still be enabled.

Fixes: #26924
commit 35201d86 (2020-08-10): Bluetooth: GATT: Unregister CCC when unregistering services

Fix issue with bt_gatt_service_unregister not clearing CCC information.

If the service is unregistered while an unbonded peer is connected and
subscribed then this CCC would be restored again when registering the
service even if the peer is now disconnected.

If the service is unregistered with bonded peers CCCs stored in settings
then this value would never be cleared. If the service is registered
again then the peer would receive a service changed for this service,
but it's CCC value would still be enabled.

Fixes: #26924
closed 2020-08-10 by carlescufi