zephyrproject-rtos/zephyr #26935: [topic-usb] USB stack rework. ToDo list.

emob-nordic, 2020-07-17:
## Introduction

This issue is created to list all things that should be reworked in [topic-usb](https://github.com/zephyrproject-rtos/zephyr/tree/topic-usb) before its merged into master.

### Problem description

USB device stack requires rework to address feature requests that we currently have. Some of them:

- #20236
- #19713
- #26857 
- #24200
- #25604

### ToDo list:

- [ ] Extend the possibility to have more than one alternate setting for interface. (follow up of #26733)
        This point requires updates to `struct usb_if_descriptor`. @emob-nordic 
- [ ] Make all drivers use bitfields introduced in #26854. Remove direction, type and recipient [getter macros](https://github.com/zephyrproject-rtos/zephyr/blob/a2d4292a96e0e3f710374b6da6d3d5060bccc905/include/usb/usbstruct.h#L39-L41). @jfischer-phytec-iot

koffes, 2020-10-14:
Geetings! Has a timeline been set for this task?

emob-nordic, 2020-10-14:
@koffes The work has started but We cannot say the exact timing.

koffes, 2021-04-26:
What is the status of this issue?

jfischer-no, 2021-04-26:
> What is the status of this issue?

Is almost ready.

jfischer-no, 2022-01-22:
Close in favor of #42066

closed 2022-01-22 by jfischer-no