zephyrproject-rtos/zephyr #26937: Kconfig choice failing

solsbarry, 2020-07-17:
**Describe the bug**
This is an odd problem. I am getting a zillion kconfig warnings all for the same issue.
Like this but wiht various things for <symbol> and <location>

warning: the choice symbol <symbol> (defined at <location>:11, <location>:11) is defined with a prompt outside the choice

And everywhere it is happening there is a very specific circumstance.
Anytime a kconfig file in the nrfconnect/sdk-nrf module (included in the project via boilerplate looking at my west configuration)
uses "source" to include another kconfig file, and that file has a "choice" that that does not have a name.

however if I give that choice a name, then the warning goes away.

For example I get warning
warning: the choice symbol AT_HOST_UART_0 (defined at C:/projects/g7/nrf\lib\at_host/Kconfig:25, C:/projects/g7/zephyr/../nrf\lib\at_host/Kconfig:25) is defined with a prompt outside the choice

for the following choice:
choice
	prompt "UART for AT Host"
	default AT_HOST_UART_0
	depends on AT_HOST_LIBRARY
	help
		Sets the UART to use for the AT Host
		-  UART 0
		-  UART 1
		-  UART 2
	config AT_HOST_UART_0
		bool "UART 0"
	config AT_HOST_UART_1
		bool "UART 1"
	config AT_HOST_UART_2
		bool "UART 2"
endchoice

However if I change the first line to:
choice UART_FOR_HOST

Then I don't get the warning.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

There are obviously other "choice"s in the kconfig files that don't have names, some do some don't. Those don't have this issue. The issue only happens when kconfig files with an unnamed choice are "source"d, such as 
 
module = ASSET_TRACKER
module-str = Asset Tracker
source "${ZEPHYR_BASE}/subsys/logging/Kconfig.template.log_config"

But the distinction is that sourcing this same file from other modules does not cause an issue.
zephyr for example sources Kconfig.template.log_config many times without issue.
But somehow when nrf sources it I have this issue with the "choice" in it.

This makes no sense to me.

**Expected behavior**
I would expect kconfig to complete normally

**Impact**
Right now I can't complete cmake. I could add a name to all of the places this is happening, but that complicates things obviously, because its in zephyr code and nrf code that I was hoping not to have to fork.  And there is no reason for it.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: GNU Arm
 - Commit SHA or Version used:

I have my open project and I use west to pull in other repos, below are the versions of things I am using right now

name: ncs
      url-base: https://github.com/nrfconnect
    - name: zephyrproject
      url-base: https://github.com/zephyrproject-rtos
    - name: throwtheswitch
      url-base: https://github.com/ThrowTheSwitch
    - name: armmbed
      url-base: https://github.com/ARMmbed
    - name: civetweb
      url-base: https://github.com/civetweb
    - name: MCUboot
      url-base: https://github.com/JuulLabs-OSS

   - name: zephyr
      repo-path: sdk-zephyr
      revision: v2.3.0-rc1-ncs1
      remote: ncs
      import:       
        name-whitelist:
          - canopennode
          - ci-tools
          - cmsis
          - civetweb
          - edtt
          - fatfs
          - hal_nordic
          - hal_st
          - libmetal
          - littlefs
          - loramac-node
          - lvgl
          - mbedtls
          - mipi-sys-t
          - net-tools
          - nrf_hw_models
          - open-amp
          - openthread
          - segger
          - tinycbor
          - tinycrypt

    #NCS repositories 
    - name: nrf
      repo-path: sdk-nrf
      path: nrf
      revision: v1.3.0
      remote: ncs   
    - name: mcuboot
      repo-path: mcuboot
      revision: v1.6.0
      path: mcuboot
      remote: MCUboot
    - name: mcumgr
      repo-path: sdk-mcumgr
      revision: v0.0.1-ncs2
      path: modules/lib/mcumgr
      remote: ncs
    - name: nrfxlib
      path: nrfxlib
      revision: v1.3.0 
      repo-path: sdk-nrfxlib
      remote: ncs
    - name: cmock
      path: test/cmock
      revision: c243b9a7a7b3c471023193992b46cf1bd1910450
      remote: throwtheswitch
    - name: unity
      path: test/cmock/vendor/unity
      revision: 031f3bbe45f8adf504ca3d13e6f093869920b091
      remote: throwtheswitch
    - name: mbedtls-nrf
      path: mbedtls
      repo-path: mbedtls
      revision: mbedtls-2.16.6
      remote: armmbed

**Additional context**
I know this involves some nordic repos, and the fault may be theirs, but really it seems more generic. It seems like an issue with how kconfig fields are being parsed.

solsbarry, 2020-07-17:
Well even my proposed horrible work around (adding a name to ech choice) doesn't work. It allows cmake to complete without warning, but my log_levels do not get generated and who knows what else.  

So for example, I added a name (LOG_LEVEL_CHOICE) to the choice in Kconfig.template.log_config

```
choice LOG_LEVEL_CHOICE
	prompt "Max compiled-in log level for $(module-str)"
	default $(module)_LOG_LEVEL_INF
	depends on LOG

config $(module)_LOG_LEVEL_OFF
	bool "Off"

config $(module)_LOG_LEVEL_ERR
	bool "Error"

config $(module)_LOG_LEVEL_WRN
	bool "Warning"

config $(module)_LOG_LEVEL_INF
	bool "Info"

config $(module)_LOG_LEVEL_DBG
	bool "Debug"

endchoice
```
And now all of the generated log levels are not in autoconf.h

The odd thing is the kconfig choices exist, I can see them in .config, and I can select them in my prj.conf file, but by default no choice is being chosen.

So now my extended work around is to config all the things in my prj file which are not getting assigned their default value.

solsbarry, 2020-07-20:
I realized our project no longer needed to include the sdk-nrf module. So removing that from our west file removed it from the project, and got rid of this issue.  So I am happy to close this if you want me to. I am not sure if my original issue exposed some underlying issue that you all might be interested in, or maybe this is just some weird edge case that you all don't care about. I'd hate for anyone to waste too much time on this.

tejlmand, 2020-09-08:
> I realized our project no longer needed to include the sdk-nrf module. So removing that from our west file removed it from the project, and got rid of this issue. 

Sorry for late response. Seems summer time caused this to drop to far down the list of notification.
Before closing, I was wondering, is your project a fork of `sdk-nrf`, so that it includes the same Kconfig files as the pure `sdk-nrf` repo ?
I have not been able to reproduce this locally.

solsbarry, 2020-09-08:
No our project is its own repository started from scratch. I didn't show our repo in the west file I put above, so I can see that would be confusing. But our repo is the main folder, containing the west file, and it pulls in all the other repos. Our west file is modeled after the NRF one, but thats the only similarity. Our Kconfig files only have new options that we created specific to our project, no repeated options.
The problem also could have stemmed from the free nordic edition of Segger Embedded Studio honestly. We use that and it have found several instances of it giving build errors where the command line doesn't, so maybe the same is true for warnings.

tejlmand, 2020-09-09:
> The problem also could have stemmed from the free nordic edition of Segger Embedded Studio honestly. 

In fact, the SES-Nordic Edition calls CMake in the background, so you should see same error messages in SES and command line.

But what I do notice, is this:
> warning: the choice symbol AT_HOST_UART_0 (defined at C:/projects/g7/nrf\lib\at_host/Kconfig:25, C:/projects/g7/zephyr/../nrf\lib\at_host/Kconfig:25) is defined with a prompt outside the choice

Those two:
```
Defined at
 C:/projects/g7/nrf\lib\at_host/Kconfig:25
 C:/projects/g7/zephyr/../nrf\lib\at_host/Kconfig:25
```
indicates that the SAME Kconfig file is sourced twice from different locations.
Note, that when you have:
```
- name: nrf
  repo-path: sdk-nrf
  path: nrf
  revision: v1.3.0
  remote: ncs   
```
in your manifest, then the `nrf/Kconfig.nrf` is automatically included by the build system, so if you ALSO do something similar to:
`source "<path>/nrf/Kconfig.nrf"` inside your own project, then you could endup with the same Kconfig files sourced twice, and leading to undefined behavior.

Judged by this: `C:/projects/g7/zephyr/../nrf`, then I would look for a line similar to: `source "$(ZEPHYR_BASE)/../nrf/Kconfig.nrf"` in your project.

tejlmand, 2020-09-10:
Closing, as I believed this has been answered.
Feel free to reopen if more details can be provided.

closed 2020-09-10 by tejlmand