zephyrproject-rtos/zephyr #26938: gpio: api to query pin configuration

cfriedt, 2020-07-17:
**Is your enhancement proposal related to a problem? Please describe.**
In some cases, it's necessary to query the configuration (e.g. direction) of a particular GPIO pin. However, the [existing GPIO API](https://docs.zephyrproject.org/latest/reference/peripherals/gpio.html) does not seem to provide a way to do that.

**Describe the solution you'd like**
```cpp
__subsystem struct gpio_driver_api {
        // ...
        // deprecate this call. make it wrap gpio_set_config() instead
	int (*pin_configure)(struct device *port, gpio_pin_t pin, gpio_flags_t flags);
        // this call would be set up to simply call pin_configure
        int (*pin_set_config)(struct device *port, gpio_pin_t pin, gpio_flags_t flags);
        // this call would return the current pin configuration
        int (*pin_get_config)(struct device *port, gpio_pin_t pin, gpio_flags_t *flags);
        // ...
};
 
__syscall int gpio_set_config(struct device *port, gpio_pin_t pin,
			  gpio_flags_t flags);

__syscall int gpio_get_config(struct device *port, gpio_pin_t pin,
			  gpio_flags_t *flags);

// deprecate this call. make it wrap gpio_set_config() instead
__syscall int gpio_config(struct device *port, gpio_pin_t pin,
			  gpio_flags_t flags);
```

**Describe alternatives you've considered**
It's possible that gpio configuration registers may be write-only on some hardware, and perhaps that's the reason that this feature was not added originally. In those cases, I would expect that a simple bitmap is used internally instead of querying hardware.

**Additional context**

henrikbrixandersen, 2020-07-17:
Could you elaborate a bit more on the use case for this?

I mean, normally you'd either have fixed input or output, or have the direction controlled by a driver, which could keep track of when to change the direction internally.

When would you need to query the direction?

cfriedt, 2020-07-17:
> Could you elaborate a bit more on the use case for this?

Sure - I'll be using it to implement Greybus GPIO support.

> When would you need to query the direction?

Anytime there is a service that reports the current state of GPIO.

PR #40325 (2022-06-24): api: gpio: add gpio direction query functions and driver api
Some services require the ability to query whether a GPIO pin is configured as an input, output, both, or neither prior to performing any operations at the service level. This is done in order to reduce state tracking within the service.

To that end, this change adds
* `gpio_port_get_direction()`
* `gpio_pin_is_input()`, and
* `gpio_pin_is_output()`

Fixes #26938

* Added tests under `tests/drivers/gpio_get_direction` and implementations for a handful of drivers.
* Add myself and @henrikbrixandersen as collaborators for GPIO
* Remov [... cut]
closed 2022-06-24 by carlescufi