zephyrproject-rtos/zephyr #26939: MCUMGR - smp shell server sends responses to wrong port

lawrence-king, 2020-07-17:
**Describe the bug**

smp_shell always outputs responses to the console instead of back to the device that the shell is running on.

The shell can be run on any port, not just the console, and we can add smp_shell to any shell. When the shell is running on a port which is not the console it is correctly passing the received data to smp_shell, its just that smp_shell sends the reply out to the console, and not to the port the shell is running on.

Every dts that uses mcumgr, also has the console on the same port as the shell, there is no existing board that routes the mcumgr to a different place than the console. Hence you would never have seen this issue.

What have you tried to diagnose or workaround this issue?

I followed the flow of the data from mcumgr on the host computer, into the USB serial port, into the shell, and into smp_shell. All is working as expected. smp_shell then formulates a response to the mcumgr request and gets it ready to send,  but at the very last step. The function smp_shell_tx_raw() in subsys/mgmt/smp_shell.c is hard coded to always send the response via k_str_out(). k_str_out() is part of  lib/os/printk and calls __char_out(). The function __char_out() is hooked into printk with the function __printk_hook_install() very early in the kernel boot phases and all the places that the hook is installed always installs it to the console.

as a workaround I changed: 
`k_str_out((void *)data, len);`
to
`shell_print_stream(shell_backend_uart_get_ptr(), data, len);`
in the function smp_shell_tx_raw().

**To Reproduce**

First build the application and load it onto a nrf52840_mdk board. This works for Zephyr v2.3.0
```
#!/bin/bash
#
# defaults if none specified on command line
set -e 
KEY=mykey-rsa-2048.pem
VERSION=1.0
#BOARD=irdeto_gen1
BOARD=nrf52840_mdk
#parse the options
while getopts v:k:b: option
do
case "${option}"
in
k) KEY=${OPTARG};;
v) VERSION=${OPTARG};;
b) BOARD=${OPTARG};;
*) options -v VERSION, -k KEY, -b BOARD
esac
done
echo rebuild the project, for $BOARD, sign the image and flash the image. 
echo - using signing key $KEY
echo - adding version number $VERSION
echo - the board is -D`echo $BOARD | tr [a-z] [A-Z]` 

rm -rf build
west build -b $BOARD -- -DDTC_OVERLAY_FILE=`pwd`/$BOARD.overlay -DSET_BOARD=`echo $BOARD | tr [a-z] [A-Z]`
west sign -t imgtool -- --key ../../$KEY --version $VERSION
west flash --hex-file build/zephyr/zephyr.signed.hex
```
At this point the console will be running on the MDK board serial port. The hardware USB2UART on the MDK board will present the console on the host as /dev/ttyACM0. The Shell will be running on the USB CDC_ACM serial port. This will be visible on the host as /dev/ttyACM1. shell commands sent to /dev/ttyACM1 using a program such as tio work as expected.

Monitor /dev/ttyACM0 with a program like tio. There should be no output.

Next try to load a file into the littlefs using mcumgr

```
#!/bin/bash
#
# defaults if none specified on command line
ASSET=../assets/82841136.bin
#parse the options
while getopts a: option
do
case "${option}"
in
a) ASSET=${OPTARG};;
*) options -a ASSET
esac
done
echo flash the asset info into the littlefs over USB
echo - programming asset $ASSET
sudo ~/.go/bin/mcumgr conn add zephyr type=serial connstring="dev=/dev/ttyACM1" #,baud=115200"
sudo ~/.go/bin/mcumgr -c zephyr fs upload $ASSET /settings/Blob
```

mcumgr will timeout waiting for a response, however the response will be visible on the tio that is monitoring /dev/ttyACM0. This is the incorrect behaviour.

**Expected behavior**

mcumgr should upload the file to the littlefs.

**Impact**
I am having to rework my system to put the console and the shell on the same port.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Ubuntu Linux, running on virtualbox under Windows
 - Toolchain  - Zephyr SDK 0.11.4
 - Commit SHA or Version used - v2.3.0 release.

support files prj.config, .config, and an overlay

[files.gz](https://github.com/zephyrproject-rtos/zephyr/files/4939286/files.gz)

carlescufi, 2020-07-20:
@de-nordic could you try to reproduce please?

de-nordic, 2020-07-20:
> 
> 
> @de-nordic could you try to reproduce please?

I am taking care of this.

mniestroj, 2020-07-20:
I think that using something from shell API (e.g. `shell_print_stream()` as in the issue description) instead of `k_str_out()` might help when having logs printed over shell during firmware upgrade / file transfer. Right now there is a possibility that log will be printed in the middle of mcumgr smp response.

jakub-uC, 2020-08-06:
@mniestroj : Function `shell_print_stream()` is not part of shell API and it shall be used only internally by the shell. Using it outside of shell thread can lead to concurrent access to the transfer medium.

de-nordic, 2020-08-28:
@lawrence-king Can you check whether https://github.com/zephyrproject-rtos/zephyr/pull/27871 fixes the problem for you?

lawrence-king, 2020-08-28:
I have been temporarily moved to another (non-Zephyr) project, so it will take me a while to get a chance to test it. Reading the code it looks like a correct solution.

Lawrence King
Principal Developer
+1(416)627-7302

ibabatov, 2020-08-30:
I am facing the same problem as the original poster.
I have manually applied the fixes from #27871 and my setup seems to work OK now. Have not done any extensive testing though.
Thank you very much!

PR #27871 (2020-09-01): mgmt: smp: Fix shell sending smp response to wrong UART
The commit fixes problem where the SMP over shell response was sent
via UART attached to the console rather than the UART attached to
the shell.

Fixes #26939
commit 88deb807 (2020-09-01): mgmt: smp: Fix shell sending SMP response to wrong UART

The commit fixes the problem where the SMP over shell responses have
been sent with use of k_str_out(), which is configured to use console
UART directly.  This caused problem when user decided to attach
shell to a different UART then the one that has been used by
the console driver, because the SMP responses would still be sent
to the console UART rather than the shell UART.

The updated code uses, for shell SMP responses, the UART that has been
selected for the shell.

Fixes #26939
closed 2020-09-01 by carlescufi