zephyrproject-rtos/zephyr #26949: sanitycheck gets overwhelmed by console output

andrewboie, 2020-07-17:
**Describe the bug**
If tests have too much console output, sanitycheck gets bogged down and can't handle it all. QEMU processes spin waiting to flush their output pipes; eventually dozens of tests time out, even though they pass when built manually.

I found this in a branch I am working on which dumps out page table information to the console with ANSI color codes. With every test emitting this, sanitycheck gets overwhelmed.

**To Reproduce**
Clone:
https://github.com/andrewboie/zephyr/tree/sanitycheck-having-a-stroke

Run sanitycheck -T tests/kernel -p qemu_x86 at a high -j level.

I think this is fairly minor as tests normally don't emit this much console data, but still the amount of data is not large and perhaps the console processing code in sanitycheck could be optimized, a modern CPU shouldn't get overwhelmed by kilobytes of data.

nashif, 2020-07-18:
```gale:zephyr(sanitycheck-having-a-stroke):  sanitycheck -T tests/kernel -p qemu_x86 -W
INFO    - JOBS: 112
INFO    - Building initial testcase list...
INFO    - 90 test configurations selected, 10 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:   90/  90  100%  skipped:   11, failed:    0
INFO    - 79 of 79 tests passed (100.00%), 0 failed, 11 skipped with 0 warnings in 93.61 seconds
INFO    - In total 100 test cases were executed on 1 out of total 267 platforms (0.37%)
INFO    - 77 tests executed on platforms, 13 tests were only built.
```

andrewboie, 2020-07-19:
@nashif this with the branch I linked?

This is what I see when I build the linked branch on my machine:

```
$ scr -W -T tests/kernel/ -p qemu_x86                                           
Deleting output directory /dev/shm/sanity-out
INFO    - JOBS: 48
INFO    - Building initial testcase list...
INFO    - 90 test configurations selected, 10 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:   38/  90  42%  skipped:   11, failed:    0
ERROR   - qemu_x86                  tests/kernel/mutex/mutex_api/kernel.mutex          FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/mutex/mutex_api/kernel.mutex/handler.log
INFO    - Total complete:   39/  90  43%  skipped:   11, failed:    1
ERROR   - qemu_x86                  tests/kernel/fpu_sharing/float_disable/kernel.fpu_sharing.float_disable.x86 FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/fpu_sharing/float_disable/kernel.fpu_sharing.float_disable.x86/handler.log
INFO    - Total complete:   40/  90  44%  skipped:   11, failed:    2
ERROR   - qemu_x86                  tests/kernel/common/kernel.common                  FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/common/kernel.common/handler.log
INFO    - Total complete:   41/  90  45%  skipped:   11, failed:    3
ERROR   - qemu_x86                  tests/kernel/timer/timer_api/kernel.timer.tickless FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/timer/timer_api/kernel.timer.tickless/handler.log
INFO    - Total complete:   43/  90  47%  skipped:   11, failed:    4
ERROR   - qemu_x86                  tests/kernel/queue/kernel.queue                    FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/queue/kernel.queue/handler.log
INFO    - Total complete:   44/  90  48%  skipped:   11, failed:    5
ERROR   - qemu_x86                  tests/kernel/threads/thread_apis/kernel.threads.apis FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/threads/thread_apis/kernel.threads.apis/handler.log
INFO    - Total complete:   45/  90  50%  skipped:   11, failed:    6
ERROR   - qemu_x86                  tests/kernel/workq/work_queue_api/kernel.workqueue.api FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/workq/work_queue_api/kernel.workqueue.api/handler.log
INFO    - Total complete:   46/  90  51%  skipped:   11, failed:    7
ERROR   - qemu_x86                  tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace/handler.log
INFO    - Total complete:   47/  90  52%  skipped:   11, failed:    8
ERROR   - qemu_x86                  tests/kernel/sched/metairq/kernel.scheduler.metairq FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/sched/metairq/kernel.scheduler.metairq/handler.log
INFO    - Total complete:   48/  90  53%  skipped:   11, failed:    9
ERROR   - qemu_x86                  tests/kernel/device/kernel.device.pm               FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/device/kernel.device.pm/handler.log
INFO    - Total complete:   49/  90  54%  skipped:   11, failed:   10
ERROR   - qemu_x86                  tests/kernel/queue/kernel.queue.poll               FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/queue/kernel.queue.poll/handler.log
INFO    - Total complete:   50/  90  55%  skipped:   11, failed:   11
ERROR   - qemu_x86                  tests/kernel/common/kernel.common.misra            FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/common/kernel.common.misra/handler.log
INFO    - Total complete:   51/  90  56%  skipped:   11, failed:   12
ERROR   - qemu_x86                  tests/kernel/pipe/pipe/kernel.pipe                 FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/pipe/pipe/kernel.pipe/handler.log
INFO    - Total complete:   52/  90  57%  skipped:   11, failed:   13
ERROR   - qemu_x86                  tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random/handler.log
INFO    - Total complete:   53/  90  58%  skipped:   11, failed:   14
ERROR   - qemu_x86                  tests/kernel/mutex/sys_mutex/system.mutex          FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/mutex/sys_mutex/system.mutex/handler.log
INFO    - Total complete:   54/  90  60%  skipped:   11, failed:   15
ERROR   - qemu_x86                  tests/kernel/mem_protect/futex/kernel.futex        FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/mem_protect/futex/kernel.futex/handler.log
INFO    - Total complete:   55/  90  61%  skipped:   11, failed:   16
ERROR   - qemu_x86                  tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool.concept FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool.concept/handler.log
INFO    - Total complete:   56/  90  62%  skipped:   11, failed:   17
ERROR   - qemu_x86                  tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing FAILED: Timeout
[... cut ...]
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/obj_tracing/kernel.objects.tracing/handler.log
INFO    - Total complete:   90/  90  100%  skipped:   11, failed:   47
INFO    - 32 of 79 tests passed (40.51%), 47 failed, 11 skipped with 0 warnings in 162.50 seconds
INFO    - In total 100 test cases were executed on 1 out of total 267 platforms (0.37%)
INFO    - 77 tests executed on platforms, 13 tests were only built.
```
I am also seeing that at least in my machine, samples/subsys/settings (a fairly verbose sample) is sometimes getting truncated output. I'm on Ubuntu 20.04.

andrewboie, 2020-07-19:
[image]

When this happens, this is what I see in `htop`. Note the very busy Python process, and all the qemu processes that are running for a long time. Also note how most cores are doing kernel rather then user stuff.

andrewboie, 2020-09-03:
This is unfortunately causing some problems if a test has a lot of console output and crashes at some point.

For example:

```
$ cat /dev/shm/sanity-out/qemu_x86_64/tests/kernel/threads/thread_apis/kernel.threads.apis/handler.log
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200625_115407-9426dddc0a1f-zephyr
)
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.3.0-2441-g8e2a7886b7b4  ***
Running test suite threads_lifecycle
===================================================================
START - test_threads_spawn_params
 PASS - test_threads_spawn_params
===================================================================
START - test_threads_spawn_priority
 PASS - test_threads_spawn_priority
===================================================================
START - test_threads_spawn_delay
 PASS - test_threads_spawn_delay
===================================================================
START - test_threads_spawn_forever
 PASS - test_threads_spawn_forever
===================================================================
START - test_thread_start
 PASS - test_thread_start
===================================================================
START - test_threads_suspend_resume_cooperative
 PASS - test_threads_suspend_resume_cooperative
===================================================================
START - test_threads_suspend_resume_preemptible
 PASS - test_threads_suspend_resume_preemptible
===================================================================
START - test_threads_priority_set
 PASS - test_threads_priority_set
===================================================================
START - test_threads_abort_self
 PASS - test_threads_abort_self
===================================================================
START - test_threads_abort_others
 PASS - test_threads_abort_others
===================================================================
START - test_threads_abort_repeat
 PASS - test_threads_abort_repeat
===================================================================
START - test_abort_handler
Child thread is running
Calling abort of child from parent
Child thread's abort handler called
 PASS - test_abort_handler
===================================================================
START - test_delayed_thread_abort
 PASS - test_delayed_thread_abort
===================================================================
START - test_essential_thread_operation
 PASS - test_essential_thread_operation
===================================================================
START - test_essential_thread_abort
ASSERTION FAIL [!(thread->base.user_options & ((1UL << (0))))] @ WEST_TOPDIR/zephyr/kernel/sched.c:493
        essential thread aborted
E: RAX: 0x0000000000000004 RBX: 0x0000000000118230 RCX: 0x0000000000000001 RDX: 0x0000000000000000
E: RSI: 0x00000000000001ed RDI: 0x0000000000116db8 RBP: 0x000000000011ac60 RSP: 0x0000000000187f78
E:  R8: 0x0000000000000001  R9: 0x0000000000000020 R10: 0x0000000000000000 R11: 0x0000000000000000
E: R12: 0x0000000000000246 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x0000000000187f78 RFLAGS: 0x0000000000000246 CS: 0x0018 CR3: 0x0000000000196000
E: RIP: 0x000000000010438e
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x000000000011c2e0 (ztest_thread)
 PASS - test_essential_thread_abort
===================================================================
START - test_systhreads_main
[... 73 lines cut ...]
===================================================================
START - test_thread_join_isr
ztest_thread: method 7, create join_thread
isr: joining join_thread
isr: k_thread_join() returned with -16
ztest_thread: method 6, create join_thread
join_thread: self-exiting
isr: joining join_thread
isr: k_thread_join() returned with 0
 PASS - test_thread_join_isr
===================================================================
START - test_thread_join_deadlock
 PASS - test_thread_join_deadlock
===================================================================
START - test_abort_from_isr
```

```
ERROR   - qemu_x86_64               tests/kernel/threads/thread_apis/kernel.threads.apis FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/threads/thread_apis/kernel.threads.apis/handler.log
```

I know the test is crashing because of the 2 exit code from the emulator, but the log is truncated.

Is this some kind of race where we detect an emulator exit and stop processing the log contents before it's completely flushed?

andrewboie, 2020-09-03:
OK this is weird -- in the above example, it claims "Exited with 2" but this is actually a timeout.

andrewboie, 2020-09-06:
Also seen this:

```
ERROR   - qemu_x86_64               tests/subsys/logging/log_core_additional/logging.add.async FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/subsys/logging/log_core_additional/logging.add.async/handler.log
INFO    - Total complete: 1570/2504  62%  skipped:  234, failed:    3
ERROR   - qemu_cortex_a53           tests/subsys/logging/log_immediate/logging.log_immediate FAILED: unexpected eof
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_a53/tests/subsys/logging/log_immediate/logging.log_immediate/handler.log
INFO    - Total complete: 1575/2504  62%  skipped:  234, failed:    4
ERROR   - qemu_cortex_m0            tests/subsys/logging/log_immediate/logging.log_immediate.clean_output FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_m0/tests/subsys/logging/log_immediate/logging.log_immediate.clean_output/handler.log
INFO    - Total complete: 1583/2504  63%  skipped:  234, failed:    5
ERROR   - mps2_an385                tests/subsys/logging/log_immediate/logging.log_immediate.clean_output FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/mps2_an385/tests/subsys/logging/log_immediate/logging.log_immediate.clean_output/handler.log
INFO    - Total complete: 1584/2504  63%  skipped:  234, failed:    6
ERROR   - qemu_cortex_m3            tests/subsys/logging/log_immediate/logging.log_immediate FAILED: unexpected eof
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_m3/tests/subsys/logging/log_immediate/logging.log_immediate/handler.log
INFO    - Total complete: 1590/2504  63%  skipped:  234, failed:    7
ERROR   - qemu_cortex_m0            tests/subsys/logging/log_immediate/logging.log_immediate FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_m0/tests/subsys/logging/log_immediate/logging.log_immediate/handler.log
INFO    - Total complete: 1615/2504  64%  skipped:  239, failed:    8
ERROR   - qemu_x86_64               tests/posix/common/portability.posix.common.newlib FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/posix/common/portability.posix.common.newlib/handler.log
INFO    - Total complete: 1764/2504  70%  skipped:  258, failed:    9
````

these are actually PASSING. but the test output is extremely long-winded:

```
$ cat  /dev/shm/sanity-out/qemu_cortex_m3/tests/subsys/logging/log_immediate/logging.log_immediate/handler.log
*** Booting Zephyr OS build v2.4.0-rc1-5-g92b3af9c2722  ***
Running test suite test_log_immediate
===================================================================
START - test_log_immediate_preemption
[00:00:00.000,000] <inf> test: CONFIG_LOG_IMMEDIATE_CLEAN_OUTPUT not enabled. Text output will be garbled.
[00:00:00.020,000] <inf> test: test string printed 1 2 0x200000c0
[00:00:00.020,000] <inf> test: data:
00 00 00 00 00 00 00 00                          |........
[00:00:00.020,000] <inf> test: test string printed 1 2 0x20000158
[00:00:00.020,000] <inf> test: data:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
[00:00:00.020,000] <inf> test: test string printed 1 2 0x200001f0
[00:00:00.020,000] <inf> test: data:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
00 00 00 00 00 00 00 00                          |........
[00:00:00.020,000] <inf> test: test string printed 1 2 0x20000288
[00:00:00.020,000] <inf> test: data:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
[00:00:00.020,000] <inf> test: test string printed 1 2 0x20000320
[00:00:00.020,000] <inf> test: data:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
00 00 00 00 00 00 00 00                          |........
[00:00:00.050,000] <inf> test: test string printed 1 2 0x200000c0
[00:00:00.050,000] <inf> test: data:
00 00 00 00 00 00 00 00                          |........
[00:00:00.060,000] <inf> test: test string printed 1 2 0x20000158
[00:00:00.060,000] <inf> test: data:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
[00:00:00.060,000] <inf> test: test string printed 1 2 0x200001f0
[00:00:00.060,000] <inf> test: data:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
00 00 00 00 00 00 00 00                          |........
[00:00:00.060,000] <inf> test: test string printed 1 2 0x20000288
[00:00:00.060,000] <inf> test: data:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
[00:00:00.060,000] <inf> test: test string printed 1 2 0x20000320
[00:00:00.060,000] <inf> test: data:
[... 1446 lines cut ...]
[00:00:02.980,000] <inf> test: test string printed 1 2 0x20000158
[00:00:02.980,000] <inf> test: data:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
[00:00:02.980,000] <inf> test: test string printed 1 2 0x200001f0
[00:00:02.980,000] <inf> test: data:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
00 00 00 00 00 00 00 00                          |........
[00:00:02.980,000] <inf> test: test string printed 1 2 0x20000288
[00:00:02.980,000] <inf> test: data:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
[00:00:02.980,000] <inf> test: test string printed 1 2 0x20000320
[00:00:02.980,000] <inf> test: data:
[... cut ...]
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
00 00 00 00 00 00 00 00                          |........
[00:00:02.990,000] <inf> test: test string printed 1 2 0x200000c0
[00:00:02.990,000] <inf> test: data:
00 00 00 00 00 00 00 00                          |........
 PASS - test_log_immediate_preemption
===================================================================
Test suite test_log_immediate succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

nashif, 2020-09-10:
just bisected and hit this as the culprit:

```
6fae53c0449d05b9fbc9763a220e4bc1e39930cb is the first bad commit
commit 6fae53c0449d05b9fbc9763a220e4bc1e39930cb
Author: Wentong Wu <wentong.wu@intel.com>
Date:   Wed Jun 24 22:55:59 2020 +0800

    sanitycheck: handle timeouts correctly with qemu

    In some cases we were not handling return code correctly and evaluating
    timeouts as a pass. Report failure correctly.

    Fixes #26065

    Signed-off-by: Wentong Wu <wentong.wu@intel.com>

 scripts/sanity_chk/sanitylib.py | 5 ++++-
 1 file changed, 4 insertions(+), 1 deletion(-)
```

nashif, 2020-09-10:
looking...

nashif, 2020-09-13:
think I finally have something to fix this and other issues related, stay tuned.

PR #28372 (2020-12-01): Various sanitycheck fixes and improvements..
use python multiprocessing instead of a thread pool, resolves
various issues.

- [x] Device testing support with multiprocessing

Fixes #28040
Fixes #26949
Fixes #24652
Fixes #30099
Fixes #27583
closed 2020-12-01 by nashif