zephyrproject-rtos/zephyr #26953: settings: ISO C++ forbids converting a string constant to 'char*'

sslupsky, 2020-07-18:
When compiling a cpp source file using the settings_handler struct, the compiler throws a warning:
```
warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
```

For example, referring to the samples/subsys/settings/src/main.c, the following code block causes the above warning when compiling as cpp:
```
struct settings_handler alph_handler = {
		.name = "alpha",
		.h_get = NULL,
		.h_set = alpha_handle_set,
		.h_commit = alpha_handle_commit,
		.h_export = alpha_handle_export
};
```

I believe the issue is the definition of the struct in settings.h:
```
struct settings_handler {

	char *name;
	/**< Name of subtree. */

	int (*h_get)(const char *key, char *val, int val_len_max);
	/**< Get values handler of settings items identified by keyword names.
	 *
	 * Parameters:
	 *  - key[in] the name with skipped part that was used as name in
	 *    handler registration
	 *  - val[out] buffer to receive value.
	 *  - val_len_max[in] size of that buffer.
	 *
	 * Return: length of data read on success, negative on failure.
	 */

	int (*h_set)(const char *key, size_t len, settings_read_cb read_cb,
		     void *cb_arg);
	/**< Set value handler of settings items identified by keyword names.
	 *
	 * Parameters:
	 *  - key[in] the name with skipped part that was used as name in
	 *    handler registration
	 *  - len[in] the size of the data found in the backend.
	 *  - read_cb[in] function provided to read the data from the backend.
	 *  - cb_arg[in] arguments for the read function provided by the
	 *    backend.
	 *
	 *  Return: 0 on success, non-zero on failure.
	 */

	int (*h_commit)(void);
	/**< This handler gets called after settings has been loaded in full.
	 * User might use it to apply setting to the application.
	 *
	 * Return: 0 on success, non-zero on failure.
	 */

	int (*h_export)(int (*export_func)(const char *name, const void *val,
					   size_t val_len));
	/**< This gets called to dump all current settings items.
	 *
	 * This happens when @ref settings_save tries to save the settings.
	 * Parameters:
	 *  - export_func: the pointer to the internal function which appends
	 *   a single key-value pair to persisted settings. Don't store
	 *   duplicated value. The name is subtree/key string, val is the string
	 *   with value.
	 *
	 * @remarks The User might limit a implementations of handler to serving
	 * only one keyword at one call - what will impose limit to get/set
	 * values using full subtree/key name.
	 *
	 * Return: 0 on success, non-zero on failure.
	 */

	sys_snode_t node;
	/**< Linked list node info for module internal usage. */
};
```

Can char *name safely be declared const char *name instead?  This will clear up the warning.

pabigot, 2020-07-20:
Yeah, that's just wrong regardless of language.  There are actually a lot of these.

sslupsky, 2020-07-21:
@pabigot Thank you for the PR.

PR #26972 (2020-07-22): settings: disallow modifying the content of a subtree name
There may be value in being able to rename a subtree, but there is no
identified need to be able modify the text of an assigned subtree
name.

Fixes #26953
closed 2020-07-22 by nashif