zephyrproject-rtos/zephyr #26954: devicetree: warning: braces around scalar initializer

sslupsky, 2020-07-18:
The new devicetree macros have created an issue when initializing the config struct within a device driver.  For instance, the following initialization code:
```
static const struct dev_config config = {
	.i2c_addr = DT_PROP(DT_NODELABEL(label), reg),
}
```
generates the following compiler warning:
```
warning: braces around scalar initializer
```
Is there a way to construct a scaler assignment and avoid the warning?

galak, 2020-07-20:
Where are you seeing this warning?  Which compiler?  All the sensor drivers in tree that are i2c aren't seeing this issue.

galak, 2020-07-20:
Have you tried using DT_REG_ADDR() macro?

sslupsky, 2020-07-20:
@galak I have not tried the DT_REG_ADDR() macro.  I'll gave that a try is it works like a charm.  No problem now.

Though, I am not sure why.  It seems to me they return the same thing?

Thank you for pointing the other macro out.

closed 2020-07-20 by sslupsky

mbolivar-nordic, 2020-07-20:
> Though, I am not sure why. It seems to me they return the same thing?

They do not, not in general. You can't read `reg` property values safely with `DT_PROP`. This is covered in the documentation:

https://docs.zephyrproject.org/latest/guides/dts/api-usage.html#reg-properties

The cells in the `reg` array need to be processed into address/size values using `#address-cells` and `#size-cells`. DT_PROP just returns the C array initializer for the entire `reg` value, which has type `array`, which mixes the address and size values in a way that is not useful.

sslupsky, 2020-07-20:
@mbolivar I was unaware the guide had been updated.  Your explanation tells me more clearly why DT_PROP generates the warning.  Thank you.