zephyrproject-rtos/zephyr #26970: usb: overflow of USB transfers leads to clogging

emob-nordic, 2020-07-20:
**Describe the bug**
After #26747 was merged I observed that some samples (hid-cdc, cdc_acm_composite) don't work anymore.
If USB stack is calling read transfer on every SUSPEND->RESUME it will quickly run out of transfer slots making it unable to finish the enumeration and make the device unusable.

Increasing the number of USB transfer slots does not help much. The issue could be seen when switching from virtual machine to native machine many times. Sooner or later there will be shortage of USB transfers and the device will no longer work.
``` diff 
diff --git a/subsys/usb/usb_transfer.c b/subsys/usb/usb_transfer.c
index 230337783b..9317b3bc8b 100644
--- a/subsys/usb/usb_transfer.c
+++ b/subsys/usb/usb_transfer.c
@@ -13,7 +13,7 @@
 
 LOG_MODULE_REGISTER(usb_transfer, CONFIG_USB_DEVICE_LOG_LEVEL);
 
-#define MAX_NUM_TRANSFERS           4 /** Max number of parallel transfers */
+#define MAX_NUM_TRANSFERS           20 /** Max number of parallel transfers */
 
 struct usb_transfer_sync_priv {
 	int tsize;
```

**Environment (please complete the following information):**
 - OS: Linux Mint
 - zephyr-sdk v0.11.2
 - 98bcc51b09 
 - Board: nrf52840dk_nrf52840. I haven't tested it with other boards but it also may happen.

**Additional context**
Probably USB transfer mechanism should overwrite existing transfer slot for particular endpoint when there is one existing.

emob-nordic, 2020-07-20:
cc @jfischer-phytec-iot

jfischer-phytec-iot, 2020-07-20:
> If USB stack is calling read transfer on every SUSPEND->RESUME it will quickly run out of transfer slots making it unable to finish the enumeration and make the device unusable.

Actually not, because SUSPEND event would cancel all transfers and cdc-acm re-triggers the transfer only for SUSPEND->RESUME transition. I need more information, logs...

emob-nordic, 2020-07-20:
> Actually not, because SUSPEND event would cancel all transfers and cdc-acm re-triggers the transfer only for SUSPEND->RESUME transition. I need more information, logs...

It seems it doesn't happen. I will post a patch that can show that. I've found that the issue could be easily addressed by this patch:

``` diff
diff --git a/subsys/usb/usb_transfer.c b/subsys/usb/usb_transfer.c
index 230337783b..f4b29f6bb7 100644
--- a/subsys/usb/usb_transfer.c
+++ b/subsys/usb/usb_transfer.c
@@ -203,6 +203,8 @@ int usb_transfer(uint8_t ep, uint8_t *data, size_t dlen, unsigned int flags,
 
 	key = irq_lock();
 
+	usb_cancel_transfer(ep);
+
 	for (i = 0; i < ARRAY_SIZE(ut_data); i++) {
 		if (!k_sem_take(&ut_data[i].sem, K_NO_WAIT)) {
 			trans = &ut_data[i];
```
I am just wondering if this should only be done for OUT endpoints.

emob-nordic, 2020-07-20:
btw @jfischer-phytec-iot  transfers are only canceled when the device was previously configured.
```
if (status == USB_DC_DISCONNECTED || status == USB_DC_SUSPEND) {
	if (usb_dev.configured) {
		usb_cancel_transfers();
		if (status == USB_DC_DISCONNECTED) {
			foreach_ep(disable_interface_ep);
			usb_dev.configured = false;
		}
	}
}
```
You can see an issue when swapping the device from native machine to virtual machine frequently. Then You run out of transfers pretty quickly. I will post more logs soon

emob-nordic, 2020-07-21:
I've added (I better like to change it to ERR instead of enabling all DBG logs) to the patch in Issue description:
``` diff
@@ -198,11 +201,13 @@ int usb_transfer(uint8_t ep, uint8_t *data, size_t dlen, unsigned int flags,
        struct usb_transfer_data *trans = NULL;
        int i, key, ret = 0;
 
-       LOG_DBG("Transfer start, ep 0x%02x, data %p, dlen %zd",
+       LOG_ERR("Transfer start, ep 0x%02x, data %p, dlen %zd",
                ep, data, dlen);
 
        key = irq_lock();
```
Here is the output.
```
. . . <reset the board>
*** Booting Zephyr OS build zephyr-v2.3.0-1124-g98bcc51b0925  ***
[00:00:00.005,126] <inf> cdc_acm_composite: Wait for DTR
[00:00:00.009,185] <inf> usb_cdc_acm: Device suspended
[00:00:00.009,185] <inf> usb_cdc_acm: Device suspended
*** Booting Zephyr OS build zephyr-v2.3.0-1124-g98bcc51b0925  ***
[00:00:00.005,126] <inf> cdc_acm_composite: Wait for DTR
[00:00:00.009,155] <inf> usb_cdc_acm: Device suspended
[00:00:00.009,185] <inf> usb_cdc_acm: Device suspended
[00:00:00.208,251] <err> usb_cdc_acm: Suspend->Resume, create transfer on ep 0x02
[00:00:00.208,251] <err> usb_transfer: Transfer start, ep 0x02, data 0x200000b4, dlen 64
[00:00:00.208,282] <err> usb_nrfx: Endpoint 0x02 is not enabled
[00:00:00.208,282] <inf> usb_cdc_acm: Define resumed
[00:00:00.208,282] <err> usb_cdc_acm: Suspend->Resume, create transfer on ep 0x01
[00:00:00.208,282] <err> usb_transfer: Transfer start, ep 0x01, data 0x20000030, dlen 64
[00:00:00.208,282] <err> usb_nrfx: Endpoint 0x01 is not enabled
[00:00:00.208,312] <inf> usb_cdc_acm: Define resumed
[00:00:05.876,739] <err> usb_cdc_acm: Configured, create transfer on ep 0x02
[00:00:05.876,770] <err> usb_transfer: Transfer start, ep 0x02, data 0x200000b4, dlen 64
[00:00:05.876,800] <inf> usb_cdc_acm: Device configured
[00:00:05.876,800] <err> usb_cdc_acm: Configured, create transfer on ep 0x01
[00:00:05.876,800] <err> usb_transfer: Transfer start, ep 0x01, data 0x20000030, dlen 64
[00:00:05.876,831] <inf> usb_cdc_acm: Device configured
[00:00:06.209,075] <inf> cdc_acm_composite: DTR set, start test
. . . <Enumeration completed. Doubled transfers for OUT endpoints, no more slots>
. . . <Type a character on 1st ACM port>
. . . <OUT transfer consumed and new one created. All slots occupied>
[00:00:15.422,302] <err> usb_transfer: Transfer start, ep 0x02, data 0x200000b4, dlen 64
. . . <Start IN transfer>
[00:00:15.422,424] <err> usb_transfer: Transfer start, ep 0x82, data 0x20002733, dlen 1
. . . <No free slots -> error>
[00:00:15.422,424] <err> usb_transfer: No transfer slot available, print all transfers
. . . <Only OUT transfers present, duplicated>
[00:00:15.422,424] <err> usb_transfer: Transfer on ep 0x02, status:-16
[00:00:15.422,424] <err> usb_transfer: Transfer on ep 0x01, status:-16
[00:00:15.422,424] <err> usb_transfer: Transfer on ep 0x02, status:-16
[00:00:15.422,454] <err> usb_transfer: Transfer on ep 0x01, status:-16
. . . <Type more>
[00:00:41.033,874] <err> usb_transfer: Transfer start, ep 0x02, data 0x200000b4, dlen 64
[00:00:45.767,547] <err> usb_transfer: Transfer start, ep 0x02, data 0x200000b4, dlen 64
[00:00:47.245,086] <err> usb_transfer: Transfer start, ep 0x02, data 0x200000b4, dlen 64
[00:00:50.546,997] <err> usb_transfer: Transfer start, ep 0x02, data 0x200000b4, dlen 64
[00:00:50.928,771] <err> usb_transfer: Transfer start, ep 0x02, data 0x200000b4, dlen 64
. . . <Type on 2nd ACM port, same result >
[00:01:04.105,072] <err> usb_transfer: Transfer start, ep 0x01, data 0x20000030, dlen 64
[00:01:04.105,194] <err> usb_transfer: Transfer start, ep 0x84, data 0x20002933, dlen 1
[00:01:04.105,194] <err> usb_transfer: No transfer slot available, print all transfers
[00:01:04.105,194] <err> usb_transfer: Transfer on ep 0x02, status:-16
[00:01:04.105,224] <err> usb_transfer: Transfer on ep 0x01, status:-16
[00:01:04.105,224] <err> usb_transfer: Transfer on ep 0x02, status:-16
[00:01:04.105,224] <err> usb_transfer: Transfer on ep 0x01, status:-16
[00:01:24.668,304] <err> usb_transfer: Transfer start, ep 0x01, data 0x20000030, dlen 64
[00:01:24.918,518] <err> usb_transfer: Transfer start, ep 0x01, data 0x20000030, dlen 64
[00:01:25.144,378] <err> usb_transfer: Transfer start, ep 0x01, data 0x20000030, dlen 64
[00:01:25.316,436] <err> usb_transfer: Transfer start, ep 0x01, data 0x20000030, dlen 64
```

Because of duplicate transfers the cdc is unable to create new IN transfer. I have opened both ports `/dev/ttyACM*`. I am able to type but unable to see redirected characters I am typing.

PR #27025 (2020-08-19): usb: cdc: Do not duplicate USB transfer slots for OUT transfers
This commit fixes an issue when more than one USB transfer is
reserved for same OUT endpoint.

There are 2 scenarios when this could happen:
- The Host sends SET_CONFIGURATION(1) request twice.
- The Host triggers Suspend->Resume->Configured
  event sequence for the device.

USB tranfers are not canceled on SUSPEND event
when the device was not configred previously.

Because of that USB transfer slot is reserved twice
for the same OUT endpoint and lead to shortage of USB
transfer slots quickly.

Without this patch CDC ACM class re [... cut]
closed 2020-08-19 by ioannisg