zephyrproject-rtos/zephyr #26975: Control never returns from stm32_i2c_msg_write(), whe SCL is pulled low permanently (hardware fault occurs)

mtahirbutt, 2020-07-20:
**Describe the bug**
A clear and concise description of what the bug is.

When there is a faulty hardware sensor connected via I2C interface, the control is not returned from stm32_i2c_msg_write() function even during initialization, I did tests with SHT3xd sensor sample app to conclude to this bug in the software. In order to further confirm the problem, I artificially pulled SCL signal line to low which resulted in the same problem.
I further debugged the stm32_i2c_msg_write(), and found that k_sem_take(&data->device_sync_sem, K_FOREVER); does not return the control, semophore resources are never released by other thread using those resources. 

What have you tried to diagnose or workaround this issue?

I tried to reduce timeout value with k_sem_take(,timeout) function and used 5s, the control is returned after 5s but data->current.is_nack is not set which does not return the function with correct error code so that the calling function (driver initialization or sensor read may react appropriately upon such hardware failure).
 

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; 
2. cmake -DBOARD=stm32L475
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

stm32_i2c_msg_write() should return error code when I2C finds a faulty hardware on the interface.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

The main application does not handle faulty hardware line effectively.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Windows)
 - Toolchain (e.g Zephyr 2.3.0, SDK 0.11.2, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

erwango, 2020-07-20:
@mtahirbutt can you check https://github.com/zephyrproject-rtos/zephyr/pull/21308/files could solve the issue you're facing ?

mtahirbutt, 2020-07-20:
@erwango, thanks for a great tip, i am working on that.

mtahirbutt, 2020-07-21:
Hi, I tested  i2c_ll_stm32: Add timeout to I2C read/write functions PR only adjusted semophore timeouts, that works fine with my zephyr 2.3.0 version. I tested with both faulty and open connector for two sensors TMP116 and shtxd35. So far, experienced no problem with 1 ms timeout value.

ydamigos, 2020-08-12:
Hi @mtahirbutt,
I have updated PR #21308, could you check again if it works for you?

mtahirbutt, 2020-08-12:
Hi @ydamigos, thanks for your nice contribution, i will work on that in some time.

closed 2020-10-26 by github-actions[bot]