zephyrproject-rtos/zephyr #26978: RFC (style) prefer ATTRIBUTE_UNUSED over ARG_UNUSED

pabigot, 2020-07-20:
It's always seemed clunky to have `ARG_UNUSED(foo);` statements in the body of a function to avoid whining from some compilers invoked with specific options.  Googling `ARG_UNUSED` shows (me) Zephyr in the top three results, with the next highest the transition in GCC from `ARG_UNUSED` to `ATTRIBUTE_UNUSED` about ten years ago.

`__attribute__((__unused__))` is supported in GCC and Keil, and likely other toolchains.

Proposal is to document the preference to use it, and update the source tree to change things like:

```
void button_pressed(struct device *gpiob, struct gpio_callback *cb,
                    uint32_t pins)
{
      ARG_UNUSED(gpiob);
      ARG_UNUSED(pins);
      ARG_UNUSED(cb);

      /* Set pin to HIGH/LOW */
      gpio_pin_set(gpio_led, PIN0, 0);
      ready_to_sleep = true;

}
```

to

```
void button_pressed(struct device *gpiob ATTRIBUTE_UNUSED,
                    struct gpio_callback *cb ATTRIBUTE_UNUSED,
                    uint32_t pins ATTRIBUTE_UNUSED)
{
      /* Set pin to HIGH/LOW */
      gpio_pin_set(gpio_led, PIN0, 0);
      ready_to_sleep = true;
}
```

which is less cluttered.

stephanosio, 2022-06-07:
Apparently `__unused` is a thing:
https://github.com/zephyrproject-rtos/zephyr/blame/fb9b3bcd9326923f8c08fd6754485c28d8f6e915/include/zephyr/toolchain/gcc.h#L239

We could extend this as `__maybe_unused` and `__always_unused` as in the Linux kernel.

stephanosio, 2022-06-07:
Related to https://github.com/zephyrproject-rtos/zephyr/issues/34413