zephyrproject-rtos/zephyr #26984: sys/device_mmio.h API design should accept generic DT node identifiers

mbolivar-nordic, 2020-07-20:
**Describe the bug**

I'm calling this a bug since I think the API design has a very minor flaw that should be easy to fix. However, validity should be discussed and determined by @galak and @andrewboie. No hurt feelings if this just gets closed :smile: .

During a Slack discussion with @andrewboie, he pointed out the way the new device_mmio.h API uses devicetree instance numbers exclusively to get information out of the DT. There is nothing wrong with instance numbers and they are recommended for use whenever possible, but this is an API design problem because it's not always possible to get nodes by instance number; in some cases, drivers need to get node identifiers from node labels, for example. (This is described in gory detail in the DT HOWTOs page.)

I haven't looked at this code in detail, but I would suggest that `DEVICE_MMIO_ROM_INIT(instance)`  be renamed `DEVICE_MMIO_ROM_INST_INIT(instance)`. The renamed macro would be written in terms of a version of  `DEVICE_MMIO_ROM_INIT(node_id)` that takes a generic node identifier instead of an instance number.

So something like this (untested, just being concrete here):

```
#ifdef DEVICE_MMIO_IS_IN_RAM
/* ... */
#define Z_DEVICE_MMIO_ROM_INITIALIZER(node_id) \
        { \
                .phys_addr = DT_REG_ADDR(node_id), \
                .size = DT_REG_SIZE(node_id) \
        }
#else
/* ... */
#define Z_DEVICE_MMIO_ROM_INITIALIZER(node_id) \
        { \
                .addr = DT_REG_ADDR(node_id) \
        }
#endif

/* ... */

#define DEVICE_MMIO_ROM_INIT(node_id) Z_DEVICE_MMIO_ROM_INITIALIZER(node_id)
#define DEVICE_MMIO_ROM_INST_INIT(instance) DEVICE_MMIO_ROM_INIT(DT_DRV_INST(instance))
```

Or something like that. The point being that there should be a non-instance-number way to do the same thing.

**Expected behavior**

I expect that all APIs which use data from devicetree nodes should have some version that accepts arbitrary node identifiers, not just instance numbers.

andrewboie, 2020-07-21:
Would like to confirm with @galak but I am happy to make this change.

galak, 2020-07-21:
I have no issue with this change.  Didn't realize we merged this, was hoping to see if we'd put addr in device struct since seems like every device needs this, so having a common way of storing and managing it would be nice. (But that's a slight aside to the issue raised here).

andrewboie, 2020-07-21:
> was hoping to see if we'd put addr in device struct since seems like every device needs this, so having a common way of storing and managing it would be nice. 

This presumes that every driver has exactly one MMIO region to manage which I don't think is necessarily the case. If you want to come up with a proposal for this change feel free, but at this time I'm just focusing on what @mbolivar has reported here.

PR #27530 (2020-08-14): drivers: use node IDs for DEVICE_MMIO.*_INIT
There is nothing wrong with instance numbers and they are
recommended for use whenever possible, but this is an API
design problem because it's not always possible to get nodes
by instance number; in some cases, drivers need to get node
identifiers from node labels, for example.

Change these APIs (which are not yet in any Zephyr release)
to take node IDs instead of instance IDs.

Fixes: #26984
commit 63c3e153 (2020-08-14): drivers: use node IDs for DEVICE_MMIO.*_INIT

There is nothing wrong with instance numbers and they are
recommended for use whenever possible, but this is an API
design problem because it's not always possible to get nodes
by instance number; in some cases, drivers need to get node
identifiers from node labels, for example.

Change these APIs (which are not yet in any Zephyr release)
to take node IDs instead of instance IDs.

Fixes: #26984
closed 2020-08-14 by carlescufi