zephyrproject-rtos/zephyr #26987: [Coverity CID :211475] Unintended sign extension in drivers/sensor/wsen_itds/itds.c

zephyrbot, 2020-07-20:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/476fc405e7/drivers/sensor/wsen_itds/itds.c#L263

Category: Integer handling issues
Function: `itds_accl_channel_get`
Component: Drivers
CID: [211475](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=211475)

Details:

```
257    
258         for (i = ofs_start; i <= ofs_stop ; i++, val++) {
259             int64_t dval;
260    
261             /* Sensitivity is exposed in ug/LSB */
262             /* Convert to m/s^2 */
>>>     CID 211475:  Integer handling issues  (SIGN_EXTENSION)
>>>     Suspicious implicit sign extension: "ddata->scale" with type "uint16_t" (16 bits, unsigned) is promoted in "ddata->samples[i] * ddata->scale" to type "int" (32 bits, signed), then sign-extended to type "long long" (64 bits, signed).  If "ddata->samples[i] * ddata->scale" is greater than 0x7FFFFFFF, the upper bits of the result will all be 1.
263             dval = (int64_t)((ddata->samples[i] * ddata->scale * SENSOR_G) /
264                     1000000LL);
265             val->val1 = (int32_t)(dval / 1000000);
266             val->val2 = (int32_t)(dval % 1000000);
267         }
268     }
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

MaureenHelm, 2020-07-21:
@ssekar15 will you please take a look?

ssekar15, 2020-07-21:
@MaureenHelm sure, I will take a look

carlescufi, 2020-09-08:
@ssekar15 could you send a patch to resolve this?

MaureenHelm, 2020-11-03:
@ssekar15 ping

closed 2021-01-17 by github-actions[bot]