zephyrproject-rtos/zephyr #26990: i2c transfers are timing out with SSD1306 display

MZachmann, 2020-07-20:
**Describe the bug**
In order to avoid waiting forever for a semaphore in the i2c code in case of hardware failure,  in pull request #25077, a fixed timeout of 100ms for i2c transfers was added. Unfortunately the ssd1306 display (and almost certainly others) take longer than that to copy the buffer to the display and now fail to work correctly. To be specific the ssd1306 takes 103ms by empirical test to execute update.

**To Reproduce**
Steps to reproduce the behavior:
Build an application using the ssd1306. I'm using a nrf52840 but doubt that matters.

**Expected behavior**
The i2c transfer should be considered good. Can we maybe just add an override timeout value in the CONFIG list instead of randomly bumping this to a different number?

**Impact**
It causes the display to fail to work entirely.

**Logs and console output**
I changed this line 112 of i2c_nrfx_twi.c :

		ret = k_sem_take(&(get_dev_data(dev)->completion_sync),
				 I2C_TRANSFER_TIMEOUT_MSEC);

to use a forever timeout and the display started working again. Before doing that it was popping this error message:

			LOG_ERR("Error on I2C line occurred for message %d", i);

for message 1.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain gnu-arm
 - Commit SHA or Version used : 5092541e2b59f4a319520d677a59b706cd3bd665

jfischer-phytec-iot, 2020-07-23:
@pabigot @anangl What is your opinion, revert 6ecc849bad44f2fdabb3ae748e637ae5afa4e93a or make it configurable with a default of 1s?

pabigot, 2020-07-23:
I see several things that concern me about #25077, including but not limited to the hard-coded wait specific to a single system.

Until we have a way to set a per-transaction/address/bus timeout, I would prefer to see a system-wide Kconfig-controllable I2C bus timeout.  A default of 1 s seems reasonable, allowing for a transfer of roughly 12.5 KiBy at the lowest standard bus speed.

pabigot, 2020-08-24:
The SSD1306 datasheet indicates the device can operate at 400 kHz which would solve this problem, but practically it might remain for other devices.

In #27715 I surveyed the existing state of I2C timeouts, which is frankly a mess.  Even adding a global default, with platform overrides, would be messy and inconsistent: it's not clear where the timeout should apply (single message or all messages in a transfer) or what to do when one is detected (report `-EIO`, report `-ETIMEDOUT`, do or do not recover the bus).

So for now I'm throwing up my hands and proposing #27744.  A more holistic approach might get attention in the 2.5 development cycle.

closed 2020-08-24 by MaureenHelm