zephyrproject-rtos/zephyr #26994: [Coverity CID :210616] Resource leak in lib/updatehub/updatehub.c

zephyrbot, 2020-07-20:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/476fc405e7/lib/updatehub/updatehub.c

Category: Resource leaks
Function: `start_coap_client`
Component: Other
CID: [210616](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210616)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

nandojve, 2020-07-21:
CC: @otavio

dleach02, 2020-07-24:
176        while (resolve_attempts--) {
   	10. alloc_arg: getaddrinfo allocates memory that is stored into addr. [Note: The source code implementation of the function has been overridden by a builtin model.]
177                ret = getaddrinfo(UPDATEHUB_SERVER, port, &hints, &addr);
178                if (ret == 0) {

The getaddrinfo() will allocate memory for the addr field. Need to pair this with a call at end to freeaddrinfo(addr)

nandojve, 2020-07-25:
Hi @dleach02 thank you for point me that. I don't have access to report yet and with this tip I know what I need fix.
By the way, I discovered by other way #26720 and opened #27028.

dleach02, 2020-07-27:
The fix for this particular problem is simple and just needs a freeaddrinfo() in the scope of this routine if there was a successful getaddrinfo(). The rest of the issues you are bringing up in the other items are somewhat orthogonal to this and can be dealt with separately. I would just submit a simple PR to fix this particular coverity (and real resources leak) issue.

PR #27222 (2020-07-30): lib: updatehub: Fix getaddrinfo resource leak
Add missing freeaddrinfo to fix memory leak.

Fixes #26994.
commit 897a1772 (2020-07-30): lib: updatehub: Fix getaddrinfo resource leak

Add missing freeaddrinfo to fix memory leak.

Fixes #26994.
commit 44fc9dca (2020-11-17): lib: updatehub: Fix getaddrinfo resource leak

Add missing freeaddrinfo to fix memory leak.

Fixes #26994.
closed 2020-07-30 by nashif