zephyrproject-rtos/zephyr #27003: CMakeLists.txt newline check is too strict

andrewboie, 2020-07-21:
**Describe the bug**
```
Please remove blank lines at end of 'arch/x86/CMakeLists.txt'
##[error]See https://docs.zephyrproject.org/latest/contribute/#coding-style for more details.
```

The problem is, the file looks like this:

```
    --kernel $<TARGET_FILE:${ZEPHYR_PREBUILT_EXECUTABLE}>
    --output pagetables.bin
    $<$<BOOL:${CMAKE_VERBOSE_MAKEFILE}>:--verbose>
    WORKING_DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}
    DEPENDS ${ZEPHYR_PREBUILT_EXECUTABLE} ${GEN_MMU}
  )

  add_bin_file_to_the_next_link(pagetables_bin_target pagetables)
endif()
```

The endif() on the last line has a newline on the end of it.

I can understand forbidding multiple blank lines at the end, but forbidding the last line of build system code to have a newline on it is ridiculous.

pabigot, 2020-07-26:
Are you absolutely certain this diagnostic wasn't appropriate at some point?  I see this content in #27001 in commit be4727c66051b6b676b00a0f0dfd29afae2d61b5, and scripts/ci/check_compliance.py doesn't complain about the last line ending in a newline.  The diagnostic does not appear in the current ci results AFAICT.

If I add a true empty line at the end of the file so the last two characters of the file are `\n\n` (and force the commit because git's pre-commit check does not allow empty lines) I do get the diagnostic, which is correct in that case.

andrewboie, 2020-08-10:
@pabigot my feeling is that `\n\n` at the end of a file can and should be flagged.

But a single `\n`? Requiring that the last line of code not end in a newline, I think that's silly, and some editors like VS Code by default add one.

andrewboie, 2020-08-10:
Maybe something is not working as intended -- I filed this bug when it flagged a CMakeLists.txt in a PR for ending in a single newline.

pabigot, 2020-08-10:
Yes, the check should complain about `lastline\n\n`, but should not complain about `lastline\n`.  AFAICT that's exactly how it does behave.

andrewboie, 2020-08-10:
Let's see if this fails: https://github.com/zephyrproject-rtos/zephyr/pull/27478

andrewboie, 2020-08-10:
That passed the nits check. Can't rule out PEBCAK or some issue on my end, closing this.

closed 2020-08-10 by andrewboie