zephyrproject-rtos/zephyr #27010: net: ieee802154: wrong header generation

Delsian, 2020-07-21:
**Describe the bug**

In ieee802154_send() as first step we use ieee802154_compute_header_size() getting ll_hdr_size value.
Next, ieee802154_create_data_frame() calling  generate_fcf_grounds() and generate_addressing_fields()
Sometimes generate_fcf_grounds() generates another set of fs->fc.src_addr_mode and fs->fc.dst_addr_mode values than  ieee802154_compute_header_size supposed before. Both variants valid, but generate_addressing_fields() tooks more bytes than we reserved for header.
As result, we get NET_ERR("Could not generate data frame %zu vs %u") and drops the packet.

**To Reproduce**

Rarely observed during retransmit of fragmented 6lowpan packets. No stable way to reproduce.

**Proposal to improve**

Join ieee802154_compute_header_size() and data_addr_to_fs_settings() into single function to sync header size and fs flags.
In this case, we can drop checking at line 767 file subsys\net\l2\ieee802154\ieee802154_frame.c
@rlubos , @jukkar what do you think about this change? Any side-effects?

**Environment (please complete the following information):**
 - OS: (Windows
 - Toolchain Zephyr SDK, GCC
 - Commit SHA 611f216708b9e6caf37df8a79875fb00803e4c97

**Logs**
Some additional log messages added to improve verbosity:
[00012473] <dbg> net_if.net_if_tx: (tx_workq): Processing (pkt 0x2000a998, prio 0) network packet iface 0x20000980/1
[00012474] <dbg> net_ieee802154_frame.ieee802154_compute_header_size: (tx_workq): Len 1 3
[00012476] <dbg> net_ipv6.nbr_lookup: (tx_workq): Get NBR info for fe80::1e00:1c00:1751:3731
[00012477] <dbg> net_ipv6.nbr_print: (tx_workq): [2] 0x200001b0 1/1/1/1/0 pending 0x00000000 iface 0x20000980/0 ll 8E:40:56:47:B8:84:14:C0 addr fe80::8c40:5647:b884:14c0
[00012479] <dbg> net_ieee802154_frame.ieee802154_compute_header_size: (tx_workq): Len 4 +10
[00012480] <dbg> net_ieee802154_frame.ieee802154_compute_header_size: (tx_workq): Len 5 +2
[00012481] <dbg> net_ieee802154_frame.ieee802154_compute_header_size: (tx_workq): Computed size of 15
[00012482] <dbg> net_ieee802154_frame.ieee802154_create_data_frame: (tx_workq): buf 0x200012ec
[00012483] <dbg> net_ieee802154_frame.generate_addressing_fields: (tx_workq): G0 0x200012ef +3
[00012484] <dbg> net_ieee802154_frame.generate_addressing_fields: (tx_workq): G1 13
[00012485] <dbg> net_ieee802154_frame.generate_addressing_fields: (tx_workq): G6 21
[00012486] <err> net_ieee802154_frame: Could not generate data frame 21 vs 15

rlubos, 2020-07-22:
I've seen that `ieee802154_compute_header_size` reaches into NBR cache to calculate the header length:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/l2/ieee802154/ieee802154_frame.c#L487

I wonder if that could be the case, the cache entry could've expired after the packet was queued, but before it was processed, it would calculate incorrect header size (too small as reported in this issue). `nbr_lookup` is also used to obtain LL address before the packet is queued, so the packet would be just passed over with a valid LL address set.

Delsian, 2020-07-24:
Both my bugs (this and https://github.com/zephyrproject-rtos/zephyr/issues/26919) were resolved with a temporary workaround.
As my system doesn't require 802.15.4 routing functionality, I put "CONFIG_NET_ROUTE=n" (and small change in Kconfig)
Now both showstopper bugs are gone and I can continue the normal development process.
I'll return to these bugs investigations later, actually treating this as technical debt.
You can close both bugs or leave them for future investigations. Thank you.

jukkar, 2020-07-24:
> You can close both bugs or leave them for future investigations.

I think this one is a valid issue, not sure about the other one.

Delsian, 2020-09-17:
Pronlem found. This issue occurs on wrong route configuration, when trying to send direct packet inside FE80:: via router.
Fixed by hardcoded net_if_ipv6_prefix_add(FE80::)

closed 2020-09-17 by Delsian