zephyrproject-rtos/zephyr #27015: Add custom transport support for MQTT

ematt, 2020-07-21:
Hello,

**Problem**
As a user, I want to have my own transport for MQTT. Unfortunately there is no way to do this as handles are hard coded. 

**Request**
Add API to support custom connect, disconnect, write, read functions to handle MQTT connectivity.

**Possible solution**

- Add `mqtt_set_transport_handle(connect_handle, disconnect_handle, read_handle, write_handle)` to allow insertion of custom handles into `transport_fn` array.

- Add a Kconfig flag similar to `MQTT_LIB_TLS` to append an additional handle set to `transport_fn`. The user provides the implementation at link time.

Best regards,
ematt

ematt, 2021-06-24:
Hello,

What is the status of this? I would like to do it if possible

Best regards,
ematt

rlubos, 2021-06-24:
To my knowledge there's no ongoing work on the issue. Feel free to propose something.

rlubos, 2021-06-30:
Resolved in https://github.com/zephyrproject-rtos/zephyr/pull/36549

closed 2021-06-30 by rlubos