zephyrproject-rtos/zephyr #27020: civetweb issues

xhpohanka, 2020-07-21:
I'm playing with `civetweb` webserver module on latest zephyr tree and I see some issues here

## stack size
When I try to serve a bit bigger pages that are in provided in samples I got stack overflow errors. Looking into code this does not lookg good to me
```
#if defined(USE_STACK_SIZE) && (USE_STACK_SIZE > 1)
#define ZEPHYR_STACK_SIZE USE_STACK_SIZE
#else
#define ZEPHYR_STACK_SIZE 8096
#endif
...
/* General purpose buffer size. */
#if !defined(MG_BUF_LEN) /* in bytes */
#define MG_BUF_LEN (1024 * 8)
#endif
...
static void
discard_unread_request_data(struct mg_connection *conn)
{
	char buf[MG_BUF_LEN];
```
`MG_BUF_LEN` sized arrays are allocated on stack on several places and this size is even bigger than default stack size used in zephyr sample.

## very slow upload
I have tested upload using form (`FileUploadForm` handler) in `modules/lib/civetweb/examples/embedded_c/embedded_c.c`. It is unacceptably slow on my embedded platform (stm32f429) and also very slow using qemu_x86, so I think it is not platform related.

Civetweb threads are made by pthread api with default priority equal to 0 which corresponds to lowest preempt priority of 15. I was able to speed up the upload only by changing the priority to highest one (-15). The same priority is used in `zephyr/samples/net/sockets/dumb_http_server_mt`. However with such setup my pages stop to work entirely with errors like `<err> net_pkt: Data buffer (28) allocation failed (arp_prepare_reply:495)`. I was not able to found the connection between packet allocations and civetweb priority yet. It seems wierd to me, this failed allocations start to appear even when the priority is inreased only by a small number, not neccessarily to -15.

Do anyone use civetweb on zephyr for some *real project*?

<details><summary>my config</summary>
<p>

```
#
# Modules
#

#
# Optional modules. Make sure they're installed, via the project manifest.
#
# CONFIG_CANOPENNODE is not set
CONFIG_CIVETWEB=y
# CONFIG_LIBMETAL is not set
# CONFIG_HAS_SEMTECH_LORAMAC is not set
# CONFIG_HAS_SEMTECH_RADIO_DRIVERS is not set
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
# CONFIG_MBEDTLS_LIBRARY is not set
CONFIG_MBEDTLS_CFG_FILE="config-tls-generic.h"

#
# TLS configuration
#

#
# Supported TLS version
#
# CONFIG_MBEDTLS_TLS_VERSION_1_0 is not set
# CONFIG_MBEDTLS_TLS_VERSION_1_1 is not set
CONFIG_MBEDTLS_TLS_VERSION_1_2=y
CONFIG_MBEDTLS_DTLS=y
# CONFIG_MBEDTLS_SSL_EXPORT_KEYS is not set
# end of Supported TLS version

#
# Ciphersuite configuration
#

#
[... 1053 lines cut ...]
# CONFIG_APPLICATION_DEFINED_SYSCALL is not set
# CONFIG_MAKEFILE_EXPORTS is not set
# CONFIG_LEGACY_DEVICETREE_MACROS is not set
CONFIG_DEPRECATED_ZEPHYR_INT_TYPES=y
# end of Build Options
# end of Build and Link Features

#
# Boot Options
#
# CONFIG_BOOTLOADER_MCUBOOT is not set
# CONFIG_BOOTLOADER_KEXEC is not set
# CONFIG_REBOOT is not set
# CONFIG_MISRA_SANE is not set
# end of Boot Options

#
# Compatibility
#
CONFIG_COMPAT_INCLUDES=y
# end of Compatibility

```
</p>
</details>

bel2125, 2020-07-21:
I do not know Zephyr (although I know some other embedded operating systems), but I might be able to provide some details on civetweb (I came here from https://github.com/civetweb/civetweb/issues/895).

## Regarding stack size: 

Indeed civetweb assumes the stack size to be some pages larger than the MG_BUF_LEN size. A buffer needs to contain at least the entire HTTP/1.x header. This is 
```
GET /some/path/to/some/file.name?query=string&max=contain&multiple=parameters HTTP/1.1
Host: www.example.com
User-Agent: Some-name-and-version-of-your-browser
Cookie: yet=another; list=of; key=value; pairs=with; unknown=length

```
Once the header is longer than MG_BUF_LEN, the request cannot be processed anymore - the current definition `#define MG_BUF_LEN (1024 * 8)` is plenty. Half of this might be sufficient. The URL+query string (/path/to...parameters) are not limited in any standard, but typical recommendations are 2,048 bytes. Microsoft Internet Explorer (if you want to call this a browser) has a limit of 2,083 bytes, Chrome of 2 MB or 32kB (depending on the platform). Cookies seem to have a limit around 4096 bytes for each cookie (test it yourself: http://browsercookielimits.squawky.net/), but you may have several of them.
If you know your URLs and query strings you plan to support on your server, and if you know the maximum cookie length, you may go lower in this limit, but keep some space for unexpected additional header fields sent by certain browsers.
Or increase the stack size - in particular if you plan to use Lua pages.

## Regarding priority:

I cannot recommend any details, since I do not know the priority levels of Zephyr.
From the CivetWeb point of view, there is one priority define `USE_MASTER_THREAD_PRIORITY` that could be used to move the master thread (the one accepting all incomming connections) to some SCHED_RR priority. If it is not set (default), CivetWeb will in itself not set any thread priorities.
CivetWeb offers a callback "init_thread" that is called from every thread after creation. You can use this to set the priority of the calling thread to something suitable for you. CivetWeb will not modify the priorities afterwards.

Not knowing Zephyr, I cannot directly tell you where the performance bottleneck might be. From another real time operating system, I know it was important to get the priorities right, together with the priorities of the network interface card interrupt and the priorities of the thread responsible for the TCP/IP stack. It was required to keep these thread priorities one level higher than the civetweb master thread. The worker threads could be the same as the master thread or one level lower.
I do not know if this experience allows any conclusion for Zephyr, or if it will be helpful in your case.

I hope this explanation "from the CivetWeb side" is of any use - probably if you add proper know how from the Zephyr side, it might help you solve this issue.

xhpohanka, 2020-07-23:
Thank you @bel2125 for sharing useful info about CivetWeb. I'm still sniffing around, trying to set correct prioirities etc. Unfortunatelly I'm still struggling with errors like `net_pkt: Data buffer (48) allocation failed (prepare_segment:444)` which appears on embedded platform as well as on `qemu_x86` simulation. This will probably need someone who is more familiar with Zephyr internals than me.

xhpohanka, 2020-07-24:
Ok. At least I have found the reason for failed allocations of packets or buffers. When I set civetweb priority close to TCP/IP stack it means that we are in cooperative range (zephyr priority < 0). In such case when civetweb is serving a bigger page (multiple `mg_write()`s in a loop) there is no chance for other thread (even with higher priority - like TCP/IP stack) to run without explicit `k_yield()` or sleep. 

My upload is still quite slow, though.

AntoineZen, 2020-08-05:
For the slow upload, you might be impacted by #26330. Have a look at the TCP window using Wireshark.

armandciejak, 2020-08-05:
@xhpohanka I noticed the same issues as you. For the allocation problem I increased the value for `CONFIG_NET_BUF_TX_COUNT`, but that's a workaround which doesn't scale.

I do not fully understand the relation between allocation failing and thread priorities. Shouldn't `mg_write` block when there are no more buffers available until some data has been transferred on the network?

bel2125, 2020-08-05:
> Shouldn't mg_write block when there are no more buffers available until some data has been transferred on the network?

From the pure civetweb point of view (no Zephyr specifics):
Civetweb uses non-blocking sockets when calling "send". If "send" returns EWOULDBLOCK, it calls "poll" to wait for a fixed time or until the socket is ready to send (if there is no "poll" it will use "select"). Anyway, if send/poll do not return success after some time (configured, default 30 seconds), mg_write will return an error.
Ín short: it blocks for a limited time (30 s) and then returns an error.

CivetWeb does not care (cannot know) why "send" is not ready ... peer not responding, no memory in TCP/IP stack, ... but will work as long as it returns "EWOULDBLOCK" (or OK). Any other return value will be treated as error, and the mg_write function returns immediately with an error.

I just found in the CivetWeb source: in most places it checks for "EWOULDBLOCK" and "EAGAIN", in some places only for "EWOULDBLOCK". In all gnuC libraries, they have the same value. According to google, there are some old (historic?) UNIX variants where they had different values. I hope this is not the case for Zephyr, since it might cause some issues.

armandciejak, 2020-08-06:
EWOULDBLOCK is the same as EAGAIN in Zephyr.
```
zephyr/lib/libc/minimal/include/errno.h:110:#define EWOULDBLOCK EAGAIN /* Operation would block */
```

xhpohanka, 2020-08-06:
> I do not fully understand the relation between allocation failing and thread priorities. Shouldn't `mg_write` block when there are no more buffers available until some data has been transferred on the network?

I also am not sure how `poll` works in zephyr. But in my case when I was runnig out of memory it could be solved by adding `k_yield` or `k_sleep` between successive `mg_write` calls.

carlescufi, 2020-09-08:
@PiotrZierhoffer could you please take a look since you are the maintainer of the civetweb module?

xhpohanka, 2020-09-09:
same new notes from my side:

- slow upload was solved with new TCP stack
- there are newer versions of civetweb than the one served in Zephyr, I'm just trying to run with actual master merged and so far it works

Nukersson, 2020-09-14:
@xhpohanka Try to gzip your files served from your civetweb http server and use mg_send_chunk() for bigger files. Still there may be issues, maybe. Let me know, if mg_send_chunk() improves your performance. If you need more information how to use it, see my PR #28323. 

P.S.: How much Bytes KiB or MiB needed your files be on your civetweb http server?

xhpohanka, 2020-10-08:
@Nukersson Thanks for suggestion of `mg_send_chunk()` I will try it. If I understand correctly it could also simplify serving of dynamic pages as you do not need to know Content-Length before the whole page.

Currently I have quite stable implementation of web server on my platform. Only time-to-time one of the pages is not transfered and need to be refreshed. I have merged mbedtls support for civetweb (originaly by qinch) and it also works with exception of loading files (firmware update). I also tried to alter civetweb source to use zephyr's internal tls_socket implementation and it works (no big testing though) the same (normal pages ok, upload fails).

The biggest web page is jquery. So around 85kB/

Nukersson, 2020-10-08:
@xhpohanka Thank you for sharing of your experience here. 

> Only time-to-time one of the pages is not transfered and need to be refreshed.
Believe me, I've seen same production code with well tested network staks and web/http servers. They do same files dropping and whole page reloading from time to time. I think, we need dig deeper her 😄 

>  The biggest web page is jquery. So around 85kB/
Take a look at #28323 PR and search for `gzip` in main `CMakeLists.txt`. You will se, how typical `*.js` files may be gzipped.
You can also gzip small and compressed version [jquery-3.5.1.slim.min.js](https://code.jquery.com/jquery-3.5.1.slim.min.js). I did it once and it worked fine for me.

Nukersson, 2020-10-08:
@xhpohanka By the way: does my answers solves your civetweb problems? Can we then close this issue? Otherways, if you have more questions I can answer - you're welcome!

Nukersson, 2020-11-07:
@xhpohanka Do you have any new insights?

xhpohanka, 2020-11-09:
Hello, I think that we can close this issue. The original problems was solved with TCP2 introduction and now I have just an occassional instabilities that are probably caused by tcp stack in connection with STM32H7 driver. I understand your suggestions with compressing the pages but I think that this should not be neccessary. We had a similar web on the same architecture and freertos and it works without any issues.
I will try to check file upload using TLS with civetweb author and eventually open a new issue here.
Thanks for support.

closed 2020-11-09 by xhpohanka