zephyrproject-rtos/zephyr #27028: Remove all HEAP allocation from Zephyr core and essential sub-systems

nandojve, 2020-07-21:
**Is your feature request related to a problem? Please describe.**

Zephyr is more than a OS core, it incorporates sub-systems and libraries following a principle to be easy to use have high scalability. This means that systems can grows very fast and we need keep eyes open to ensures that core and essential sub-systems are always safe.

The #26720 seems to open a point of great concern once dynamic memory allocation from HEAP could create problems to keep system safe all the time. Core and sub-systems can not suffer from any interference from bad memory usage over time. Most of time, memory problems are difficult to solve once product is on the field. A typical IoT device on real environment, in general, can only be accessed remotely and needs a network almost with a perfect implementation to achieve that. The change a74137f6 fix thread safe issue but open the door to a system instability situation for those that want a more secure and safe implementation using small memory footprint.

**Describe the solution you'd like**

1- Define the boundaries of core and essential sub-systems.
2- Core and essential sub-systems should never use HEAP and/or dynamic memory, they are essentials, and most not suffer from NOMEM error.
3- Define the memory allocation strategy with reserved memory that core and essential sub-systems should use it. It should be thread safe and should be deterministic.
4- Define a simple and clear checklist that must be followed by core and essential sub-system developers to ensure proper memory usage.

**Describe alternatives you've considered**

I'm assuming system need work for 10 years without reset and that should be the goal. If we open to Dynamic Memory allocation on core and essential sub-system we can have a starvation point over time. That can disable remote access to device and/or stop device to access network.

The concept of services can create a new option to handle that situation. They can be monitored and system can have a hook to determine if the service is memory hungry. This way, system can have a fallback, disable that service till next order. System than can call a default way that signal to a back end that a device may have a problematic firmware. Again, if system can't sent message because there is no available memory, system can close services by priority and enter on a state that allows at least send alerts to a back end and allows firmware upgrade to run. That kind of strategy allows recover from memory starvation and currently Zephyr allows memory allocation but there is no way to ensure that memory exists over time.

**Additional context**

It reminds me about the infinite memory problem. Big systems with process and services works pretty well with Dynamic Memory allocation because there are ways to clean up the system. Embedded software in general uses pre-defined memory because are the other side. There is no right or wrong but we need a recovery from a possible starvation. Watchdog is not the end solution, instead it can be part of the solution and it helps to identify the problem.

CC @otavio @pfalcon @cfriedt @andrewboie

andrewboie, 2020-07-21:
@nandojve Agree with all the points raised here.

I think for the particular case of a spec-compliant `getaddrinfo()`, we don't have great options because by its upstream design it returns a pointer from somewhere that has to be later freed, that isn't shared. It doesn't provide a way for the caller to provide this memory, so out of the heap it comes.

If we're willing, I think a workable solution would be to provide a Zephyr-specific variant of `getaddrinfo()` API which, as part of its parameter list, the caller provides a buffer that the API fills in with the addrinfo structures, returning an error code if the buffer wasn't big enough.

I don't think there are any other core kernel spots that require allocations (other than some alloc_ APIs which are documented as doing this and have non-allocating equivalents) but if you find them lets get rid of them too.

I think codifying in the Zephyr documentation where, when, and how heaps can be used is a great idea.

I've been thinking it might be useful to also introduce a "kill switch" for allocators, for situations where heaps or exhaustable pools are needed at early boot or application initialization (like allocating page tables), but after a certain point no more allocations should be necessary. Once the kill switch is thrown, no allocations from the OS are permitted.

> I'm assuming system need work for 10 years without reset and that should be the goal.

Definitely!

andrewboie, 2020-07-21:
cc @nashif @andyross @ceolin

andrewboie, 2020-07-21:
@nandojve this documentation might be helpful when building the docs proposed here. We have an enumeration of all the specific core kernel APIs which allocate, on the kernel side, memory to service a system call:

https://docs.zephyrproject.org/latest/reference/usermode/memory_domain.html#thread-resource-pools

These all only perform allocations when invoked from user mode, due to a strong need to keep kernel memory separate from untrusted threads. They all have non-allocating equivalents except for `k_object_alloc()`, which isn't usable unless CONFIG_DYNAMIC_OBJECTS is turned on.

Other than these documented cases, and the specific case of `getaddrinfo()` (which is just a wrapper function to the actual system call `zsock_getaddrinfo()` calling `calloc()` along the way) I hope there isn't anything else.

andrewboie, 2020-07-22:
One other subsystem I found: if asynchronous logging is enabled, logging a string requires the use of `log_strdup()`. I think we need to iterate on this so that you don't need to do this if you want to log a string asynchronously, I am thinking a different API specifically for logging strings or other binary data.

dleach02, 2020-07-27:
while I agree that having HEAP allocation with embedded systems is problematic, the definition of getaddrinfo() has it doing an allocation of memory from some location that needs to be paired with the free. Not sure how to get around that if you want access to getaddrinfo() function.

andrewboie, 2020-07-27:
> while I agree that having HEAP allocation with embedded systems is problematic, the definition of getaddrinfo() has it doing an allocation of memory from some location that needs to be paired with the free. Not sure how to get around that if you want access to getaddrinfo() function.

@dleach02 my idea was to introduce a parallel API to `getaddrinfo()` which offers the same functionality but gets the memory buffer as a parameter and not an allocation. Unless there is a strict requirement that Zephyr's official networking API has to follow Berkeley Sockets with no alternatives or deviations I think this will work.

Actually we sort of already have it in the sense that `zsock_getaddrinfo()` does exactly this. It is the actual system call; `getaddrinfo()` calls `calloc()` and then passes that memory to `zsock_getaddrinfo()`.

andyross, 2020-07-27:
FWIW: getaddrinfo() isn't a system call in traditional resolver implementations.  It's done in the user process using the lower level primitives directly.  We're sort of in uncharted territory with the existing design anyway.

andrewboie, 2020-07-27:
> FWIW: getaddrinfo() isn't a system call in traditional resolver implementations. It's done in the user process using the lower level primitives directly. We're sort of in uncharted territory with the existing design anyway.

@andyross I think the placement in the stack of the syscall is orthogonal to this particular GH ticket, in that we'd like a way to have the functionality of `getaddrinfo()` without a heap present. regardless of where the syscall is, I think we need to expose a different API to the application.

nandojve, 2020-10-10:
Hi @andrewboie. Is there any news on this subject?

Besides that:
> I've been thinking it might be useful to also introduce a "kill switch" for allocators, for situations where heaps or exhaustable pools are needed at early boot or application initialization (like [...]

I'm favor to have a "kill switch" thing. I can imagine many advantages to have such mechanism. I was wondering if the proposal could already be on 2.5. It may be interesting have this explored before LTS to collect some reactions.
What do you think?

andrewboie, 2020-10-11:
> It may be interesting have this explored before LTS to collect some reactions. What do you think?

Sure, although as an optional feature I expect this doesn't need universal consensus. If people don't want it they won't use it.

Are you willing to contribute some patches for this?