zephyrproject-rtos/zephyr #27032: zephyr network stack socket APIs are not thread safe

andrewboie, 2020-07-21:
**Describe the bug**
The current Zephyr implementation of the BSD Socket APIs is not thread safe. Concurrent use of networking APIs by multiple threads may lead to kernel crashes and/or data corruption.

**To Reproduce**
Found by code inspection and also this comment from the author:

>  You're absolutely right that it's not thread-safe, that's why I don't seriously suggest to revert that commit. On the other hand, I don't remember seeing a bug report of users hitting that issue. So, your heroic effort of fixing it is somewhat misplaced, because the whole socket subsystem is not thread safe. That's because its initial goal was simplicity and small resource usage, it was literally a proof of concept that we can use standards-based API on top of existing adhoc Zephyr API without significant overhead imposed on applications. That worked, and Sockets API was even made the only Zephyr's networking API. Its suggested mode of operation is that a single thread performs all networking activity, while other threads handle sensors, actuators, etc.

**Expected behavior**
The socket implementation for this operating system uses the necessary locking (spinlocks, semaphores) to be safe to use by multiple threads without crashes or data corruption.

The Open Group specifications for the socket APIs specifically cite threat safety as a requirement. For example https://pubs.opengroup.org/onlinepubs/009695399/functions/freeaddrinfo.html

**Impact**
Showstopper for users who use the network stack by more than one thread.

pfalcon, 2020-07-22:
As clear from the quoted context, "significant overhead" is a rhetoric figure, comparing situation which was before introduction of the Sockets API (all the callbacks, etc.) and concerns at that time that adding Socket API on top of that will be "too much".

Also, to clarify, Sockets API was developed with single-thread usecase in mind, or more specifically: socket
is an I/O object, and thus should be (per best practices) owned by a single thread and not shared with other threads. So no special provisions were made for inter-thread synchronization, instead that was left to be done on the application side, for applications which want to share the same socket across threads (over all this time, no such usecase in relation to Zephyr surfaced). But sockets implementation is anyway just a thin layer on top of net_context's (and other kernel IPC primitives), so multi-threading support for sockets largely depends on how well they support it (there shouldn't be much problem with that). There still apparently would be need to add locking when checking various flags and performing operations based on that.

carlescufi, 2020-09-15:
Blocked by #28406

PR #27054 (2021-04-15): net: socket: Add locking to prevent concurrent access
The BSD API calls were not thread safe. Add locking to fix this.

Fixes #27032
commit dde03c67 (2021-04-15): net: socket: Add locking to prevent concurrent access

The BSD API calls were not thread safe. Add locking to fix this.

Fixes #27032
closed 2021-04-15 by galak