zephyrproject-rtos/zephyr #27033: RFC: Update terminology related to I2C

pabigot, 2020-07-21:
Following [Linux](https://www.zdnet.com/article/linux-team-approves-new-terminology-bans-terms-like-blacklist-and-slave/) and [OSHWA](https://www.oshwa.org/a-resolution-to-redefine-spi-signal-names) I'd like to see the language associated with the I2C bus roles changed to be less insensitive.

At this time I haven't been able to identify an I2C-specific precedent to follow, but candidate terminology to replace "master" (device that drives the clock signal) and "slave" (reacts to the clock signal changes) include:

* leader/follower
* primary/secondary
* controller/peripheral

These are listed in decreasing order of my personal preference taking into account the semantics and pragmatics of the different terms and [how the protocol works](https://www.i2c-bus.org/).

At this time I'm floating this to see whether there's significant support or opposition to such a change.

mrrosen, 2020-07-27:
I would suggest "master" -> "host" and "slave" -> "device". This terminology is common in other bus protocols like USB; and at least the term "device" is used interchangeably with "slave" in some documentation Ive seen (Ive seen "device address" used a quite alot instead of "slave address"). I would recommend against at least two of the suggestions you provided:

- primary/secondary: some I2C devices have multiple I2C device addresses they can listen for (common in camera sensors); these addresses are referred to as secondary addresses, so it might be overloading terms leading to confusion.
- controller/peripheral: these terms are very common to refer the I2C module on an SoC level, either as an SoC peripheral or an I2C controller; so again, overloading terms might lead to confusion as to whether something is in SoC or not.

Of course, Im sure theres some overloading of "device" at very least in I2C device documentation, but I think its the best choice.

sslupsky, 2020-08-13:
I do not want to appear to be "insensitive" ... but unless you can get NXP to change the I2C specification, I see changing terminology here will result is a lot of confusion.  I think this is an issue to NXP to address first.

pabigot, 2020-09-03:
More details on why I'm using leader/follower:

"controller" already has a meaning in the Zephyr I2C world: it's the I2C peripheral that interacts with the bus.  The driver for that peripheral is the I2C controller, which can be configured into leader ("master") or follower ("slave") modes.

"Leader" is because it's the one that controls the clock, which the "follower" reacts to (absent clock stretching).

pabigot, 2020-09-07:
A little more background to keep this moving:

As a software professional and student of social science I feel strongly that the terms "master" and "slave" subtly reinforce structural racism in tech culture.  They also blunt the impact of the word "slave" in relation to the more euphemistic "human trafficking" that continues today world-wide.

Other major open source projects have already taken steps to address these problems.

As a contributing member of the open source community I'm in a position to act.  As the Zephyr I2C maintainer I have a responsibility to act.  So #27996 starts the process.

As I note there the text projecting my views upwards to Zephyr as a whole in the explanation of the new language is overstepping, and I'm willing to adjust that to TSC-approved text, or take it out entirely (though as I2C maintainer I think that would be foolish as it makes it harder for people to understand).

But if the requirement to merge the new documentation I've written is that it must be changed to use the terms "master" and "slave", I want that to be a positive decision made by Zephyr as a project, which I must then follow as a community member.

pabigot, 2021-02-06:
The requested decision has been made and the I2C terminology will change when the [conditions specified in the coding guidelines](https://docs.zephyrproject.org/latest/guides/coding_guidelines/index.html?highlight=inclusive#rule-a-2-inclusive-language) have been met.

Until that point this can make no progress.

pabigot, 2022-02-24:
Thanks.  For reference the updated specification can be found: https://www.nxp.com/docs/en/user-guide/UM10204.pdf

teburd, 2022-02-24:
Wonderful, glad to finally have some updated names. Glad to see you comment here @pabigot

closed 2022-06-29 by carlescufi