zephyrproject-rtos/zephyr #27037: No network interface found when running wifi sample

vanti, 2020-07-21:
**Describe the bug**
When running the wifi sample on cc3220sf_launchxl (or any wifi-based sample for that matter), it reports that there is no network interface to work with:

```
*** Booting Zephyr OS build zephyr-v2.3.0-1168-g62580748e536  ***

[00:00:00.002,000] <dbg> wifi_simplelink.simplelink_init: SimpleLink driver Initialized
[00:00:00.003,000] <err> net_if: There is no network interface to work with!
```

The sample works fine prior to commit 62580748e536d5ff14ecc290e114c034f1ccccfd. It seems there is a change in how interfaces are iterated over in net_if_init():

```
	Z_STRUCT_SECTION_FOREACH(net_if, iface) {
		init_iface(iface);
		if_count++;
	}
```

This code is no longer finding any interface.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr/samples/net/wifi
2. west build -b cc3220sf_launchxl
3. west flash
4. See error

**Expected behavior**
The sample should get to displaying the wifi shell's prompt without error.

**Impact**
This is a showstopper as far as using wifi on the TI cc32xx.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.11.3
 - Commit SHA or Version used: 62580748e536d5ff14ecc290e114c034f1ccccfd

jukkar, 2020-07-22:
Thanks for the report, I will look into it.

jukkar, 2020-07-22:
Latest master works ok with Ethernet in nucleo_767zi, lowering the priority.

jukkar, 2020-07-22:
I have a fix proposal, I was not able to test it with real hw as I do not have any board with offloaded network stack. Could you try it and +1 the PR if it works for you?

PR #27041 (2020-07-22): net: if: Network interface was missing for offloaded drivers
The NET_IF_OFFLOAD_INIT() macro was not converted to use
Z_STRUCT_SECTION_ITERABLE() macros in commit 62580748e536
("net: Use section foreach macros"). This meant that for those boards
that use offloaded network stack, the network interface was not created
and network connectivity was lost.

Fixes #27037
commit 2705a9d6 (2020-07-22): net: if: Network interface was missing for offloaded drivers

The NET_IF_OFFLOAD_INIT() macro was not converted to use
Z_STRUCT_SECTION_ITERABLE() macros in commit 62580748e536
("net: Use section foreach macros"). This meant that for those boards
that use offloaded network stack, the network interface was not created
and network connectivity was lost.

Fixes #27037
closed 2020-07-22 by nashif