zephyrproject-rtos/zephyr #27047: zefi.py assumes host GCC is x86

galak, 2020-07-22:
zefi.py will fail if this assumption isn't meet.  Which may be the case on either an arm64 host environment or on something like MacOS X.

Is it possible to use the x86 cross compiler that build zephyr here?

jhedberg, 2020-07-22:
AFAIK the plan is to eventually move to using the same build environment as for the rest of Zephyr, as per #26737

I'll let @andyross comment if there's some quick/simple intermediate fix we could do to zefi.py to eliminate the x86 dependency.

andyross, 2020-07-22:
We could pass the compiler to the script via the environment or the command line for a quick fix.  But note that there's been absolutely no thought given to non-x86 EFI environments.  It's very much an arch-specific feature right now.  The framework is there, I guess, but I'd expect significant changes would be needed to get the COFF linkage working on other platforms, etc...

I'd mostly view this as a duplicate of #26737, yeah.

jenmwms, 2020-11-18:
> We could pass the compiler to the script via the environment or the command line for a quick fix. But note that there's been absolutely no thought given to non-x86 EFI environments. It's very much [...]

@andyross I'll follow-up with you about this bug. @nashif clarified this is not a duplicate. I'll update this bug as I learn more.

galak, 2021-01-23:
Re-open this issue since we need an update to the SDK to enable EFI binary targets in binutils.

PR #31434 (2021-01-20): zefi.py: Use cross compiler while building zephyr
Currently, zefi.py takes host GCC OBJCOPY as
default. Fixing the script to use CMAKE_C_COMPILER
and CMAKE_OBJCOPY.

Fixes: #27047
commit 513b9f59 (2021-01-20): zefi.py: Use cross compiler while building zephyr

Currently, zefi.py takes host GCC OBJCOPY as
default. Fixing the script to use CMAKE_C_COMPILER
and CMAKE_OBJCOPY.

Fixes: #27047
PR #31595 (2021-01-25): x86: Fix zefi.py generation to use SDK toolchain
Fixes #27047
commit ae4e4b78 (2021-01-25): x86: Fix zefi.py generation to use SDK toolchain

With SDK 0.12.2 we have support to generation EFI binaries in binutils
which is needed by the zefi.py script.  Now that is there we can utilize
the SDK objcopy instead of assuming the host objcopy can do this (which
would only be the case on x86 linux host systems).

Fixes #27047
closed 2021-01-25 by nashif