zephyrproject-rtos/zephyr #27055: BlueZ with ESP32 boards supported or not?

danergo, 2020-07-22:
Guys, can you confirm this for me please?

I wrote an application (on Linux) which uses BlueZ to control Bluetooth devices (both Classic and BLE).

I found that Zephyr can actually work together with BlueZ, and it also supports ESP32 boards but I don't want to buy any ESP32 module until this is confirmed at least here :)

As I found a lot of Wifi issues and incompatibility with Zephyr and ESP32, the question seems obvious:

**Can I use Bluetooth (Classic and BLE) with ESP32?**

carlescufi, 2020-07-22:
No, you cannot because Zephyr does not support Bluetooth on the ESP32. You can however use a Nordic board for that, but it only has BLE, no Classic.

closed 2020-07-22 by carlescufi

Cimex97, 2020-10-02:
@danergo you can build and flash a [sample](https://github.com/espressif/esp-idf/tree/8bc19ba893e5544d571a753d82b44a84799b94b1/examples/bluetooth/hci/controller_hci_uart), that comes out of the esp-idf. That is a controller with hci_uart. That works fine with linux.