zephyrproject-rtos/zephyr #27056: Local header found before system header of same name

nslowell, 2020-07-22:
**Describe the bug**
If a sample application adds a local include directory that includes a local header file that happens to match the name of a header file in the zephyr include directory, because all the directories are added to the -I list, there is no distinction between #include <file.h> and #include "file.h" like there typically should be.  #include <file.h> should be found in sysroot locations before local directories.  Even if the order were reversed, because there is no distinction, any #include "file.h" references would find the zephyr header instead of the local header.  There should not be a restriction on local header file names.

**To Reproduce**
Steps to reproduce the behavior:
1. Use Getting Started: https://docs.zephyrproject.org/latest/getting_started/index.html to build blinky
2. make directory in samples/basic/blinky/src/ called "include"
     `mkdir samples/basic/blinky/src/include`
3. add empty header file called device.h to new include directory
     `touch samples/basic/blinky/src/include/device.h`
4. add include directory to project by adding the following line to samples/basic/blinky/CMakeLists.txt
   ` zephyr_library_include_directories(${ZEPHYR_BASE}/samples/basic/blinky/src/include)`
5. attempt to rebuild blinky
6. observe error

**Expected behavior**
Zephyr system files should be able to find the zephyr system file device.h since they #include <device.h>

**Beginning of Error**
 ```
ccache /users/nlowell/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.3.0-898-g668e359d8558 -DKERNEL -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/users/nlowell/zephyrproject/zephyr/samples/basic/blinky/src/include -I/users/nlowell/zephyrproject/zephyr/include -Izephyr/include/generated -I/users/nlowell/zephyrproject/zephyr/soc/arm/atmel_sam0/samd21 -I/users/nlowell/zephyrproject/zephyr/drivers -I/users/nlowell/zephyrproject/zephyr/soc/arm/atmel_sam0/common/. -I/users/nlowell/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/users/nlowell/zephyrproject/modules/hal/atmel/asf/sam0/include/samd21 -isystem /users/nlowell/zephyrproject/zephyr/lib/libc/minimal/include -isystem /users/nlowell/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /users/nlowell/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Os -imacros/users/nlowell/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m0plus -mthumb -mabi=aapcs -imacros/users/nlowell/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/users/nlowell/zephyrproject/zephyr/samples/basic/blinky=CMAKE_SOURCE_DIR -fmacro-prefix-map=/users/nlowell/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/users/nlowell/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c /users/nlowell/zephyrproject/zephyr/samples/basic/blinky/src/main.c
In file included from /users/nlowell/zephyrproject/zephyr/samples/basic/blinky/src/main.c:10:
/users/nlowell/zephyrproject/zephyr/include/drivers/gpio.h:353:48: warning: 'struct device' declared inside parameter list will not be visible outside of this definition or declaration
  353 | typedef void (*gpio_callback_handler_t)(struct device *port,
```

**Impact**
This limits/dictates header file names in local source and causes very confusing errors when the name conflict occurs.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK and West build system
 - v2.3.0

tejlmand, 2020-08-26:
This seems correct behavior to me.
When you do:
```
zephyr_library_include_directories(${ZEPHYR_BASE}/samples/basic/blinky/src/include)
```
inside `samples/basic/blinky/CMakeLists.txt`, then you are effectively adding `src/include` to the default search path of the `app` library, but not to other Zephyr libraries.

This can be verified by:
```
$ cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -Bbuild samples/basic/blinky/
# Build only libkernel.a and it's dependencies
$ ninja -Cbuild kernel
...
 /opt/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.3.0-2118-g94721b316f63 -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../kernel/include -I../arch/arm/include -I../include -Izephyr/include/generated -I../soc/arm/nordic_nrf/nrf52 -I/projects/github/ncs/modules/hal/cmsis/CMSIS/Core/Include -I/projects/github/ncs/modules/hal/nordic/nrfx -I/projects/github/ncs/modules/hal/nordic/nrfx/drivers/include -I/projects/github/ncs/modules/hal/nordic/nrfx/mdk -I/projects/github/ncs/modules/hal/nordic/. -I/projects/github/ncs/modules/debug/segger/rtt -isystem ../lib/libc/minimal/include -isystem /opt/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /opt/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Os -imacros/projects/github/ncs/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros/projects/github/ncs/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/projects/github/ncs/zephyr/samples/basic/blinky=CMAKE_SOURCE_DIR -fmacro-prefix-map=/projects/github/ncs/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/projects/github/ncs=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/kernel/CMakeFiles/kernel.dir/device.c.obj -MF zephyr/kernel/CMakeFiles/kernel.dir/device.c.obj.d -o zephyr/kernel/CMakeFiles/kernel.dir/device.c.obj   -c /projects/github/ncs/zephyr/kernel/device.c
...
[33/33] : && /opt/cmake-3.17.1-Linux-x86_64/bin/cmake -E rm -f zephyr/kernel/libkernel.a && /opt/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-ar qc zephyr/kernel/libkernel.a  zephyr/kernel/CMakeFiles/kernel.dir/device.c.obj zephyr/kernel/CMakeFiles/kernel.dir/errno.c.obj zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj zephyr/kernel/CMakeFiles/kernel.dir/idle.c.obj zephyr/kernel/CMakeFiles/kernel.dir/init.c.obj zephyr/kernel/CMakeFiles/kernel.dir/kheap.c.obj zephyr/kernel/CMakeFiles/kernel.dir/mailbox.c.obj zephyr/kernel/CMakeFiles/kernel.dir/mem_slab.c.obj zephyr/kernel/CMakeFiles/kernel.dir/mempool.c.obj zephyr/kernel/CMakeFiles/kernel.dir/msg_q.c.obj zephyr/kernel/CMakeFiles/kernel.dir/mutex.c.obj zephyr/kernel/CMakeFiles/kernel.dir/pipes.c.obj zephyr/kernel/CMakeFiles/kernel.dir/queue.c.obj zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj zephyr/kernel/CMakeFiles/kernel.dir/sem.c.obj zephyr/kernel/CMakeFiles/kernel.dir/stack.c.obj zephyr/kernel/CMakeFiles/kernel.dir/system_work_q.c.obj zephyr/kernel/CMakeFiles/kernel.dir/thread.c.obj zephyr/kernel/CMakeFiles/kernel.dir/thread_abort.c.obj zephyr/kernel/CMakeFiles/kernel.dir/version.c.obj zephyr/kernel/CMakeFiles/kernel.dir/work_q.c.obj zephyr/kernel/CMakeFiles/kernel.dir/smp.c.obj zephyr/kernel/CMakeFiles/kernel.dir/timeout.c.obj zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj && /opt/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-ranlib zephyr/kernel/libkernel.a && :
```
this works as expected with no errors.

But building of `libapp.a` fails, cause you requested `app` library to use `src/include` when compiling.
Note, `main.c` is part of `app` library.
```
$ ninja -Cbuild app
[1/2] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
/opt/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.3.0-2118-g94721b316f63 -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/projects/github/ncs/zephyr/samples/basic/blinky/src/include .... -o CMakeFiles/app.dir/src/main.c.obj   -c /projects/github/ncs/zephyr/samples/basic/blinky/src/main.c
In file included from /projects/github/ncs/zephyr/samples/basic/blinky/src/main.c:10:
```

`zephyr_library_include_directories()` will call CMake `target_include_directories()` with the current Zephyr library, which in this case will be `app`.
https://cmake.org/cmake/help/v3.18/command/target_include_directories.html

tejlmand, 2020-08-26:
Closing as this is not a bug.

closed 2020-08-26 by tejlmand

nslowell, 2020-08-26:
Can you offer a solution to the scenario?

tejlmand, 2020-08-26:
what are your expectations ?
You request all files belonging to `app` library to be compiled with `${ZEPHYR_BASE}/samples/basic/blinky/src/include` as include path, so did you expect them to NOT be compiled with that include path ?

`main.c` belongs to `app` lib, as can be seen by this line:
```
target_sources(app PRIVATE src/main.c)
```

tejlmand, 2020-08-26:
maybe you are mixing up how includes are handled by the compiler.
Zephyr currently supports gcc based compilers (and clang for native builds), and with gcc it would seem that you would not want to use `target_include_directories()` but handle this include path manually with `-iquote` as compile flag to get the behavior you might be looking for:
https://gcc.gnu.org/onlinedocs/gcc-4.3.1/cpp/Search-Path.html
https://gcc.gnu.org/onlinedocs/gcc-9.3.0/cpp/Invocation.html#Invocation

> Directories specified with -iquote apply only to the quote form of the directive, #include "file". Directories specified with -I, -isystem, or -idirafter apply to lookup for both the #include "file" and #include <file> directives.

tejlmand, 2020-08-27:
If you do the following in CMakeLists.txt instead:
```
zephyr_library_compile_options(-iquote ${ZEPHYR_BASE}/samples/basic/blinky/src/include)
```
then everything builds as expected:
```
$  ninja -Cbuild app 
ninja: Entering directory `build'
[2/2] Linking C static library app/libapp.a
```

If we then change `main.c` to use:
```
include "device.h"
```
instead, then we'll get:
```
$ ninja -Cbuild app 
ninja: Entering directory `build'
[1/2] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
/opt/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.3.0-2118-g94721b316f63 -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/projects/github/ncs/zephyr/include -Izephyr/include/generated -I/projects/github/ncs/zephyr/soc/arm/nordic_nrf/nrf52 -I/projects/github/ncs/modules/hal/cmsis/CMSIS/Core/Include -I/projects/github/ncs/modules/hal/nordic/nrfx -I/projects/github/ncs/modules/hal/nordic/nrfx/drivers/include -I/projects/github/ncs/modules/hal/nordic/nrfx/mdk -I/projects/github/ncs/modules/hal/nordic/. -I/projects/github/ncs/modules/debug/segger/rtt -isystem /projects/github/ncs/zephyr/lib/libc/minimal/include -isystem /opt/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /opt/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Os -imacros/projects/github/ncs/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros/projects/github/ncs/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/projects/github/ncs/zephyr/samples/basic/blinky=CMAKE_SOURCE_DIR -fmacro-prefix-map=/projects/github/ncs/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/projects/github/ncs=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -iquote /projects/github/ncs/zephyr/samples/basic/blinky/src/include -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c /projects/github/ncs/zephyr/samples/basic/blinky/src/main.c
In file included from /projects/github/ncs/zephyr/samples/basic/blinky/src/main.c:8:
/projects/github/ncs/zephyr/samples/basic/blinky/src/include/device.h:1:1: error: static assertion failed: "fail"
    1 | _Static_assert(0, "fail");
      | ^~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```
because the `-iquote` tells gcc to prefer the `src/include/` path first when quotes are used as `"device.h"`.
Note: I added `_Static_assert(0, "fail");` to `src/include/device.h` to get a failure during compilation, otherwise it would just continue and `device.h` from system would ALSO be included.

Diff with the failure:
```
diff --git a/samples/basic/blinky/CMakeLists.txt b/samples/basic/blinky/CMakeLists.txt
index 54f8909407..6961388b9f 100644
--- a/samples/basic/blinky/CMakeLists.txt
+++ b/samples/basic/blinky/CMakeLists.txt
@@ -4,4 +4,6 @@ cmake_minimum_required(VERSION 3.13.1)
 find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
 project(blinky)
 
+zephyr_library_compile_options(-iquote ${ZEPHYR_BASE}/samples/basic/blinky/src/include)
+
 target_sources(app PRIVATE src/main.c)
diff --git a/samples/basic/blinky/src/main.c b/samples/basic/blinky/src/main.c
index 5e21be3436..3f1625bf72 100644
--- a/samples/basic/blinky/src/main.c
+++ b/samples/basic/blinky/src/main.c
@@ -5,7 +5,7 @@
  */
 
 #include <zephyr.h>
-#include <device.h>
+#include "device.h"
 #include <devicetree.h>
 #include <drivers/gpio.h>
```

tejlmand, 2020-08-27:
@nslowell FYI ^^^

nslowell, 2020-08-27:
Thanks! I guess I was wondering if there should be a zephyr_library_... macro that resulted in the -iquote setting, but I guess this solution works.