zephyrproject-rtos/zephyr #27071: USB: CDC-ACM uart console hijacks usb_enable call preventing user applications from registering callbacks

pabigot, 2020-07-22:
The UART console driver, when associated with USB_CDC_ACM device via `CONFIG_USB_UART_CONSOLE`, improperly invokes `usb_enable(NULL)` although that function is documented to be under the control of the application, which may want to provide a callback.

This has been true "forever", but became more visible due to #25894 which causes the call to `usb_enable()` to be invoked for the console even when `CONFIG_USB_UART_DTR_WAIT` is not selected.

There are two concerns exposed by this behavior:

* First, as long as `usb_enable()` includes this requirement:

      * Class-specific initialization and registration must be performed by the user
      * before invoking this, so that any data or events on the bus are processed
      * correctly by the associated class handling code.

  then drivers should not be invoking it unconditionally if their doing so prevents the user from ensuring this initialization is performed in the right order.  (It may be that it doesn't prevent this, but that's certainly not obvious. and the obvious alternative solution of initialization in `SYS_INIT()` is not consistent with how this would be done if the application did not select `CONFIG_USB_UART_CONSOLE=y` and instead invoked `usb_enable()` from main or some other application context.)

* Second, it's not obvious that the correct decision when the console device can't be opened is to cause the application to busy-wait until USB becomes available, even if `WAIT_FOR_DTR` is selected.

The same issue applies to #26122 which proposes to add another situation where the infrastructure, rather than the application, is invoking `usb_enable()`.

It's not immediately clear how this should can be resolved.  One question is whether for non-USB-UART console devices failure to initialize causes the application to hang.  If so, then changing this may be harder to justify.

If not, then perhaps the console could left in a dormant neutral state until something signals that USB (and possibly DTR) becomes available.  Transitioning out of that state would require the UART console driver to be notified when these conditions are met, something that can be signaled only by the USB subsystem.

sslupsky, 2020-07-22:
@pabigot That title is a better description of the problem.

sslupsky, 2020-07-22:
There are a couple other things that might be helpful with regard to the DTR functionality of the startup.  Namely, if you want the console output to begin right away without having to hit a key on the console, you need to set a couple other flow control settings:
```
uart_line_ctrl_set(dev, UART_LINE_CTRL_DCD, 1);
uart_line_ctrl_set(dev, UART_LINE_CTRL_DSR, 1);
```

pabigot, 2020-07-22:
> There are a couple other things that might be helpful with regard to the DTR functionality of the startup.

Please open an issue for that.  It's not something that'd be addressed here.  (I haven't noticed that need, myself.)

sslupsky, 2020-07-22:
It can be done in the application.  I thought I would mention it in case you found it useful.

pabigot, 2020-07-22:
`netusb`, `tracing`, and `openthread/platform/uart` all try to enable USB without coordination also.

carlescufi, 2020-07-29:
The problem that we need to decide on is:

- We can enable USB when the console is initialized (i.e. what is being done now) and then provide the application with a way of adding its own callback **and** finding out the current USB state (this is not feasible for all USB drivers). This allows early print statements to go out during the boot process.

- We remove the auto-initialization of USB in the console and require the application to call `usb_enable()`. This will cause some early boot messages . The other issue is waiting for DTR. One possibility would be:

```
main()
{
  /* anything sent to the console so far has been dropped */
   int res = usb_enable(callback);
   /* console is enabled now */
   while(!dtr) {
       uart_line_ctrl_get(uart_console_dev, UART_LINE_CTRL_DTR, &dtr);
       k_msleep(100);
  }
  /* console is ready, messages won't be dropped */
}
```

pabigot, 2020-07-29:
And  why does USB have to be initialized at application runlevel?  (Rhetorical question.)

> This is our preferred solution at this point.

OK, just make very clear that using a USB console (which is not optional on some platforms) means any pre-APPLICATION log messages will be dropped (which is very much not ideal; I'm going to have to look into non-console log destinations).

I do like that this path allows removal of `USB_UART_DTR_WAIT` which is either badly named or mis-implemented as it only (ever?) applied to a console UART.

I do not, though, believe that applications should be forced to implement that magic check.  It should be abstracted to a function `uart_console_wait_until_ready()` which would ideally delegate to `cdc_acm_wait_for_dtr()` which would do the magic.  (`uart_console_wait_until_ready()` should return `-ENOTSUP` for non-USB consoles.)

Also (reminder to self): the Kconfig help for `USB_UART_CONSOLE` needs to be updated to make much more clear what functionality selecting it does and does not affect.

pabigot, 2020-07-29:
Also don't lose sight of [the other places](https://github.com/zephyrproject-rtos/zephyr/issues/27071#issuecomment-662679087) that might be breaking the API.
```
drivers/console/uart_console.c: ret = usb_enable(NULL);
subsys/net/lib/openthread/platform/uart.c:      int ret = usb_enable(NULL);
subsys/tracing/tracing_backend_usb.c:   ret = usb_enable(NULL);
subsys/usb/class/netusb/netusb.c:       ret = usb_enable(NULL);
```

emob-nordic, 2020-07-30:
> Also don't lose sight of [the other places](https://github.com/zephyrproject-rtos/zephyr/issues/27071#issuecomment-662679087) that might be breaking the API.

I am working on PR adding this. In case of any doubts will post a message. We talked about the issue together with USB folks and that's the only way out we see for now.

emob-nordic, 2020-08-04:
>Also don't lose sight of the other places that might be breaking the API.
drivers/console/uart_console.c: ret = usb_enable(NULL);
subsys/net/lib/openthread/platform/uart.c:      int ret = usb_enable(NULL);
subsys/tracing/tracing_backend_usb.c:   ret = usb_enable(NULL);
subsys/usb/class/netusb/netusb.c:       ret = usb_enable(NULL);

I put some work towards that and I can't say I like that. For `uart_console.c` and `tracing_backend_usb.c` changes are acceptable but for `netusb.c` putting `usb_enable` to application (main) means forcing main to execute a lot of dependent code I personally don't like. Zperf sample is using those and calling `usb_enable` from application main means that some additional code must be added (I am not a fan of that) just because the user wants USB. I enclose a wip [branch](https://github.com/emob-nordic/zephyr/tree/usb_console_enable) as reference.

I would like to bring it back to discussion. @carlescufi @anangl @jfischer-phytec-iot (@pabigot if You're intrested).

PR #27427 (2020-08-13): usb: Enable USB by the application.
The only reasonable solution right now to address #27071 is forcing the application to manually enable USB subsystem.
This patch updates remaining samples (not included in #20375) to call `usb_enable()` in `main`.

@markus-becker-tridonic-com please review the sample You added recently (last commit). I don't know what was expected to be on usb-cdc but right now I can see some garbage output (with master as well).
Furthermore:
- the sample has several Kconfig warnings that could be addressed (both overlays & without any).
- The sample builds with warnings whe [... cut]
closed 2020-08-13 by carlescufi