zephyrproject-rtos/zephyr #27081: missing `python3-psutil` dependency

jenmwms, 2020-07-23:
**Describe the bug**
Sanitycheck fails to run tests in Qemu due to unsatisfied dependency on `psutil` python module 

**To Reproduce**
Steps to reproduce the behavior:
1. Follow  [Getting Started Guide](https://docs.zephyrproject.org/latest/getting_started/index.html#getting-started-guide)
2. Fail on last step of [ Get Zephyr and install Python dependencies](https://docs.zephyrproject.org/latest/getting_started/index.html#getting-started-guide)

**Expected behavior**
No errors after `pip3 install --user -r ~/zephyrproject/zephyr/scripts/requirements.txt`

**Impact**
Prevents running tests in Qemu using sanitycheck until resolved. 

**Logs and console output**
```
    $ ./scripts/sanitycheck --platform=qemu_x86 -T tests/subsys/shell/
    Install the anytree module to use the --test-tree option
    Install tabulate python module with pip to use --device-testing option.
    Install tabulate python module with pip to use --device-testing option.
    Install psutil python module with pip to run in Qemu.
    Renaming output directory to /home/jennife2/zephyrproject/zephyr/sanity-out.5
    INFO    - JOBS: 4
    INFO    - Building initial testcase list...
    INFO    - 2 test configurations selected, 0 configurations discarded due to filters.
    INFO    - Adding tasks to the queue...
    Exception in thread qemu_x86/tests/subsys/shell/shell_history/shell.history:
    Traceback (most recent call last):
      File "/usr/lib64/python3.7/threading.py", line 926, in _bootstrap_inner
        self.run()
      File "/usr/lib64/python3.7/threading.py", line 870, in run
        self._target(*self._args, **self._kwargs)
      File "/home/jennife2/zephyrproject/zephyr/scripts/sanity_chk/sanitylib.py", line 816, in _thread
        cpu_time = QEMUHandler._get_cpu_time(pid)
      File "/home/jennife2/zephyrproject/zephyr/scripts/sanity_chk/sanitylib.py", line 771, in     _get_cpu_time
        proc = psutil.Process(pid)
    NameError: name 'psutil' is not defined

    Exception in thread qemu_x86/tests/subsys/shell/shell_flash/shell.history:
    Traceback (most recent call last):
      File "/usr/lib64/python3.7/threading.py", line 926, in _bootstrap_inner
        self.run()
      File "/usr/lib64/python3.7/threading.py", line 870, in run
        self._target(*self._args, **self._kwargs)
      File "/home/jennife2/zephyrproject/zephyr/scripts/sanity_chk/sanitylib.py", line 816, in _thread
        cpu_time = QEMUHandler._get_cpu_time(pid)
      File "/home/jennife2/zephyrproject/zephyr/scripts/sanity_chk/sanitylib.py", line 771, in     _get_cpu_time
        proc = psutil.Process(pid)
    NameError: name 'psutil' is not defined
```

**Environment (please complete the following information):**
 - OS: Fedora 30
 - Toolchain: Zephyr SDK 0.11.3 
 - Commit SHA or Version used: e47866c

**Additional context**
- Resolved with `sudo dnf install python3-psutil`

galak, 2020-07-23:
Hmm, psutils is listed in https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/requirements-run-test.txt

jenmwms, 2020-07-23:
> Hmm, psutil is listed in https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/requirements-run-test.txt

Yes, it is listed there. Does that file specify to install it, or is it used to check/display if the package needs to be installed to support the run time testing in sanitycheck?

jenmwms, 2020-07-23:
I submitted a PR #27083. For reference, please see related bug for Ubuntu recently resolved in similar approach #25129

nashif, 2020-07-23:
we cant have every python dependency installed via system package manager, this is done using pip3 and is documented already, so all you need to do is ```pip3 install -r scripts/requirements.txt -U``` when something stops working or when a new dependency is added.

jenmwms, 2020-07-23:
Thanks @nashif, I did not realize that distinction. I took this approach with system package manager because the suggested method in the guide (same as your suggestion basically) was failing to install psutil. Please see the attached console output when I run  `pip3 install --user -r ~/zephyrproject/zephyr/scripts/requirements.txt` and similarly for `pip3 install --user psutil`

[consoleUsingReqtScriptforInstallFails.txt](https://github.com/zephyrproject-rtos/zephyr/files/4967750/consoleUsingReqtScriptforInstallFails.txt)

[consoleUsingpsutilforInstallFails.txt](https://github.com/zephyrproject-rtos/zephyr/files/4967784/consoleUsingpsutilforInstallFails.txt)

I have success installing when I use this patch (using system package manager for install on Fedora). Can you please suggest a better placement for this, or shall it remain a user exercise if they choose to use Fedora rather than Ubuntu?

galak, 2020-07-23:
@jenmwms I think what you need to update for the Fedora docs is having the python devel package - I think (`python3-devel`).  See if that lets the `pip3 install --user -r ~/zephyrproject/zephyr/scripts/requirements.txt` work.

jenmwms, 2020-07-23:
Thanks, @galak! That (python3-devel) did not fix the issue either.

galak, 2020-07-23:
> Thanks, @galak! That (python3-devel) did not fix the issue either.

Same issue w/regards to Python.h?

On my fedora 31 box I have:
```
[galak@spiff ~]$ rpm -qf /usr/include/python3.8/Python.h
python38-3.8.3-1.fc31.x86_64
[galak@spiff ~]$ rpm -qf /usr/include/python^C
[galak@spiff ~]$ rpm -qf /usr/include/python3.7m/Python.h
python3-devel-3.7.7-3.fc31.x86_64
[galak@spiff ~]$ rpm -qf /usr/include/python3.6m/Python.h
python36-3.6.11-1.fc31.x86_64
```

jenmwms, 2020-07-23:
Interesting. I see the following:

jenmwms, 2020-07-23:
> Thanks, @galak! That (python3-devel) did not fix the issue either.

Ok, I tried a few things to clarify this. 
- The issue persists if I add `python3-devel` to the `installation_linux.rst` instead of `python3-psutil` in the patch #27083.
- But if I separately run `sudo dnf install python3-devel` the install is successful, and the step of running requirement script is successful too.

nashif, 2020-07-23:
not sure what you mean, what is it that you do in this step

> The issue persists if I add python3-devel to the installation_linux.rst instead of python3-psutil in the patch #27083.

that is different from 

> sudo dnf install python3-devel

jenmwms, 2020-07-23:
> not sure what you mean, what is it that you do in this step
> 
> > The issue persists if I add python3-devel to the installation_linux.rst instead of python3-psutil in the patch #27083.

This is what I meant to describe (slight modification from the patch which suggests python3-psutil):
```
diff --git a/doc/getting_started/installation_linux.rst b/doc/getting_started/installation_linux.rst
index 6ae0a2f93f..e7b6c5a7dc 100644
--- a/doc/getting_started/installation_linux.rst
+++ b/doc/getting_started/installation_linux.rst
@@ -87,7 +87,7 @@ need one.
 
          sudo dnf group install "Development Tools" "C Development Tools and Libraries"
          dnf install git cmake ninja-build gperf ccache dfu-util dtc wget \
-           python3-pip python3-tkinter xz file glibc-devel.i686 libstdc++-devel.i686 \
+           python3-devel python3-pip python3-tkinter xz file glibc-devel.i686 libstdc++-devel.i686 \
            SDL2-devel
```

jenmwms, 2020-07-23:
I'm having trouble reproducing this issue.  We can close this and reopen if someone else experiences the symptoms. I think `sudo dnf install python3-devel` helped in the end (no patch), but not sure why I cannot walk it back to reproduce.

jenmwms, 2020-07-23:
> I'm having trouble reproducing this issue.

I can reproduce this on my system with `pip3 uninstall --user psutil` etc.

jenmwms, 2020-09-09:
I suggest we close this as a won't fix. Based on discussion, I think it was a mistake in my setup that most will not encounter as a bug.

closed 2020-09-09 by jenmwms