zephyrproject-rtos/zephyr #27090: LE Coded PHY scanning on nRF9160DK fails

ChristianHirsch, 2020-07-23:
**Describe the bug**
If LE Coded PHY scanning is executed on `nrf9160dk_nrf9160` it exits with `Scanning failed to start (err -5)`.

**To Reproduce**
Steps to reproduce the behavior:

### Flash `nrf9160dk_nrf52840`
In order to prepare nRF52840 on the nRF9160DK board, the `hci_uart` sample with the following `prj.conf` has been used:
```
CONFIG_CONSOLE=n
CONFIG_STDOUT_CONSOLE=n
CONFIG_UART_CONSOLE=n
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_BT=y
CONFIG_BT_HCI_RAW=y
CONFIG_BT_HCI_RAW_H4=y
CONFIG_BT_HCI_RAW_H4_ENABLE=y

CONFIG_BT_CTLR_PHY_CODED=y
CONFIG_BT_LL_NRFXLIB=y
CONFIG_BT_LLL_VENDOR_NORDIC=y
CONFIG_BT_LL_SW_SPLIT=y
```

### Flash `nrf9160dk_nrf9160`
The [`central_hr_coded`](https://github.com/nrfconnect/sdk-nrf/pull/2617) sample is used, with the following modifications:
```
$ mkdir boards
$ cat > boards/nrf9160dk_nrf9160ns.conf << EOF
CONFIG_BT_H4=y
CONFIG_BT_WAIT_NOP=y
CONFIG_UART_INTERRUPT_DRIVEN=y

# Set UARTE2 as non-secure, used for HCI
CONFIG_SPM_NRF_UARTE2_NS=y

CONFIG_BT_CTLR_PHY_CODED=n
CONFIG_BT_LL_NRFXLIB=n

CONFIG_BSD_LIBRARY_TRACE_ENABLED=n
EOF
$ cat > boards/nrf9160dk_nrf9160ns.overlay << EOF
CONFIG_BT_H4=y
CONFIG_BT_WAIT_NOP=y
CONFIG_UART_INTERRUPT_DRIVEN=y

# Set UARTE2 as non-secure, used for HCI
CONFIG_SPM_NRF_UARTE2_NS=y

CONFIG_BT_CTLR_PHY_CODED=n
CONFIG_BT_LL_NRFXLIB=n

CONFIG_BSD_LIBRARY_TRACE_ENABLED=n
```

**Expected behavior**
Sample scans and connects to a board running `peripheral_hr_coded`. This works if the `central_hr_coded` is flashed to `nrf9160dk_nrf52840`.

**Logs and console output**
```
*** Booting Zephyr OS build v2.3.0-rc1-ncs1-1437-g9c6523e0349e  ***
Bluetooth initialized
Scanning failed to start (err -5)
[00:00:00.027,008] <wrn> bt_hci_core: opcode 0x0000 pool id 5 pool 0x2002049c != &hci_cmd_pool 0x20020474
[00:00:00.049,499] <inf> bt_hci_core: Identity: d4:a3:d4:62:c8:b0 (random)
[00:00:00.049,530] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1
[00:00:00.049,530] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff
[00:00:00.052,734] <wrn> bt_hci_core: opcode 0x2041 status 0x01
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - sdk-nrf v1.3.0 (https://github.com/nrfconnect/sdk-nrf/pull/2617)
 - Zephyr SDK: 0.11.2

**Additional context**
Another strange thing that I experienced is that the `CONFIG_BT_LL*` configs in the `prf.conf` file for the `hci_uart` sample has to be in exact this order. Otherwise it will generate warnings that some configs aren't selected.
The communication between the nRF52840 and nRF9160 has been tested with the `central_hr` sample and another board running the `peripheral_hr` sample. This works well.

ChristianHirsch, 2020-07-23:
FYI @joerchan

carlescufi, 2020-07-23:
I think this is likely because coded phy scanning is not implemented at the LL level right @joerchan and @andrzej-kaczmarek?

joerchan, 2020-07-23:
> I think this is likely because coded phy scanning is not implemented at the LL level right @joerchan and @andrzej-kaczmarek?

@carlescufi No this is with CONFIG_BT_LL_NRFXLIB=y, so it should be supported. He is saying that it works with board nrf52840dk_nrf52840, but not on nRF9160DK and doing the same with a HCI UART to the nrf52840 on the DK.

joerchan, 2020-07-23:
@ChristianHirsch You have mixed up a few configs, and hci_uart has been compiled with the wrong controller.

`CONFIG_BT_LL_SW_SPLIT` and `CONFIG_BT_LL_NRFXLIB` are choices, you can only select one of them, and in this case you want  `CONFIG_BT_LL_NRFXLIB=y`.`

So remove these from hci_uart/prj.conf
```
CONFIG_BT_LLL_VENDOR_NORDIC=y
CONFIG_BT_LL_SW_SPLIT=y
```

Also you can remove these from the nrf9160 conf files:
```
CONFIG_BT_CTLR_PHY_CODED=n
CONFIG_BT_LL_NRFXLIB=n
```
Not needed when you have `CONFIG_BT_H4=y`

ChristianHirsch, 2020-07-23:
@joerchan well `CONFIG_BT_LL_NRFXLIB=y` is what I had in the first place, but I got the following warning when compiling/configuring:
```
warning: BT_CTLR_DTM_HCI (defined at subsys/bluetooth/controller/Kconfig:492) was assigned the value
'y' but got the value 'n'. Check these unsatisfied dependencies: BT_CTLR_DTM_HCI_SUPPORT (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_BT_CTLR_DTM_HCI.html and/or look up
BT_CTLR_DTM_HCI in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.
```
And in order to get rid of this warning, I had to add `CONFIG_BT_LL_SW_SPLIT=y` and `CONFIG_BT_LLL_VENDOR_NORDIC=y`. Anyway I removed it and flashed the nRF52840 with it. I also changed the nRF9160 config and flashed it and now, it tells me that scanning successfully started, but it is not receiving any advertising packets:
```
*** Booting Zephyr OS build v2.3.0-rc1-ncs1-1437-g9c6523e0349e  ***
Bluetooth initialized
Scanning successfully started
[00:00:00.643,188] <wrn> bt_hci_core: opcode 0x0000 pool id 5 pool 0x2002049c != &hci_cmd_pool 0x20020474
[00:00:00.659,881] <inf> bt_hci_core: Identity: db:ea:20:3f:c0:4b (random)
[00:00:00.659,881] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x10be, manufacturer 0x0059
[00:00:00.659,881] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0x10be
[00:00:01.291,625] <wrn> bt_hci_core: opcode 0x0000 pool id 5 pool 0x2002049c != &hci_cmd_pool 0x20020474
```

joerchan, 2020-07-23:
`BT_CTLR_DTM_HCI` is enabled in the hci_uart/boards/<board>.conf file. But the controller does not support it, so it gets deselected because of that. Which is fine, you don't need DTM (direct test mode).

> I also changed the nRF9160 config and flashed it and now, it tells me that scanning successfully started, but it is not receiving any advertising packets:

Well did you flash the peripheral_hr_coded as well? You are scanning on coded only after all.

ChristianHirsch, 2020-07-23:
> Well did you flash the peripheral_hr_coded as well? You are scanning on coded only after all.

Of course, and I also checked that it is working by flashing the `central_hr_coded` on the nRF52840 and everything works fine. This is the output of the nRF52840 running it:
```
Scanning successfully started
[DEVICE]: 33:bc:66:12:b8:a1 (random), AD type 5, AD props 17, AD data len 11, RSSI -37, Primary PHY 4, Secondary PHY 4
[AD]: 1 data_len 1
[AD]: 3 data_len 6
Connection pending
Connected: 33:bc:66:12:b8:a1 (random), tx_phy 4, rx_phy 4
[ATTRIBUTE] handle 20
[ATTRIBUTE] handle 21
[ATTRIBUTE] handle 23
[SUBSCRIBED]
[NOTIFICATION] data 0x20006a67 length 2
[NOTIFICATION] data 0x20006a67 length 2
```

joerchan, 2020-07-23:
Ok, just checking :)

One thing I noticed is that the `hci_uart`  buffer lengths needs to be increased when enabling EXT_ADV.
Can you try adding these to the hci_uart/prj.conf?
```
BT_RX_BUF_LEN=258
BT_DISCARDABLE_BUF_SIZE=257
```
These are the default values when `CONFIG_BT_EXT_ADV=y`, please confirm that they are set correctly in the `build/zephyr/.config` file for the `central_hr_coded` and `peripheral_hr_coded`.

joerchan, 2020-07-23:
@ChristianHirsch I had a look at some of my older branches where I was using hci_uart and extended advertising. And there I had these configuration options in addition:
```
CONFIG_MAIN_STACK_SIZE=1024
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=1024
```
You should give those a try as well.

ChristianHirsch, 2020-07-24:
Yes! Finally it works, thank you so much for your help :)

Increasing `CONFIG_MAIN_STACK_SIZE` and `CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE` in the `hci_uart` sample did the trick. It is not necessary to increase `BT_RX_BUF_LEN` and `BT_DISCARDABLE_BUF_SIZE` though, but I did it anyway.

Wouldn't it make sense to permanently increase the stack sizes in the `hci_uart` sample, at least for the `nrf9160dk_nrf52840` in the `boards/nrf9160dk_nrf52840.conf` file? I could also imagine that increasing the stack size for the `hci_usb` samples may be useful as well if somebody want to use LE Coded PHY on a PC with a dongle..

joerchan, 2020-07-24:
> Wouldn't it make sense to permanently increase the stack sizes in the hci_uart sample, at least for the nrf9160dk_nrf52840 in the boards/nrf9160dk_nrf52840.conf file?

Actually. I had already done this. But I missed the nrf9160dk_nrf52840 board. So I will have to update that one as well.
It is only the `CONFIG_MAIN_STACK_SIZE=1024` that is needed `, the system worqueue stack size is large enough at 512.
```
thread_analyzer:  main                : unused 208 usage 816 / 1024 (79 %)
thread_analyzer:  sysworkq            : unused 376 usage 136 / 512 (26 %)
```

> It is not necessary to increase BT_RX_BUF_LEN and BT_DISCARDABLE_BUF_SIZE though, but I did it anyway.

No, it is. hci_uart crashed for me using the shell when trying to set advertising data. Advertising data with extended advertising can be up to 251 bytes in a single HCI command.

TwistedSim, 2020-07-24:
Does this apply to hci_usb example as well?

ChristianHirsch, 2020-07-27:
@TwistedSim I guess so, but I'm not an expert. @joerchan does it apply to the `hci_usb` sample as well?

joerchan, 2020-07-27:
@TwistedSim Increasing the buffer sizes should be done if you want to use Extended Advertising in the `hci_usb` sample.
```
BT_RX_BUF_LEN=258
BT_DISCARDABLE_BUF_SIZE=257
```
Changing the MAIN_STACK size is not needed it seems, the main function of that sample is simple, and the sample prj.conf does not change the default, which means it should be 1024 by default.

PR #27126 (2020-07-28): Bluetooth: hci_uart: Update configuration for extended advertising
Update HCI uart sample so that it can be used for extended advertising without having to override the configuration defaults.

Fixes: #27090
closed 2020-07-28 by carlescufi