zephyrproject-rtos/zephyr #27095: Track LVGL ROM footprint improvements

gmarull, 2020-07-23:
**Is your enhancement proposal related to a problem? Please describe.**

ROM usage is significantly higher on LVGL v7 even for minimal configurations. This is in part due to new features such as the new drawing system. However, it looks like there is room to make some improvements on LVGL to reduce the ROM footprint.

**Describe the solution you'd like**

LVGL maintainers are currently investigating ways for reducing the library footprint when some options are not enabled. When these changes land in new releases of LVGL, Zephyr should also be updated (preferably before 2.4 release)

There is an open issue to track progress on ROM usage improvements: https://github.com/lvgl/lvgl/issues/1669

**Describe alternatives you've considered**

N/A

**Additional context**

https://github.com/zephyrproject-rtos/zephyr/pull/26982

pabigot, 2020-07-23:
Converted to bug since enhancements aren't tracked and this is a memory-usage regression that was recognized at the time the PR was reviewed.

gmarull, 2020-08-24:
Update: improvements on ROM footprints will be included in v8 and v9 releases. v8 may be ready by the end of this year.

pabigot, 2020-09-17:
@vwj should this still be a medium priority bug that affects the release?  That classification performed its intent by making sure this was re-evaluated before 2.4 came out, but the changes necessary to reduce footprint will not be available for this release so it can't be "fixed".

We could also close it since the check was done.

@nashif

gmarull, 2022-01-20:
I think it is safe to close this now. If Zephyr users find an issue or are concerned about LVGL memory usage, they should report it to the project itself. We can't do anything about that here.

closed 2022-01-20 by gmarull