zephyrproject-rtos/zephyr #27110: lis2dh driver - set other registers on the device

Satkarni, 2020-07-24:
**Is your enhancement proposal related to a problem? Please describe.**
Current lis2dh driver function for setting attributes is not generic enough to modify other registers. Only accelerometer data rate, sampling frequency etc can be set. I need support for changing FIFO mode, power down mode etc.

**Describe the solution you'd like**
Modify the current attribute set function so that any reg inside lis2dh can be set. 

**Describe alternatives you've considered**
Alternatives will include writing an out of tree driver or hacking the driver. However I do not wish to break zephyr abstraction. 

**Additional context**
As can be seen below, the function is not generic enough. 

<pre><font color="#00CD00">static</font> <font color="#00CD00">int</font> lis2dh_acc_config(<font color="#00CD00">struct</font> device *dev, <font color="#00CD00">enum</font> sensor_channel chan, 
<font color="#6C6C6C"> </font><font color="#FFFF00">271 </font>                <font color="#00CD00">enum</font> sensor_attribute attr,     
<font color="#6C6C6C"> </font><font color="#CDCD00">  1 </font>                <font color="#00CD00">const</font> <font color="#00CD00">struct</font> sensor_value *val) 
<font color="#6C6C6C"> </font><font color="#CDCD00">  2 </font>{
<font color="#6C6C6C"> </font><font color="#CDCD00">  3 </font>    <font color="#CDCD00">switch</font> (attr) {
<font color="#6C6C6C"> </font><font color="#CDCD00">  4 </font><font color="#CD00CD">#ifdef CONFIG_LIS2DH_ACCEL_RANGE_RUNTIME</font>
<font color="#6C6C6C"> </font><font color="#CDCD00">  5 </font>    <font color="#CDCD00">case</font> SENSOR_ATTR_FULL_SCALE:    
<font color="#6C6C6C"> </font><font color="#CDCD00">  6 </font>        <font color="#CDCD00">return</font> lis2dh_acc_range_set(dev, sensor_ms2_to_g(val));
<font color="#6C6C6C"> </font><font color="#CDCD00">  7 </font><font color="#CD00CD">#endif</font>
<font color="#6C6C6C"> </font><font color="#CDCD00">  8 </font><font color="#CD00CD">#ifdef CONFIG_LIS2DH_ODR_RUNTIME</font>
<font color="#6C6C6C"> </font><font color="#CDCD00">  9 </font>    <font color="#CDCD00">case</font> SENSOR_ATTR_SAMPLING_FREQUENCY: 
<font color="#6C6C6C"> </font><font color="#CDCD00"> 10 </font>        <font color="#CDCD00">return</font> lis2dh_acc_odr_set(dev, val-&gt;val1);
<font color="#6C6C6C"> </font><font color="#CDCD00"> 11 </font><font color="#CD00CD">#endif</font>
<font color="#6C6C6C"> </font><font color="#CDCD00"> 12 </font><font color="#CD00CD">#if defined(CONFIG_LIS2DH_TRIGGER)</font>
<font color="#6C6C6C"> </font><font color="#CDCD00"> 13 </font>    <font color="#CDCD00">case</font> SENSOR_ATTR_SLOPE_TH:
<font color="#6C6C6C"> </font><font color="#CDCD00"> 14 </font>    <font color="#CDCD00">case</font> SENSOR_ATTR_SLOPE_DUR:
<font color="#6C6C6C"> </font><font color="#CDCD00"> 15 </font>        <font color="#CDCD00">return</font> lis2dh_acc_slope_config(dev, attr, val);
<font color="#6C6C6C"> </font><font color="#CDCD00"> 16 </font><font color="#CD00CD">#endif</font>
<font color="#6C6C6C"> </font><font color="#CDCD00"> 17 </font>    <font color="#CDCD00">default</font>:
<font color="#6C6C6C"> </font><font color="#CDCD00"> 18 </font>        LOG_DBG(<font color="#AF5F00">&quot;Accel attribute not supported.&quot;</font>);
<font color="#6C6C6C"> </font><font color="#CDCD00"> 19 </font>        <font color="#CDCD00">return</font> -<font color="#AF5F00">ENOTSUP</font>;
<font color="#6C6C6C"> </font><font color="#CDCD00"> 20 </font>    }
<font color="#6C6C6C"> </font><font color="#CDCD00"> 21 </font>
<font color="#6C6C6C"> </font><font color="#CDCD00"> 22 </font>    <font color="#CDCD00">return</font> <font color="#AF5F00">0</font>;
<font color="#6C6C6C"> </font><font color="#CDCD00"> 23 </font>}
<font color="#6C6C6C"> </font><font color="#CDCD00"> 24 </font>
<font color="#6C6C6C"> </font><font color="#CDCD00"> 25 </font><font color="#00CD00">static</font> <font color="#00CD00">int</font> lis2dh_attr_set(<font color="#00CD00">struct</font> device *dev, <font color="#00CD00">enum</font> sensor_channel chan, 
<font color="#6C6C6C"> </font><font color="#CDCD00"> 26 </font>               <font color="#00CD00">enum</font> sensor_attribute attr,     
<font color="#6C6C6C"> </font><font color="#CDCD00"> 27 </font>               <font color="#00CD00">const</font> <font color="#00CD00">struct</font> sensor_value *val) 
<font color="#6C6C6C"> </font><font color="#CDCD00"> 28 </font>{
<font color="#6C6C6C"> </font><font color="#CDCD00"> 29 </font>    <font color="#CDCD00">switch</font> (chan) {
<font color="#6C6C6C"> </font><font color="#CDCD00"> 30 </font>    <font color="#CDCD00">case</font> SENSOR_CHAN_ACCEL_X:
<font color="#6C6C6C"> </font><font color="#CDCD00"> 31 </font>    <font color="#CDCD00">case</font> SENSOR_CHAN_ACCEL_Y:
<font color="#6C6C6C"> </font><font color="#CDCD00"> 32 </font>    <font color="#CDCD00">case</font> SENSOR_CHAN_ACCEL_Z:
<font color="#6C6C6C"> </font><font color="#CDCD00"> 33 </font>    <font color="#CDCD00">case</font> SENSOR_CHAN_ACCEL_XYZ:
<font color="#6C6C6C"> </font><font color="#CDCD00"> 34 </font>        <font color="#CDCD00">return</font> lis2dh_acc_config(dev, chan, attr, val);
<font color="#6C6C6C"> </font><font color="#CDCD00"> 35 </font>    <font color="#CDCD00">default</font>:
<font color="#6C6C6C"> </font><font color="#CDCD00"> 36 </font>        LOG_WRN(<font color="#AF5F00">&quot;attr_set() not supported on this channel.&quot;</font>);
<font color="#6C6C6C"> </font><font color="#CDCD00"> 37 </font>        <font color="#CDCD00">return</font> -<font color="#AF5F00">ENOTSUP</font>;
<font color="#6C6C6C"> </font><font color="#CDCD00"> 38 </font>    }
<font color="#6C6C6C"> </font><font color="#CDCD00"> 39 </font>
<font color="#6C6C6C"> </font><font color="#CDCD00"> 40 </font>    <font color="#CDCD00">return</font> <font color="#AF5F00">0</font>;
<font color="#6C6C6C"> </font><font color="#CDCD00"> 41 </font>}
</pre>

kartben, 2024-02-08:
@avisconti is it still relevant?

avisconti, 2024-02-08:
> @avisconti is it still relevant?

It is a quite old request. I think this request is not clearly identifyng what attributes to add. Moreover FIFO with sensor api v1 cannot be supported.

avisconti, 2024-02-14:
I closed the issue as it is not clearly defining what are the attributes requested. Pls feel free to reopen it with more details.

closed 2024-02-14 by avisconti