zephyrproject-rtos/zephyr #27112: mcumgr fs download crashes with v2.3.0

haipas, 2020-07-24:
**Describe the bug**
Hi, I use the mcumgr to exchange files between Zephyr (nrf52840 with SD-Card via SPI1) and a host.
After updating from v2.2.0 to v2.3.0 the command mcumgr fs download causes a bus fault error.
The commands echo and fs upload work without problems.

**To Reproduce**
Steps to reproduce the behavior:
ok: /go # mcumgr --conntype ble --connstring peer_name=Zephyr echo test
ok: /go # mcumgr --conntype ble --connstring peer_name=Zephyr fs upload test.txt /SD:/test.txt

Fail: /go # mcumgr --conntype ble --connstring peer_name=Zephyr fs download /SD:/test.txt test.cntrl

**Expected behavior**
The desired file should be transferred. As it works with the fs upload command and also in v2.2.0

**Impact**
No measurement data can be downloaded from the device via BLE

**Logs and console output**
*** Booting Zephyr OS build zephyr-v2.3.0-142-g52207f6b7fb1  ***
[00:00:00.838,714] <inf> spi_nrfx_spi: CS control inhibited (no GPIO device)
[00:00:00.906,066] <inf> sdhc_spi: Found a ~30436 MiB SDHC card.
[00:00:00.907,318] <inf> sdhc_spi: Manufacturer ID=3 OEM='SD' Name='SC32G' Revision=0x80 Serial=0x4d0d3bf5
[00:00:00.907,348] <inf> smp_sample: Block count 62333952
[00:00:00.907,348] <inf> smp_sample: Sector size 512
[00:00:00.907,379] <inf> smp_sample: Memory Size(MB) 1764
[00:00:00.974,670] <inf> sdhc_spi: Found a ~30436 MiB SDHC card.
[00:00:00.975,952] <inf> sdhc_spi: Manufacturer ID=3 OEM='SD' Name='SC32G' Revision=0x80 Serial=0x4d0d3bf5
[00:00:00.975,982] <inf> spi_nrfx_spi: CS control inhibited (no GPIO device)
[00:00:00.990,722] <inf> smp_sample: Disk mounted.
[00:00:00.990,722] <inf> littlefs: LittleFS version 2.2, disk version 2.0
[00:00:00.990,936] <inf> littlefs: FS at NRF_FLASH_DRV_NAME:0xfc000 is 4 0x1000-byte blocks with 512 cycle
[00:00:00.990,966] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.991,210] <inf> littlefs: /lfs mounted
[00:00:00.993,194] <inf> smp_bt_sample: Bluetooth initialized
[00:00:00.993,682] <inf> smp_bt_sample: Advertising successfully started
[00:00:00.993,835] <inf> smp_sample: build time: Jul 24 2020 09:28:22
[00:00:06.136,932] <inf> smp_bt_sample: Connected
[00:00:06.343,322] <err> os: ***** BUS FAULT *****
[00:00:06.343,322] <err> os:   Instruction bus error
[00:00:06.343,353] <err> os: r0/a1:  0xffffffbc  r1/a2:  0x0002ff50  r2/a3:  0x00000000
[00:00:06.343,353] <err> os: r3/a4:  0x00000001 r12/ip:  0xffffffff r14/lr:  0x00011cbd
[00:00:06.343,353] <err> os:  xpsr:  0x20000000
[00:00:06.343,353] <err> os: Faulting instruction address (r15/pc): 0x9c676fc6
[00:00:06.343,353] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:06.343,353] <err> os: Current thread: 0x20003804 (unknown)
[00:00:06.654,663] <err> os: Halting system

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - zephyr-sdk-0.11.2
 - zephyr-v2.3.0-142-g52207f6b7fb1

**Additional context**
smp_svr sample v2.3.0
[smp_svr_v2.3.0.zip](https://github.com/zephyrproject-rtos/zephyr/files/4970792/smp_svr_v2.3.0.zip)
smp_svr sample v2.2.0
[smp_svr_v2.2.0.zip](https://github.com/zephyrproject-rtos/zephyr/files/4970801/smp_svr_v2.2.0.zip)

MaureenHelm, 2020-07-28:
Can you reproduce this issue on master?

haipas, 2020-07-29:
Yes, the problem still exists with the current master. But another error message occurs

`*** Booting Zephyr OS build zephyr-v2.3.0-1372-g5e2e7a0fcf6f  ***
[00:00:00.000,000] <inf> spi_nrfx_spi: CS control inhibited (no GPIO device)
[00:00:00.213,043] <inf> sdhc_spi: Found a ~30436 MiB SDHC card.
[00:00:00.214,324] <inf> sdhc_spi: Manufacturer ID=3 OEM='SD' Name='SC32G' Revision=0x80 Serial=0x4d0d3bf5
[00:00:00.214,355] <inf> smp_sample: Block count 62333952
[00:00:00.214,355] <inf> smp_sample: Sector size 512
[00:00:00.214,355] <inf> smp_sample: Memory Size(MB) 1764
[00:00:00.281,677] <inf> sdhc_spi: Found a ~30436 MiB SDHC card.
[00:00:00.282,928] <inf> sdhc_spi: Manufacturer ID=3 OEM='SD' Name='SC32G' Revision=0x80 Serial=0x4d0d3bf5
[00:00:00.340,637] <inf> smp_sample: Disk mounted.
[00:00:00.340,637] <inf> littlefs: LittleFS version 2.2, disk version 2.0
[00:00:00.340,850] <inf> littlefs: FS at NRF_FLASH_DRV_NAME:0xfc000 is 4 0x1000-byte blocks with 512 cycle
[00:00:00.340,881] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.341,125] <inf> littlefs: /lfs mounted
[00:00:00.343,231] <inf> smp_bt_sample: Bluetooth initialized
[00:00:00.343,719] <inf> smp_bt_sample: Advertising successfully started
[00:00:00.343,902] <inf> smp_sample: build time: Jul 29 2020 08:09:09
[00:00:05.102,630] <inf> smp_bt_sample: Connected
[00:00:06.033,325] <inf> smp_bt_sample: Disconnected (reason 0x08)
[00:00:06.034,332] <inf> smp_bt_sample: Advertising successfully started
[00:00:55.288,116] <inf> smp_bt_sample: Connected
[00:00:55.502,166] <err> os: ***** USAGE FAULT *****
[00:00:55.502,197] <err> os:   Illegal use of the EPSR
[00:00:55.502,197] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000001  r2/a3:  0x001bac3c
[00:00:55.502,197] <err> os: r3/a4:  0x00000000 r12/ip:  0x6d617375 r14/lr:  0x00020a20
[00:00:55.502,197] <err> os:  xpsr:  0x60000000
[00:00:55.502,197] <err> os: Faulting instruction address (r15/pc): 0x00020a20
[00:00:55.502,227] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:55.502,227] <err> os: Current thread: 0x200013f8 (idle 00)
[00:00:55.817,077] <err> os: Halting system`

de-nordic, 2020-08-24:
Are you able to check if the issue persists when performing the same operations on lfs attached to internal storage area instead of external SPI connected SD?

haipas, 2020-09-14:
Sorry for the late answer I was on vacation.
The problem also occurs with lfs. 

```
*** Booting Zephyr OS build zephyr-v2.3.0-151-g3279a87dc96b  ***
[00:00:05.020,233] <inf> littlefs: LittleFS version 2.2, disk version 2.0
[00:00:05.020,446] <inf> littlefs: FS at NRF_FLASH_DRV_NAME:0xfc000 is 4 0x1000-byte blocks with 512 cycle
[00:00:05.020,446] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:05.020,690] <inf> littlefs: /lfs mounted
[00:00:05.022,583] <inf> smp_bt_sample: Bluetooth initialized
[00:00:05.023,071] <inf> smp_bt_sample: Advertising successfully started
[00:00:05.023,223] <inf> smp_sample: build time: Sep 14 2020 15:55:24
[00:00:12.481,048] <inf> smp_bt_sample: Connected
[00:00:12.680,053] <err> fs: failed get file or dir stat (-2)
[00:00:13.413,665] <inf> smp_bt_sample: Disconnected (reason 0x08)
[00:00:13.414,642] <inf> smp_bt_sample: Advertising successfully started
[00:00:57.411,224] <inf> smp_bt_sample: Connected
[00:00:57.633,300] <err> os: ***** USAGE FAULT *****
[00:00:57.633,300] <err> os:   Illegal load of EXC_RETURN into PC
[00:00:57.633,300] <err> os: r0/a1:  0x20000188  r1/a2:  0xe000ed00  r2/a3:  0x200001f0
[00:00:57.633,300] <err> os: r3/a4:  0x00000001 r12/ip:  0x0000003c r14/lr:  0x200057b9
[00:00:57.633,300] <err> os:  xpsr:  0x20000000
[00:00:57.633,331] <err> os: Faulting instruction address (r15/pc): 0x200057b1
[00:00:57.633,331] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:57.633,331] <err> os: Current thread: 0x20001c8c (unknown)
[00:00:57.950,958] <err> os: Halting system
```

de-nordic, 2020-09-14:
I do not think that the problem is with mcumgr. The problem is with SPI or storage subsystem.

haipas, 2020-09-15:
@de-nordic I don't think I have a general problem with the SPI or the SD-Card. My application can read/write to the SD card without any problems and I can send files to the SD card with the fs upload command.

The error occurs with the following function in the file subsys/mgmt/smp.c 

```
/**
 * Processes all received SNP request packets.
 */
static void
zephyr_smp_handle_reqs(struct k_work *work)
{
	struct zephyr_smp_transport *zst;
	struct net_buf *nb;

	zst = (void *)work;

	while ((nb = k_fifo_get(&zst->zst_fifo, K_NO_WAIT)) != NULL) {
		zephyr_smp_process_packet(zst, nb);
	}
}
```

de-nordic, 2020-09-15:
Have you tried to follow this advice: https://github.com/zephyrproject-rtos/zephyr/issues/27367#issuecomment-692098385?

haipas, 2020-09-15:
Ok, increasing `CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE` resolved the problem... Many thanks for the help

closed 2020-09-15 by haipas