zephyrproject-rtos/zephyr #27151: buildkite: samples: net: echo_server: Doesn't run all configurations from atmel_rf2xx shield

nandojve, 2020-07-26:
**Describe the bug**
Inspection on **buildkite 2/20** from #26655 reveals that not all configuration have been built. I suspect that boards that don't have a **netif** interface will not build/run with shields. This need more inspection to make sure it is not affecting more configurations.

**To Reproduce**
Clone main repository and run **./scripts/sanitycheck --testcase-root samples/net/sockets/echo_server**.
It seems to run even less combinations.

**Expected behavior**
All configurations should run and must build all combination at buildkite to ensure that CI is OK and Zephyr cover all test scenarios.

**Impact**
Many configurations may not be running as expected and we can accept broke code.

**Logs and console output**
1- Goto https://buildkite.com/zephyr/zephyr/builds/2728#8c66b445-9f4f-41ed-a431-3c8769e3247e
2- Search by **rf2xx** and check builds 76 up to 93.
3- Confirm that below configurations hadn't build:
3.1 - sample.net.sockets.echo_server.802154.rf2xx
3.2 - sample.net.sockets.echo_server.802154.rf2xx.xplained
3.3 - sample.net.sockets.echo_server.802154.rf2xx.mikrobus

**Environment (please complete the following information):**
Current Buildkite CI

CC: @nashif @jukkar

nashif, 2020-08-31:
> All configurations should run and must build all combination at buildkite to ensure that CI is OK and Zephyr cover all test scenarios.

This is not the expected behavior, we do not build everything and all combination, this is done on a nightly basis, but not per PR.

nandojve, 2020-08-31:
Hi @nashif 

Maybe I not expressed my self on a clear way. When I wrote "all configurations" I'm referring at sample/test with a whitelist. So, not all whitelist combination are building. In this case, I created combinations to make sure that a change on that overlays + SoCs are Ok on a minimal set. Those are the minimal build set that should pass on CI when a PR is sent to cover all code variations. Otherwise, code can be changed and not compile, however PR can be accepted as OK.

For instance, on my view the below fragment should build on CI, but it is not.

```
  sample.net.sockets.echo_server.802154.rf2xx.mikrobus:
    extra_args: SHIELD=atmel_rf2xx_mikrobus OVERLAY_CONFIG="overlay-802154.conf"
    platform_whitelist: lpcxpresso55s69_ns
```

I'm raising this because I've been seen code changes that not even built but PRs were accepted as OK. Than people start to open issues. This could create a feeling that CI is not trustworthy, and that kind thing I'm trying to avoid opening this PR.

nandojve, 2020-09-12:
CI not cover all paths, see #28341

nashif, 2020-09-13:
the one is not building in CI because the testcase depends on netif, this board does not provide it, so this:

```
diff --git a/boards/arm/lpcxpresso55s69/lpcxpresso55s69_ns.yaml b/boards/arm/lpcxpresso55s69/lpcxpresso55s69_ns.yaml
index a78352ecf7..fb91153a41 100644
--- a/boards/arm/lpcxpresso55s69/lpcxpresso55s69_ns.yaml
+++ b/boards/arm/lpcxpresso55s69/lpcxpresso55s69_ns.yaml
@@ -19,3 +19,4 @@ supported:
   - gpio
   - spi
   - watchdog
+  - netif
```
would make it at least buildable... 
However, you then hit the next problem that the test/app itself does not fit on the board, so it overflows. None of this is a bug of sanitycheck, this is just test configuration.

nandojve, 2020-09-13:
Hi @nashif , this helps a lot to understand the problem but not solve them. The boards that are using shields could never set the depends on at yaml file. Why? These are are shields and can be used on any board `extra_args: SHIELD=atmel_rf2xx_mikrobus OVERLAY_CONFIG="overlay-802154.conf"`.

Are there any way to address the depends on for shields? How shields can expose "netif", "ieee802.15.4", etc for instance? I'm asking because I don't see any *.yaml on shields directory and I'm not sure about it.

nashif, 2020-09-13:
Can you please stop reopening this as a high priority bug, it is not a bug and it is not a high priority one for that matter.
Please open an enhancement request with the support you are looking for, this is going to be fixed probably with https://github.com/zephyrproject-rtos/zephyr/pull/25703 when it lands, then the filter would work on DTS and what the shield provides.

closed 2020-09-13 by nashif

nandojve, 2020-09-14:
Hi @nashif , sorry if I'm disturbing, this is not my intention. Now I have a good picture about what is happening and how things will move forward.
Thank you!