zephyrproject-rtos/zephyr #27154: bt_conn_le_param_update doesn't return an error when setting the timeout >30sec, stops device from sleeping on nrf52840

jeremyherbert, 2020-07-26:
**Describe the bug**
When setting the connection timeout to >30 seconds on an nrf52840, no error is returned. But by looking at the current draw of the device, it appears to prevent the device going into idle sleep mode.

**To Reproduce**
1. Use any of the bluetooth examples where the device connects
2. in the connected callback, add the following code:

```
    int err;

    struct bt_le_conn_param param = BT_LE_CONN_PARAM_INIT(BT_GAP_INIT_CONN_INT_MIN, BT_GAP_INIT_CONN_INT_MIN, 30, 3100);
    err = bt_conn_le_param_update(conn, &param);

    if (err) {
        printk("Failed to update connection parameters\n");
        return;
    }
```

**Expected behavior**
Some sort of error code should be returned indicating that the timeout could not be set. Device should still be able to sleep.

**Environment (please complete the following information):**
 - OS: Mac OS Catalina (10.15.4)
 - Toolchain: gnuarmemb
 - Commit SHA: bbd71e23a2d56f05cefc9dca7bef609c22c1966c

**Additional context**
Using a custom nrf52840 board, but should work just the same on any nrf52840.

joerchan, 2020-07-29:
Please elaborate on what you mean by device cannot sleep?
There is an active Bluetooth LE connection, it will consume power to maintain the connection.
The connection parameters you are setting is 30 ms, with a latency of 30, meaning that the peripheral will be active in the connection every 900 ms.

> Use any of the bluetooth examples where the device connects
Please be more specific, roles can have different roles.

> Some sort of error code should be returned indicating that the timeout could not be set.
The timeout could be set, the maximum timeout is 32 seconds, you have set 31 seconds. So this should be fine.
Also the connection parameters are not set until you get `le_param_updated` callback. There you will see the actual parameters set, it is the central that decides the connection parameters, so if you are the peripheral role your connection parameters may be denied.

jeremyherbert, 2020-07-29:
I am closing this, it turns out that the issue was that the change was rejected by the central. I didn't know about the le_param_updated callback, thank you for the tip.

closed 2020-07-29 by jeremyherbert