zephyrproject-rtos/zephyr #27160: fragile coordination between USB configuration and USB console

pabigot, 2020-07-26:
Since introduced in fca0add22b93b18f292d1693ca3f59c60256cde3 the USB console infrastructure has included a 1 s busy-wait in console device startup after DTR has been detected.  There is no documentation of why this is here.

Experimentation indicates that if the delay is removed the first couple bytes of output on the console do not appear.  This suggests that there is some other USB event that should be used to gate completion of the console driver initialization: DTR alone is not sufficient.

This in turn relates to #27071 as it suggests that the UART console may need to register for USB callbacks before it starts to service console output.  Unless some ideas behind https://github.com/zephyrproject-rtos/zephyr/pull/27025#discussion_r458659540 allow communication of the ready state to the console driver which is busy spin-waiting for it.

And possibly related: If `CONFIG_USB_UART_DTR_WAIT=n` then `console_out()` in the console driver may be asked to output data before the console is ready, which can among other things produce "Device not configured, dropping" log warnings from the CDC_ACM driver, which in turn might cause more console output, more messages, and a cascade that (as it turns out) triggers #26970.

In short, the coordination between USB startup and USB console being ready for use needs some attention.

emob-nordic, 2020-08-07:
Does it also drop the characters if You remove 1s busy wait and wait for detected baudrate?

pabigot, 2020-08-10:
> Does it also drop the characters if You remove 1s busy wait and wait for detected baudrate?

@emob-nordic That won't work, because the baud rate is set only when the device is first opened AND only if the rate differs from the last configured rate (at least on Linux).  In other words in many cases there is no `USB_CDC_LINE_CTRL_BAUD_RATE` control signal.

To avoid:

    <wrn> usb_cdc_acm: Device not configured, drop 1 bytes

and loss of the first character of output it's necessary to wait for the `USB_DC_CONFIGURED` event (which precedes DTR being asserted, and is the sixth event received on my system), or for DTR.

pabigot, 2020-08-10:
Previous comment about avoiding loss of initial data was incorrect.

On my system it takes a between 500 ms and 700 ms after boot before `USB_DC_CONFIGURED` is signalled.

It's another 490 ms (exactly) until DTR is asserted.

Multiple bytes of data are lost if output is initiated before DTR.

Multiple bytes of data are lost if output is initiated as soon as DTR is asserted.

The only way to reliably get the first byte of data is to `k_msleep(1)` after DTR is asserted.

On my system.

emob-nordic, 2020-08-11:
> Multiple bytes of data are lost if output is initiated before DTR.

What if:
``` diff
diff --git a/subsys/usb/class/cdc_acm.c b/subsys/usb/class/cdc_acm.c
index 9de34b9232..91c1a42ead 100644
--- a/subsys/usb/class/cdc_acm.c
+++ b/subsys/usb/class/cdc_acm.c
@@ -573,6 +573,13 @@ static int cdc_acm_fifo_fill(struct device *dev,
 		return 0;
 	}
 
+#ifdef CONFIG_UART_LINE_CTRL
+	if (!(dev_data->line_state & SET_CONTROL_LINE_STATE_DTR)) {
+		LOG_WRN("Terminal is not ready, drop %d bytes", len);
+		return 0;
+	}
+#endif
+
 	dev_data->tx_ready = false;
 
 	wrote = ring_buf_put(dev_data->tx_ringbuf, tx_data, len);
```

pabigot, 2020-08-11:
I think that would reduce the number of generated errors when using printk too early, but it doesn't help the application find out when it's OK to send data.

I would like to see something like this API added to `<drivers/uart/cdc_acm.h>`:

```

struct cdc_acm_wait_ready_state {
   sys_snode_t node;
   struct sys_notify notify;
};

/** Signal when the UART is in a ready state.
 *
 * This is a convenience function that waits until the DTR signal has
 * been asserted plus a small delay to complete internal setup.  At
 * this point transmission should be successful without dropping data.
 */
int cdc_acm_wait_ready(struct device *dev,
                       struct cdc_acm_wait_ready_state *sp);
```		       

This would be pretty easy to implement, and be safe to invoke from all contexts.

The problem with it is that it would only work on the initial startup. If the conop for the device is that it is connected and disconnected from a USB host during normal operations there are unavoidable race conditions in detecting the ready state.

To handle that we should change the semantics of the callback passed to usb_enable() to instead provide notifications of the current state of the USB infrastructure, rather than the events that cause changes to that state.

Then the conop can be to register for state changes, and use those to detect loss of service.  When service is lost (and on startup) use cdc_acm_wait_ready() to be notified of restoration of service.

Is there a plan to provide the application information about the state of the USB rather than just the DC events?

jfischer-phytec-iot, 2020-08-12:
>I would like to see something like this API added to <drivers/uart/cdc_acm.h>: ...

One reason I was against _drivers/uart/cdc_acm.h_ is that it is like pandora's box. There is common UART API which should be used or expanded.  Special treatment of the UART (emulation) over USB makes also application less portable.

>Is there a plan to provide the application information about the state of the USB rather than just the DC events?

It should be one of the major points but it was not even discussed yet. Device controller (DC) events should not be simple forwarded to the function/classes or even application. I think there should be thee types of the notification: from the device controller/driver to the stack, from stack to the function/classes, and notifications from the stack(ch9)/class to the application.

Is this issue not resolved by #27427?

pabigot, 2020-08-12:
> Is this issue not resolved by #27427?

IMO, no.  There's still no standard infrastructure that the application can use to tell whether it can write data without loss.  "Wait for DTR, then wait a few milliseconds more" is just going to result in code duplication, even if it gets documented somewhere.  (CDC ACM may be one of the few UART implementations where it's possible to detect that there is something on the other end of the serial connection, so It doesn't seem suitable to a generic UART API extension.)

For my applications I know the UARTs are going to be CDC ACM devices, and that they have behavior that is very different from SoC peripheral UARTs.  For the near term need I can work around the functionality gaps because the system is powered by USB so I only need to handle startup.  For others where the system is intermittently plugged in I can't do anything, but it'll be months before I can get to those.

emob-nordic, 2020-08-12:
> To handle that we should change the semantics of the callback passed to usb_enable() to instead provide notifications of the current state of the USB infrastructure, rather than the events that cause changes to that state.

This is what I would like to take care of asap. We've created a list of ToDos for topic branch and this particular issue is there as well (#26935).

emob-nordic, 2020-08-19:
> Multiple bytes of data are lost if output is initiated as soon as DTR is asserted.
The only way to reliably get the first byte of data is to k_msleep(1) after DTR is asserted.

I've checked if those first bytes just after DTR are send - > they are. From samples/subsys/usb/console/src/main.c
```
	/* Poll if the DTR flag was set, optional */
	while (!dtr) {
		uart_line_ctrl_get(dev, UART_LINE_CTRL_DTR, &dtr);
	}

	printk("ABCDEFGH\n");

```
Here is an USB line sniffer.
SetControlLineState - setting DTR
SetLineCoding - setting baudrate
DataIn - Characters, the first one is 'A'

[image]

'A' character is send on USB line, the problem is with console terminal used on HOST side.
I don't think we can do much about it, adding `k_msleep(1);` means we repair the problem that is on host side.
For one set up this will work with 1ms but for others who knows.

jfischer-no, 2024-05-21:
The original description is outdated. There have been several improvements to the sample and CDC ACM UART driver since this issue was opened.

closed 2024-05-21 by jfischer-no