zephyrproject-rtos/zephyr #27161: shell:  start_shell() and stop_shell() can freeze the shell

sslupsky, 2020-07-26:
When using start_shell() and stop_shell() to control the start and stop of the shell when the USB is connected and disconnected (USB is used for console), the shell freezes.  It appears the freeze occurs because the underlying poll that waits for the data never receives the signal.

I previously attempted to resolve this through changes to uninit() in #24404.  It was pointed out that there are other issues with init() and uninit() and recommended I separate the two issues.

So, I removed the uninit() changes from that PR and created this issue to track the outstanding issue of starting and stopping the shell.  My use case is to disable the shell when the USB is disconnected and reenable it when the USB is reconnected.

I found a temporary work around by disabling the transmit and receive interrupts but this doesn't really "disable" and "enable" the shell ... rather, it simply avoids sending anything.

jakub-uC, 2020-08-03:
@sslupsky I think that there is a good solution in PR.

sslupsky, 2020-08-07:
Hi @jakub-uC I had a look at the changes and I do believe they address some outstanding issues with init() and uninit().  Thank you for those.

Regarding the issue with shell_stop() / shell_start() I do not think we are out of the woods yet.  When using the USB_CDC for the shell, I seem to recall that the shell gets hung up at shell_pend_on_txdone() in shell_ops.c.  There is a poll where the signal (SHELL_SIGNAL_TXDONE) is never received so the poll waits forever.  This gets triggered after stopping the shell and starting it again.  When the shell is restarted, there are a few print statements that set the colour and send a couple new lines.  These prints cause the shell to freeze.  I could not determine exactly why that was happening.

It appears the shell uart (USB_CDC) is not stopped when the shell is stopped.  So, it seems possible that the shell uart is still processing data even though the shell is stopped.  And I wonder if that is causing a problem.

jakub-uC, 2020-08-07:
@sslupsky Thank you for testing that. There was one more proposal in your PR that I did not implement: adding mutex in `shell_start`. I couldn't understand how it could help. Most likely I need to spend some more time on the issue.

In the meantime would you mind checking if adding mutex there will help?

PR #27342 (2020-08-18): shell: fix possible deadlock in shell_fprintf
Disable shell print functions if the shell is not initialized.

Fixes #27161
commit 920e64cf (2020-08-18): shell: fix possible deadlock in shell_fprintf

Disable shell print functions if the shell is not initialized.
Update tests accordingly.

Fixes #27161
closed 2020-08-18 by nashif